openwrt/openwrt #17835: No LAN Ports on WRT1900ACS with latests snapshots

juceda, 2025-02-02:
### Describe the bug

Hi

I've noticed that I'm unable to run properly latests snapshots on WRT1900ACS.

Right now I have a working firmware image (SNAPSHOT r28605-9decfc82dc), but for a while now I've noticed that every snapshot I build and flash, the LAN switch doesn't work.

I can connect to the router via WIFI and in LuCI I've seen that the physical network devices have disappeared, therefore there is no LAN communication between my hosts connected via cable.

### OpenWrt version

r28739-69890e16b3

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Linksys WRT1900ACS

### Image kind

Self-built image

### Steps to reproduce

Just upgrade from SNAPSHOT r28605-9decfc82dc to r28739-69890e16b3 (eg.).. and some earlier commits suffer this issue too (I'm sorry, I don't remember the exact commit from which this started to happen).

### Actual behaviour

No LAN ports (lan 1 to 4). Lan bridge contains only wifi aps. 

### Expected behaviour

LAN interface group with LAN switch devices (and wifi) together.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

hauke, 2025-02-02:
The switch driver is build as a module now, see eaa82118eadfd495f8512d55c01c1935b8b42c51

Do you have `kmod-dsa-mv88e6xxx` in your image, this is a default package now?

juceda, 2025-02-02:
Hi hauke,

On my .config is enabled as default package:

CONFIG_DEFAULT_kmod-dsa-mv88e6xxx=y

Also on target/linux/mvebu/image/cortexa9.mk (lines 259 - 273), the module is supposed to be added "automatically?" as well.
    270   DEVICE_PACKAGES += mwlwifi-firmware-88w8864 kmod-dsa-mv88e6xxx

 In fact that line is similar to a recent fix for WRT32X. For that model, the kmod-dsa-mv88e6xxx module was missing, and fixed on commit eb5f98fdf97ac2293c63c4002f16bb92f1e796eb

hauke, 2025-02-02:
Do you have `CONFIG_kmod-dsa-mv88e6xxx=y` in your configuration? This is needed.

Do you have the file `/lib/modules/6.6.74/mv88e6xxx.ko`

Is the mv88e6xxx module listed in the `lsmod` output?

juceda, 2025-02-03:
Well... manually selecting kmod-dsa-mv88e6xxx on menu config fixed the issue.

It seems I incorrectly assumed that if I selected the proper target profile (WRT1900ACS/Shelby) the required modules would be auto-checked as well.

Thanks for your support Hauke!

closed 2025-02-03 by juceda