openwrt/openwrt #17836: Raspberry Pi 5: Slow WiFi bandwidth when USB WiFi adapter is plugged

Neirop, 2025-02-02:
### Describe the bug

I want to use a Raspberry pi 5 as a WiFi extender, I have a MediaTek MT7921AU WiFi adapter.
When I'm not plugging my WiFi adapter, the bandwidth is what I expect for 2.4GHz.

[image]

But if I plug my WiFi adapter, the bandwidth drops:

[image]

Also I can't connect to the AP with the WiFi adapter, it doesn't detect the AP.

Note:
I'm using the 5.1V/5A recommended charger for the Raspberry Pi 5.
I tried the same build config for a Raspberry Pi 3B with the same setup and I don't have any problem:

[image]

### OpenWrt version

r28417-daef29c75d

### OpenWrt release

24.10.0-rc7

### OpenWrt target/subtarget

bcm27xx/bcm2712

### Device

Raspberry Pi 5 Model B Rev 1.0

### Image kind

Self-built image

### Additional info

I tried to plug the WiFi adapter on USB2 or USB3, I tried with a minimal build config and I get the same results.

The system log:
[system_log.txt](https://github.com/user-attachments/files/18633208/system_log.txt)

### Diffconfig

```text
CONFIG_TARGET_bcm27xx=y
CONFIG_TARGET_bcm27xx_bcm2712=y
CONFIG_TARGET_bcm27xx_bcm2712_DEVICE_rpi-5=y
CONFIG_DOCKER_STO_EXT4=y
CONFIG_DRIVER_11AX_SUPPORT=y
CONFIG_GNUTLS_ALPN=y
CONFIG_GNUTLS_ANON=y
CONFIG_GNUTLS_DTLS_SRTP=y
CONFIG_GNUTLS_HEARTBEAT=y
CONFIG_GNUTLS_OCSP=y
CONFIG_GNUTLS_PSK=y
CONFIG_HTOP_LMSENSORS=y
CONFIG_KERNEL_PAGE_POOL=y
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_MBEDTLS=y
CONFIG_LIBCURL_NGHTTP2=y
CONFIG_LIBCURL_NO_SMB="!"
CONFIG_LIBCURL_PROXY=y
[... 238 lines cut ...]
CONFIG_PACKAGE_ucode-mod-lua=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_usbids=y
CONFIG_PACKAGE_usbutils=y
CONFIG_PACKAGE_v4l2camera=y
CONFIG_PACKAGE_v4l2tools=y
CONFIG_PACKAGE_vnstat2=y
CONFIG_PACKAGE_vnstati2=y
CONFIG_PACKAGE_wireguard-tools=y
CONFIG_PACKAGE_xtables-nft=y
CONFIG_PACKAGE_zlib=y
CONFIG_SQLITE3_COLUMN_METADATA=y
CONFIG_SQLITE3_DYNAMIC_EXTENSIONS=y
CONFIG_SQLITE3_FTS3=y
CONFIG_SQLITE3_FTS4=y
CONFIG_SQLITE3_FTS5=y
CONFIG_SQLITE3_RTREE=y
CONFIG_TARGET_ROOTFS_PARTSIZE=4096
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

Noltari, 2025-02-03:
@Neirop assuming that you're connecting the wifi adapter to an USB3 port, what happens if you connect it to an USB2 port?
Can you also try using an extension usb cable?
I wonder if this is some kind of interference coming from the USB3 port, which is known to cause issues with 2.4GHz wireless.

Neirop, 2025-02-03:
I already tried to plug it to an USB2, it doesn't change.
I didn't try with an extension cable, I can try later but I doubt it would change anything.

Neirop, 2025-02-03:
Well, I tried Raspberry Pi OS and got the same results, so it's not a software issue (at least not from OpenWRT).
I tested it with an extension cable and got better bandwidth.
So, in conclusion, I either have a faulty unit or there’s a major design flaw with the Raspberry Pi 5. I wouldn’t recommend this device if you plan to use a USB WiFi adapter