openwrt/openwrt #17837: AVM FritzBox 4040 USB power is turned off during booting

cnbr, 2025-02-02:
### Describe the bug

I tried to integrate an overlay/extroot filesystem on a USB stick, as described at https://openwrt.org/docs/guide-user/additional-software/extroot_configuration. 

The USB stick is recognized with its partitions, at  5.078149, but then disconnects a few seconds later at 7.870130, before the overlay file system can be integrated.

A usb power meter shows that the power supply from the USB port is switched on at starting booting, then, at the time of disconnection, power is switched off.
Approximately 50 seconds later power  will be switched on again at the end of the boot process and the USB stick will be recognized again, but it is to late to mount the overlay.

I also noticed that the PowerLED was switched off and no longer flashed. When the USB stick gets power again, the Power LED lights up again.

```
[    1.780686] Freeing unused kernel image (initmem) memory: 1024K
[    1.785000] Run /sbin/init as init process
[    2.697176] init: Console is alive
[    2.697537] init: - watchdog -
[    4.414108] kmodloader: loading kernel modules from /etc/modules-boot.d/*
[    4.476289] gpio_button_hotplug: loading out-of-tree module taints kernel.
[    4.486602] SCSI subsystem initialized
[    4.719051] xhci-hcd xhci-hcd.0.auto: xHCI Host Controller
[    4.719142] xhci-hcd xhci-hcd.0.auto: new USB bus registered, assigned bus number 1
[    4.723683] xhci-hcd xhci-hcd.0.auto: hcc params 0x0228f665 hci version 0x100 quirks 0x0000008002000010
[    4.731157] xhci-hcd xhci-hcd.0.auto: irq 59, io mem 0x08a00000
[    4.740640] xhci-hcd xhci-hcd.0.auto: xHCI Host Controller
[    4.746301] xhci-hcd xhci-hcd.0.auto: new USB bus registered, assigned bus number 2
[    4.751877] xhci-hcd xhci-hcd.0.auto: Host supports USB 3.0 SuperSpeed
[    4.760270] hub 1-0:1.0: USB hub found
[    4.766042] hub 1-0:1.0: 1 port detected
[    4.770285] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
[    4.774482] hub 2-0:1.0: USB hub found
[    4.785694] hub 2-0:1.0: 1 port detected
[    4.786314] xhci-hcd xhci-hcd.1.auto: xHCI Host Controller
[    4.789600] xhci-hcd xhci-hcd.1.auto: new USB bus registered, assigned bus number 3
[    4.795020] xhci-hcd xhci-hcd.1.auto: USB3 root hub has no ports
[    4.802400] xhci-hcd xhci-hcd.1.auto: hcc params 0x0220f665 hci version 0x100 quirks 0x0000008002000010
[    4.808715] xhci-hcd xhci-hcd.1.auto: irq 60, io mem 0x06000000
[    4.818808] hub 3-0:1.0: USB hub found
[    4.823678] hub 3-0:1.0: 1 port detected
[    4.840847] usbcore: registered new interface driver usb-storage
[    4.842731] usbcore: registered new interface driver uas
[    4.846292] kmodloader: done loading kernel modules from /etc/modules-boot.d/*
[    4.852709] init: - preinit -
[    4.998135] random: crng init done
[    5.078149] usb 1-1: new high-speed USB device number 2 using xhci-hcd
[    5.262704] scsi host0: uas
[    5.265188] scsi 0:0:0:0: Direct-Access      USB      SanDisk 3.2Gen1 1.00 PQ: 0 ANSI: 7
[    5.275135] sd 0:0:0:0: [sda] 120164352 512-byte logical blocks: (61.5 GB/57.3 GiB)
[    5.276209] sd 0:0:0:0: [sda] Write Protect is off
[    5.282465] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    5.286788] sd 0:0:0:0: [sda] Preferred minimum I/O size 4096 bytes
[    5.300146]  sda: sda1 sda2 sda3
[    5.302754] sd 0:0:0:0: [sda] Attached SCSI removable disk
Cannot parse config file '/etc/fw_env.config': No such file or directory
Failed to find NVMEM device
[    7.374071] ipqess-edma c080000.ethernet eth0: configuring for fixed/internal link mode
[    7.374883] qca8k-ipq4019 c000000.switch lan1: configuring for phy/psgmii link mode
[    7.380950] qca8k-ipq4019 c000000.switch: PSGMII calibration!
[    7.388634] ipqess-edma c080000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[    7.870130] usb 1-1: USB disconnect, device number 2
[    7.948104] sd 0:0:0:0: [sda] Synchronizing SCSI cache
[    8.568219] sd 0:0:0:0: [sda] Synchronize Cache(10) failed: Result: hostbyte=0x07 driverbyte=DRIVER_OK
Press the [f] key and hit [enter] to enter failsafe mode
Press the [1], [2], [3] or [4] key and hit [enter] to select the debug level
[   11.608468] qca8k-ipq4019 c000000.switch lan1: Link is Up - 1Gbps/Full - flow control rx/tx
```
Many seconds later, USB power is switched on
```
[   64.758176] usb 1-1: new high-speed USB device number 3 using xhci-hcd
[   64.943308] scsi host0: uas
[   64.946162] scsi 0:0:0:0: Direct-Access      USB      SanDisk 3.2Gen1 1.00 PQ: 0 ANSI: 7
[   64.956942] sd 0:0:0:0: [sda] 120164352 512-byte logical blocks: (61.5 GB/57.3 GiB)
[   64.958050] sd 0:0:0:0: [sda] Write Protect is off
[   64.964154] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[   64.968710] sd 0:0:0:0: [sda] Preferred minimum I/O size 4096 bytes
[   64.981907]  sda: sda1 sda2 sda3
[   64.984415] sd 0:0:0:0: [sda] Attached SCSI removable disk
[   67.129213] qca8k-ipq4019 c000000.switch lan1: Link is Up - 1Gbps/Full - flow control rx/tx
```

Bootlog on the serial console is attached

[bootlog_fritzbox4040.log.txt](https://github.com/user-attachments/files/18633402/bootlog_fritzbox4040.log.txt)

### OpenWrt version

r28417-daef29c75d

### OpenWrt release

 24.10.0-rc7  

### OpenWrt target/subtarget

 ipq40xx/generic

### Device

 AVM FRITZ!Box 4040

### Image kind

Official downloaded image

### Steps to reproduce

Boot the device with an USB-Stick or connect USB Power Meter, watch to the PowerLED during booting.
Watch the logs or dmesg, and see, when the USB-Stick ist connected or disconneced

### Actual behaviour

Power of the USB-Port is switch on at starting boot, than switched off during the  boot  procress and at the end of the booting procress it ist switch on.

### Expected behaviour

Power of the USB-Port is allways on during booting 

### Additional info

During testing I found that the problem also exists with 23.05.5. 
With the  22.03.7 the problem does not exist, here the power supply to the USB port always remains on during the boot process.

All image where created with the OpenWrt firmware-selector and adding the kmod-usb-storage-uas block-mount kmod-fs-ext4 e2fsprogs kmod-usb-storage modules

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

aimacintyre, 2025-02-03:
This might be a side effect of the DSA conversion of the ipq40xx target between 22.03 and 23.05 - in the FB4040 DTS the USB power control is tied to `ethphy1` (`lan2` switch port) and this hasn't changed since 22.03, so I surmise that the DSA switch initialisation is triggering the ~50 second poweroff you see.

robimarko, 2025-02-03:
DSA cannot influence this as USB power is controlled by the QCA807x PHY pin and it has its own completely separate driver.

aimacintyre, 2025-02-05:
> DSA cannot influence this as USB power is controlled by the QCA807x PHY pin and it has its own completely separate driver.

Sorry, my terminology was poor.  My intention was to highlight that there appeared to be a behaviour difference coincident with the swconfig to DSA transition possibly as a consequence of some change in the QCA807x initialisation.  There don't appear to be any real differences in the DTS or the QCA807x GPIO code across this transition that I can find (other than for the `ess` to `edma` network driver change) but I don't understand the Linux driver framework enough to trace any related code that might have changed.  Even though the device power LED is driven by a separate GPIO pin (pin 1 on `ethphy2`, USB power enable is GPIO pin 1 on `ethphy1`), it's odd that it appears to be affected as well - almost as though the power LED is gated by the USB power enable output.

mschomm, 2025-05-11:
Same problem here with a fresh install of 24.10.1 on a FritzBox 4040.

neheb, 2025-06-08:
there's a GPIO hog for this. This should probably be converted to a regulator and tied in to usb3_ss_phy