openwrt/openwrt #17838: Asus TUF Gaming AX6000: configure the LAN port LEDs

andrey-le, 2025-02-02:
### Describe the bug

LUCI-LED Configuration

### OpenWrt version

r28417-daef29c75d

### OpenWrt release

24.10.0-rc7

### OpenWrt target/subtarget

mediatek/filogic

### Device

ASUS TUF-AX6000

### Image kind

Official downloaded image

### Actual behaviour

On  24.10-rc7 version I succeded only with turning off system power LED and WLAN LED.
TUF AX4200 landing page says: the LAN port LEDs are driven by the switch, but OpenWrt does not correctly configure the output.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

aimacintyre, 2025-02-03:
If the LAN port LEDs are driven directly by the switch, it is unlikely they can be controlled by OpenWrt in the normal way.  If they can be controlled at all (e.g. by manipulating the switch configuration), someone with the need/desire to do so will have to figure out how and then implement a solution.

andrey-le, 2025-02-03:
Seems like a big new feature, issue is closed

maurerle, 2025-02-04:
@andrey-le - as the opener of the issue, you can close it yourself :)

closed 2025-02-04 by andrey-le

abajk, 2025-02-09:
I don't think it requires much change. The LEDs are probably connected to the PHY. The [driver](https://github.com/openwrt/openwrt/commit/5fd518b6e3d5447acb862cc4119425316e22a1ba) supports LED control. They need to be described in the dts.

abajk, 2025-02-09:
It needs to be done by someone with access to the hardware.

abajk, 2025-02-09:
Example https://github.com/openwrt/openwrt/commit/85c41253d927ae6a31845d888780a4462efe61ef