openwrt/openwrt #17841: docker: failed to register layer: lsetxattr security.capability operation not supported.

uysalerinc, 2025-02-03:
### Describe the bug

I am trying to pull a docker image that uses newgidmap an i got the docker: failed to register layer: lsetxattr security.capability /usr/bin/newgidmap: operation not supported. error.

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

openwrt-24.10

### OpenWrt target/subtarget

bcm27xx/bcm2711

### Device

Raspberry Pi Compute Module 4 Rev 1.1

### Image kind

Self-built image

### Steps to reproduce

git pull blah/blah

### Actual behaviour

0ed94f65e4fb: Extracting  37.65MB/37.65MB
6230635da2f6: Download complete
99f2c4d28ecf: Download complete
beeb2daa7322: Download complete
b863d97ebbfc: Download complete
82c6367beae6: Download complete
c1f2efe029d7: Download complete
b303c082fc76: Download complete
861a45fdafed: Download complete
fcafb0dc0e11: Download complete
docker: failed to register layer: lsetxattr security.capability /usr/bin/newgidmap: operation not supported.
See 'docker run --help'.

### Additional info

bcm2711/config-6.6

CONFIG_EXT4_FS=y
CONFIG_EXT4_FS_POSIX_ACL=y
CONFIG_EXT4_FS_SECURITY=y
CONFIG_FS_POSIX_ACL=y

.config

#
# Storage
#
# CONFIG_DOCKER_STO_DEVMAPPER is not set
CONFIG_DOCKER_STO_EXT4=y
CONFIG_DOCKER_STO_BTRFS=y
# end of Storage

# Filesystem ACL and attr support options
#
CONFIG_USE_FS_ACL_ATTR=y
CONFIG_KERNEL_FS_POSIX_ACL=y
CONFIG_KERNEL_BTRFS_FS_POSIX_ACL=y
CONFIG_KERNEL_EXT4_FS_POSIX_ACL=y

CONFIG_KERNEL_SQUASHFS_FRAGMENT_CACHE_SIZE=3
[... 540 lines cut ...]
CONFIG_VERSION_PRODUCT=""
CONFIG_VERSION_REPO="https://downloads.openwrt.org/releases/23.05.2"
CONFIG_VERSION_SUPPORT_URL=""
# CONFIG_WPA_MBO_SUPPORT is not set
# CONFIG_PACKAGE_kmod-crypto-kpp is not set
CONFIG_PACKAGE_kmod-hid=y
CONFIG_PACKAGE_kmod-hid-generic=y
CONFIG_PACKAGE_kmod-input-evdev=y
CONFIG_PACKAGE_kmod-lib-crc-ccitt=y
# CONFIG_PACKAGE_kmod-mppe is not set
CONFIG_PACKAGE_kmod-ppp=y
# CONFIG_PACKAGE_kmod-ppp-synctty is not set
# CONFIG_PACKAGE_kmod-pppoa is not set
CONFIG_PACKAGE_kmod-pppoe=y
# CONFIG_PACKAGE_kmod-pppol2tp is not set
CONFIG_PACKAGE_kmod-pppox=y
# CONFIG_PACKAGE_kmod-pptp is not set
CONFIG_PACKAGE_kmod-slhc=y
# CONFIG_VERSION_CODE_FILENAMES is not set
# CONFIG_VERSION_FILENAMES is not set
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

uysalerinc, 2025-02-03:
its 24.10.0-rc7

hauke, 2025-02-03:
Activate these options and then it should work:
```
CONFIG_KERNEL_CIFS_ACL=y
CONFIG_KERNEL_F2FS_FS_POSIX_ACL=y
CONFIG_KERNEL_F2FS_FS_SECURITY=y
CONFIG_KERNEL_HFSPLUS_FS_POSIX_ACL=y
CONFIG_KERNEL_HFS_FS_POSIX_ACL=y
CONFIG_KERNEL_IO_URING=y
CONFIG_KERNEL_JFFS2_FS_POSIX_ACL=y
CONFIG_KERNEL_JFFS2_FS_SECURITY=y
CONFIG_KERNEL_JFS_POSIX_ACL=y
CONFIG_KERNEL_NET_L3_MASTER_DEV=y
CONFIG_KERNEL_NFS_ACL_SUPPORT=y
CONFIG_KERNEL_REISER_FS_POSIX_ACL=y
CONFIG_KERNEL_TMPFS_POSIX_ACL=y
CONFIG_KERNEL_UBIFS_FS_SECURITY=y
CONFIG_KERNEL_XFS_POSIX_ACL=y
```

closed 2025-02-03 by hauke