openwrt/openwrt #17846: Aerohive BR200-WP bootloops on 24.10.0-rc7

konus-owt, 2025-02-03:
### Describe the bug

Upgrading from 23.05.5 to 24.10.0-rc7 results in a bootloop. It starts loading the kernel but then crashes when it can't load the rootfs.

`
U-Boot 2009.11 (Dec 27 2017 - 21:56:12), Build: jenkins-SDW-Havana_BR200_Rel-33

CPU:   P1011E, Version: 1.1, (0x80ed0011)
Core:  E500, Version: 5.1, (0x80212051)
Clock Configuration:
       CPU0:660  MHz, 
       CCB:330  MHz,
       DDR:330  MHz (660 MT/s data rate) (Asynchronous), LBC:20.625 MHz
L1:    D-cache 32 kB enabled
       I-cache 32 kB enabled
I2C:   ready
SPI:   ready
DRAM:  Configuring DDR for 660 MT/s data rate
DDR: 256 MB
FLASH: 64 MB
L2:    256 KB enabled
MMC:  

    PCIE2: disabled

    PCIE1 connected to Slot 2 as Root Complex (base addr ffe0a000)
               Scanning PCI bus 01
        01  00  168c  0030  0280  ff
    PCIE1 on bus 00 - 01

In:    serial
Out:   serial
Err:   serial
Net:   initialization for Atheros AR8327/AR8328,addr 4
AR8327/AR8328 v1.2 is found!
set for PHY4
initialization for Atheros AR8327/AR8328,addr 0
eth0, eth1
Hit any key to stop autoboot:  0 
Hit any key to interrupt boot from flash:  0 
## Booting kernel from Legacy Image at ee840000 ...
   Image Name:   POWERPC OpenWrt Linux-6.6.73
   Created:      2025-01-27  22:53:29 UTC
   Image Type:   PowerPC Linux Kernel Image (uncompressed)
   Data Size:    3447228 Bytes =  3.3 MB
   Load Address: 01500000
   Entry Point:  01500000
   Verifying Checksum ... OK
## Loading init Ramdisk from Legacy Image at 02000000 ...
   Image Name:   OpenWrt fake ramdisk
   Created:      2025-01-27  22:53:29 UTC
   Image Type:   PowerPC Linux RAMDisk Image (uncompressed)
   Data Size:    0 Bytes =  0 kB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at ec000000
   Booting using the fdt blob at 0xec000000
   Loading Kernel Image ... OK
OK
   Loading Ramdisk to 10000000, end 10000000 ... OK
   Loading Device Tree to 00ff9000, end 00fff11d ... OK
ft_fixup_l2cache: FDT_ERR_NOTFOUND

zImage starting: loaded at 0x01500000 (sp: 0x0184afa0)
Allocating 0x1118e7c bytes for kernel...
Decompressing (0x00000000 <- 0x01515000:0x018499bc)...
Done! Decompressed 0x10d9b30 bytes

Linux/PowerPC load: console=ttyS0,9600
[... 149 lines cut ...]
### Image kind

Official downloaded image

### Steps to reproduce

sysupgrade from 23.05.5 to 24.10.0-rc7

### Actual behaviour

sysupgrade completes but device can't find rootfs and ends in a bootloop

### Expected behaviour

boot correctly

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

konus-owt, 2025-02-04:
The 24.10.0-rc7 initramfs file can be tftp booted without issues. When I try to sysupgrade it immediately crashes and reboots the router. 

Full log included below:

`

U-Boot 2009.11 (Dec 27 2017 - 21:56:12), Build: jenkins-SDW-Havana_BR200_Rel-33

CPU:   P1011E, Version: 1.1, (0x80ed0011)
Core:  E500, Version: 5.1, (0x80212051)
Clock Configuration:
       CPU0:660  MHz, 
       CCB:330  MHz,
       DDR:330  MHz (660 MT/s data rate) (Asynchronous), LBC:20.625 MHz
L1:    D-cache 32 kB enabled
       I-cache 32 kB enabled
I2C:   ready
SPI:   ready
DRAM:  Configuring DDR for 660 MT/s data rate
DDR: 256 MB
FLASH: 64 MB
L2:    256 KB enabled
MMC:  

    PCIE2: disabled

    PCIE1 connected to Slot 2 as Root Complex (base addr ffe0a000)
               Scanning PCI bus 01
        01  00  168c  0030  0280  ff
    PCIE1 on bus 00 - 01

In:    serial
Out:   serial
Err:   serial
Net:   initialization for Atheros AR8327/AR8328,addr 4
AR8327/AR8328 v1.2 is found!
set for PHY4
initialization for Atheros AR8327/AR8328,addr 0
eth0, eth1
Hit any key to stop autoboot:  3  0 
Password:

=> dhcp; setenv serverip 192.168.1.254; tftpboot 0x2004000 openwrt-rc7.bin; bootm 0x2004000
Check for phy link status
Speed: 1000, full duplex
BOOTP broadcast 1
*** Unhandled DHCP Option in OFFER/ACK: 28
*** Unhandled DHCP Option in OFFER/ACK: 28
DHCP client bound to address 192.168.1.164
Check for phy link status
Speed: 1000, full duplex
Using eth0 device
TFTP from server 192.168.1.254; our IP address is 192.168.1.164
Filename 'openwrt-rc7.bin'.
Load address: 0x2004000
Loading: *start tftp get.#################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
[... 274 lines cut ...]
SPI:   ready
DRAM:  Configuring DDR for 660 MT/s data rate
DDR: 256 MB
FLASH: 64 MB
L2:    256 KB enabled
MMC:  

    PCIE2: disabled

    PCIE1 connected to Slot 2 as Root Complex (base addr ffe0a000)
               Scanning PCI bus 01
        01  00  168c  0030  0280  ff
    PCIE1 on bus 00 - 01

In:    serial
Out:   serial
Err:   serial
Net:   initialization for Atheros AR8327/AR8328,addr 4
AR8327/AR8328 v1.2 is found!
set for PHY4
initialization for Atheros AR8327/AR8328,addr 0
eth0, eth1
Hit any key to stop autoboot:  3  2 

`

konus-owt, 2025-02-04:
Someone on the forum with a Sophos Red 15 also has bootloop issues. That device has the same cpu / target. See the link below:

[https://forum.openwrt.org/t/howto-installing-openwrt-on-sophos-red-15w/177928/25?u=konus](https://forum.openwrt.org/t/howto-installing-openwrt-on-sophos-red-15w/177928/25?u=konus)

konus-owt, 2025-02-25:
Someone with a Watchguard Firebox T10 is also having problems upgrading to 24.10.0. Looks like there is something wrong with the partitions / storage for the whole mpc85xx/p1010 target.

[https://forum.openwrt.org/t/watchguard-t10-upgrade-failure-while-updating-to-24-10-partitioning-issue/225904?u=konus](url)

konus-owt, 2025-09-07:
I tried upgrading from 23.05 to today's snapshot and the issue is still there. It starts booting but can't read the rootfs and then crashes and bootloops

`## Booting kernel from Legacy Image at ee840000 ...
   Image Name:   POWERPC OpenWrt Linux-6.6.104
   Created:      2025-09-07   9:45:08 UTC
   Image Type:   PowerPC Linux Kernel Image (uncompressed)
   Data Size:    3667336 Bytes =  3.5 MB
   Load Address: 01500000
   Entry Point:  01500000
   Verifying Checksum ... OK
## Loading init Ramdisk from Legacy Image at 02000000 ...
   Image Name:   OpenWrt fake ramdisk
   Created:      2025-09-07   9:45:08 UTC
   Image Type:   PowerPC Linux RAMDisk Image (uncompressed)
   Data Size:    0 Bytes =  0 kB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at ec000000
   Booting using the fdt blob at 0xec000000
   Loading Kernel Image ... OK
OK
   Loading Ramdisk to 10000000, end 10000000 ... OK
   Loading Device Tree to 00ff9000, end 00fff11d ... OK
ft_fixup_l2cache: FDT_ERR_NOTFOUND

zImage starting: loaded at 0x01500000 (sp: 0x01880fa0)
Allocating 0x1118e84 bytes for kernel...
Decompressing (0x00000000 <- 0x01515000:0x0187f588)...
Done! Decompressed 0x10d9a94 bytes

Linux/PowerPC load: console=ttyS0,9600
Finalizing device tree... flat tree at 0x1881cc0
[    0.000000] Memory CAM mapping: 16/16/16/16/64/64/64 Mb, residual: 0Mb
[    0.000000] Activating Kernel Userspace Access Protection
[    0.000000] Activating Kernel Userspace Execution Prevention
[    0.000000] Linux version 6.6.104 (builder@buildhost) (powerpc-openwrt-linux-musl-gcc (OpenWrt GCC 14.3.0 r30972-0ee0846be1) 14.3.0, GNU ld (GNU Binutils) 2.44) #0 Sun Sep  7 09:45:08 2025
[    0.000000] Hardware name: Aerohive BR200-WP e500v2 0x80212051 P1020 RDB
[    0.000000] printk: bootconsole [udbg0] enabled
[    0.000000] -----------------------------------------------------
[    0.000000] phys_mem_size     = 0x10000000
[    0.000000] dcache_bsize      = 0x20
[    0.000000] icache_bsize      = 0x20
[    0.000000] cpu_features      = 0x0000000010010108
[    0.000000]   possible        = 0x0000000010010108
[    0.000000]   always          = 0x0000000010010108
[    0.000000] cpu_user_features = 0x84e08000 0x08000000
[    0.000000] mmu_features      = 0x00020210
[    0.000000] -----------------------------------------------------
br200_wp_setup_arch()
[    0.000000] br200-wp board from Aerohive
[    0.000000] barrier-nospec: using isync; sync as speculation barrier
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] MMU: Allocated 1088 bytes of context maps for 255 contexts
[    0.000000] Kernel command line: console=ttyS0,9600 noinitrd
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Kernel virtual memory layout:
[    0.000000]   * 0xffbdf000..0xfffff000  : fixmap
[    0.000000]   * 0xd1000000..0xffbdf000  : vmalloc & ioremap
[    0.000000] Memory: 242012K/262144K available (8280K kernel code, 672K rwdata, 8104K rodata, 200K init, 251K bss, 20132K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 512, nr_irqs: 512, preallocated irqs: 16
[... 70 lines cut ...]
[    5.142706] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[    5.265571] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 251)
[    5.353843] pcieport 0001:00:00.0: enabling device (0106 -> 0107)
[    5.426380] Serial: 8250/16550 driver, 16 ports, IRQ sharing enabled
[    5.505096] printk: console [ttyS0] disabled
����ʚ���ɥ�ѭ��ͽ����喅��������ʚ���ɥ�ѭ��ͽ����喅��������ʒ���ɥ�ѭ��э��ͽ��Ց���ͅ��������ʒ���ɥ�ѭ��э��ͽ��Ց���ͅ����������ɥ������Ң��Ѫ����������ꢒ�͕��Ցꒂ������͊�������²����ɥ�ѭ��ͽ������ͅ����ᲂ�ʪ�ɥ�ѭ��ͽ����喅����ᲂʂ�����������ɥ�����Ѫ����������ꢒ�͕��Ցꒂ������͊����Გ���ͱ������������������Ლ�ʲ��ͱ�����ɚ����������ѡ�ɹ�щ����ͥ����������ɕ�͂���ʚ��Ҫ�Ტ�����ͱ�����ɚ����������ѡ�ɹ�щ���Җ������Ʌх��ɭ�ɽչ�ͱ��͂�Ც�����ͱ�����ɚ����������ѡ�ɹ�щ���ѡ҅�����ʚ��Ҫ���������ͱ�����ɚ����������ѡ�ɹ�щ���ѡ�չ������ѡ������Ჺ��²�ͱ�����ɚ����������ѡ�ɹ�щ���ѡ҅���������ɵ����������ͱ�����ɚ����������ѡ�ɹ�щ���ѡ҅���������ɵ��������ª��ͱ�����ɚ����������ѡ�ɹ�щ���ѡ҅���������ɵ�������ʢ���ͱ�����ɚ����������ѡ�ɹ�щ���ѡ҅���������ɵ�������ʂ��ʪ����ѡ�ɥٕɨ������ѡ�ɹ�Ѩ��ɽ����Ẋ�������������������ъ��������Ḻ����������������������ъ��������Ẓ����뭕ݑт�ݕ����뭺�э����ɥٕ�녑������ʂ��ɕ�͍������ɥٕɒ�����х٥�х���݅ɕ��������ͱ������ɽ��ꨪ�͍ͥ�ٕɕ�Ảʪ����ҥ٥���ѕ٥������ձ���ᕑ�����ɽ��������ᱦ��ɔ����ᱦ���ʩ���������ᕑ����եɕ��Ŋ����ɡ���¢�ª�ᕑ���ɒ����ѕɕ�Ẻ��ª�ҥ٥���ѕ٥������ձ���ᕑ�����ɽ��������������ɔ����ႍ������ѽʩa�ºʪ����ᕑ����եɕ��Ŋ����ɡ�ʚ�ʲ���ᕑ���ɒ����ѕɕ����ʂ��¥����ѕɕ��ʕ�ɽѽ�����������������Ѫ�ѥ����ѡ�ʲ�ʢʫ����꺥ѡ�����������ѕɕ�Ԩ�ɽѽ��������a�����ɥ�����ѕɥ������������������˝��م������啙�ձ����ѕʽ�ɚ�ɥ��͢�녑ɹ�љ��ѕəʽՕ�����ͩ��ʊ������������Ѳ¸¢�ʒʢɵ���幅������������Ʌѥ��������չ��ªʪº��������������Ҋ�Ҭ˙���ɥ���ɥᕑ���������둕�²²��������������Ҋ�ҥ��������ձ���ݬ��ɽ�����¢�����������������Ҋ������ [... cut ...]
[   10.576482] Call Trace:
[   10.605656] [c202fdb0] [c07e2d3c] dump_stack_lvl+0x30/0x4c (unreliable)
[   10.684858] [c202fdd0] [c002f754] panic+0x124/0x2dc
[   10.743219] [c202fe30] [c1001878] mount_root_generic+0x1e0/0x29c
[   10.815105] [c202fea0] [c1001d78] prepare_namespace+0x214/0x260
[   10.885950] [c202fee0] [c00032b8] kernel_init+0x20/0x134
[   10.949505] [c202ff00] [c000f224] ret_from_kernel_user_thread+0x10/0x18
[   11.028689] --- interrupt: 0 at 0x0
[   11.070369] Rebooting in 1 seconds..

`

CHKDSK88, 2025-10-16:
Fix:  #20419

closed 2025-10-16 by konus-owt