openwrt/openwrt #17854: RT3593 detected as 0x0000 and fails to allocate device

nariox, 2025-02-05:
### Describe the bug

```
[   28.864762] ieee80211 phy0: rt2x00_set_rt: Info - RT chipset 3593, rev 0402 detected
[   28.880340] ieee80211 phy0: rt2800_init_eeprom: Error - Invalid RF chipset 0x0000 detected
[   28.896850] ieee80211 phy0: rt2x00lib_probe_dev: Error - Failed to allocate device
[   29.033755] ieee80211 phy1: rt2x00_set_rt: Info - RT chipset 3883, rev 0400 detected
[   29.049514] ieee80211 phy1: rt2x00_set_rf: Info - RF chipset 3853 detected
[   29.065710] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
```

### OpenWrt version

r28431-92e020b50f

### OpenWrt release

24.10-SNAPSHOT

### OpenWrt target/subtarget

ramips/rt3883

### Device

TRENDnet TEW-692GR

### Image kind

Self-built image

### Steps to reproduce

Happens during boot

### Actual behaviour

During boot, the 5GHz radio does not get initialized correctly

### Expected behaviour

RF chipset is detected correctly, then proceeds to initialize

### Additional info

This image is self-built, but without any patches and using the config.buildinfo from 24.10-rc7

### Diffconfig

```text
CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_rt3883=y
CONFIG_TARGET_ramips_rt3883_DEVICE_trendnet_tew-692gr=y
CONFIG_ALL_KMODS=y
CONFIG_ALL_NONSHARED=y
CONFIG_DEVEL=y
CONFIG_AUTOREMOVE=y
CONFIG_BPF_TOOLCHAIN_BUILD_LLVM=y
# CONFIG_BPF_TOOLCHAIN_NONE is not set
CONFIG_BUILDBOT=y
CONFIG_COLLECT_KERNEL_DEBUG=y
CONFIG_HAS_BPF_TOOLCHAIN=y
CONFIG_IB=y
CONFIG_IMAGEOPT=y
CONFIG_JSON_CYCLONEDX_SBOM=y
CONFIG_KERNEL_BUILD_DOMAIN="buildhost"
CONFIG_KERNEL_BUILD_USER="builder"
CONFIG_KERNEL_DEBUG_INFO=y
CONFIG_KERNEL_DEBUG_INFO_REDUCED=y
CONFIG_KERNEL_DEBUG_KERNEL=y
CONFIG_MAKE_TOOLCHAIN=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_libbpf=m
CONFIG_PACKAGE_libelf=m
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_libpcap=m
CONFIG_PACKAGE_libxdp=m
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-package-manager=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-ssl=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_px5g-mbedtls=y
CONFIG_PACKAGE_qosify=m
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_tc-tiny=m
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_xdp-filter=m
CONFIG_PACKAGE_xdp-loader=m
CONFIG_PACKAGE_xdpdump=m
CONFIG_PACKAGE_zlib=m
# CONFIG_PKG_ASLR_PIE_NONE is not set
CONFIG_PKG_ASLR_PIE_REGULAR=y
CONFIG_REPRODUCIBLE_DEBUG_INFO=y
CONFIG_SDK=y
CONFIG_SDK_LLVM_BPF=y
CONFIG_USE_LLVM_BUILD=y
CONFIG_VERSIONOPT=y
CONFIG_VERSION_BUG_URL=""
CONFIG_VERSION_CODE=""
CONFIG_VERSION_CODE_FILENAMES=y
CONFIG_VERSION_DIST="OpenWrt"
CONFIG_VERSION_FILENAMES=y
CONFIG_VERSION_HOME_URL=""
CONFIG_VERSION_HWREV=""
CONFIG_VERSION_MANUFACTURER=""
CONFIG_VERSION_MANUFACTURER_URL=""
CONFIG_VERSION_NUMBER=""
CONFIG_VERSION_PRODUCT=""
CONFIG_VERSION_REPO="https://downloads.openwrt.org/releases/24.10.0-rc7"
CONFIG_VERSION_SUPPORT_URL=""
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

nariox, 2025-02-05:
Confirming the same issue on the official 24.10.0. Here's the output observed on 23.05.5:
```
[   21.142719] ieee80211 phy0: rt2x00_set_rt: Info - RT chipset 3593, rev 0402 detected
[   21.161736] ieee80211 phy0: rt2x00_set_rf: Info - RF chipset 000d detected
[   21.483820] ieee80211 phy1: rt2x00_set_rt: Info - RT chipset 3883, rev 0400 detected
[   21.499424] ieee80211 phy1: rt2x00_set_rf: Info - RF chipset 3853 detected
```

Seems like the error is happening either [here](https://github.com/torvalds/linux/blob/5c8c229261f14159b54b9a32f12e5fa89d88b905/drivers/net/wireless/ralink/rt2x00/rt2800lib.c#L11147), where the EEPROM is read or [here](https://github.com/torvalds/linux/blob/5c8c229261f14159b54b9a32f12e5fa89d88b905/drivers/net/wireless/ralink/rt2x00/rt2800lib.c#L11168) where the u16 is parsed, my guess being on the first link, but from my understanding (not having dug too deep), this should be a simple read from memory and nothing should have fundamentally changed since 23.05.

hauke, 2025-02-05:
This looks very similar to #17818.

I added this revert patch: 6df0e3d02ac89d3ad7da1ade7fc51b15163bd136, but I am not sure if it fixes it. You tested it on a more recent version. 

Could you please try 24.10.0-rc6 please.

DragonBluep, 2025-02-06:
Please try this patch:
```
--- a/package/kernel/mac80211/patches/rt2x00/602-01-wifi-rt2x00-Add-support-for-loading-EEPROM-from-user.patch
+++ b/package/kernel/mac80211/patches/rt2x00/602-01-wifi-rt2x00-Add-support-for-loading-EEPROM-from-user.patch
@@ -151,7 +151,7 @@ Signed-off-by: Christian Marangi <ansuelsmth@gmail.com>
 +	}
 +
 +	if (!ee_name)
-+		return 0;
++		return -ENOENT;
 +
 +	rt2x00_info(rt2x00dev, "Loading EEPROM data from '%s'.\n", ee_name);
 +
```

nariox, 2025-02-06:
> This looks very similar to [#17818](https://github.com/openwrt/openwrt/issues/17818).
> 
> I added this revert patch: [6df0e3d](https://github.com/openwrt/openwrt/commit/6df0e3d02ac89d3ad7da1ade7fc51b15163bd136), but I am not sure if it fixes it. You tested it on a more recent version.
> 
> Could you please try 24.10.0-rc6 please.

This does look very similar. Unfortunately, I cannot test it on those, because of this bug: https://github.com/openwrt/openwrt/issues/17677
But thank you for pointing this to me.

> Please try this patch:
>
> ```
> --- a/package/kernel/mac80211/patches/rt2x00/602-01-wifi-rt2x00-Add-support-for-loading-EEPROM-from-user.patch
> +++ [...]
This one solves my issue, wifi config now produces the correct config and I can use it to connect. Thank you

closed 2025-02-12 by openwrt-bot

nariox, 2025-02-13:
Thank you very much, @DragonBluep and all others involved!