openwrt/openwrt #17855: In kernel 6.6 (used in version 24.10), PPPoE configurations fail to work properly with Hysteria 2 (hy2).

NogKCcn, 2025-02-05:
### Describe the bug

Under kernel version 6.6, many applications such as OpenClash and Mihomo fail to properly utilize the hy2 proxy protocol. The symptoms vary depending on hy2 nodes: inability to ping (while working normally with mobile Meta or other apps), successful pinging without internet connectivity, or intermittent connection with high latency and frequent packet loss.

These issues are resolved when reverting to version 23.05 with identical configurations. Notably, these problems appear to occur exclusively under PPPoE network conditions.

### OpenWrt version

24.10-rc7

### OpenWrt release

24.10-rc7

### OpenWrt target/subtarget

/

### Device

r3s nanopi

### Image kind

Official downloaded image

### Steps to reproduce

When testing the official OpenWrt source and ImmortalWrt's official source with kernel 6.6, network connections almost completely failed under the same settings. However, switching to kernel 6.1 worked normally. The network environment uses an internal IP behind NAT (192.168.x.x).  
(PS: Kernel versions before 6.6.35 appeared normal, but versions from 6.6.47 onward became unusable.)

For OpenWrt's official snapshot x64 builds:  
- When configured with an internal IP behind NAT (192.168.x.x), it works normally.  
- However, using the same image with PPPoE network configuration results in complete connection failure (tested on kernel 6.6.58).  
- Reverting to OpenWrt 22.03.7 or 23.05.5 (kernels 5.10 and 5.15 respectively) with nearly identical configurations works normally.

### Expected behaviour

The 6.6 version of the kernel can use hy2 normally.

### Additional info

https://github.com/apernet/hysteria/issues/1233
There is someone people find this bug too.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

wherss1, 2025-02-05:
我也遇到同样的问题，我的设备是x86  openwrt官方或 imm24.10版本软路由拨号OpenClash MihomoTProxy passwall(节点使用类型选Hysteria2）使用Hysteria2节点，科学不通的情况

NogKCcn, 2025-02-05:
> 报告的目标/子目标无效。这是来自受支持的设备吗？`/`

是的

NogKCcn, 2025-02-05:
> Invalid Release reported. `24.10-rc7` Is this from a clean repository?

yes

LGA1150, 2025-02-05:
I suspect the following commits related to UDP GRO/GSO are the cause:
openwrt/openwrt@79fce424046fed521026df7c7f838441fb46add8
openwrt/openwrt@17f07040e7808a0c9cba2afc0e0b22e5589da133

Try disabling GRO on all interfaces and see if it works.
To disable GRO on an interface, run `ethtool -K ethX gro off`. "ethX" should be the physical interface name.

ping @nbd168

wherss1, 2025-02-05:
禁用了所有接口GRO，还是无法与 Hysteria 2 （hy2） 一起正常工作。

LGA1150, 2025-02-05:
> 禁用了所有接口GRO，还是无法与 Hysteria 2 （hy2） 一起正常工作。

Try disabling GSO too

wherss1, 2025-02-05:
禁用了eth1 0接口GRO GRO,还是一样无法正常

wherss1, 2025-02-05:
我现在只开了2个接口

NogKCcn, 2025-02-05:
> 我现在只开了2个接口

我试了很多方式无法解决

wherss1, 2025-02-05:
我也是，用了我所有想得到的办法，ai都问了很多遍，没有一个能行的

NogKCcn, 2025-02-06:
> > 禁用了所有接口GRO，还是无法与 Hysteria 2 （hy2） 一起正常工作。
> 
> Try disabling GSO too

It's not work too

NogKCcn, 2025-02-06:
> 我也是，用了我所有想得到的办法，ai都问了很多遍，没有一个能行的

我找到问题所在了，是新版clash内核和fw4兼容的问题，降低内核版本即可，等作者修复

wherss1, 2025-02-07:
> > 我也是，用了我所有想得到的办法，ai都问了很多遍，没有一个能行的
> 
> 我找到问题所在了，是新版clash内核和fw4兼容的问题，降低内核版本即可，等作者修复

那passwa ll无法直接用hy2又是怎么回事，必须切换sing-box才可用

NogKCcn, 2025-02-07:
> > > 我也是，用了我所有想得到的办法，ai都问了很多遍，没有一个能行的
> > 
> > 我找到问题所在了，是新版clash内核和fw4兼容的问题，降低内核版本即可，等作者修复
> 
> 那passwa ll无法直接用hy2又是怎么回事，必须切换sing-box才可用

passwall调用的内核不一样啊，总之就是mihomo新版本内核和fw4兼容不好，降低版本一切正常

NogKCcn, 2025-02-09:
> > 禁用了所有接口GRO，还是无法与 Hysteria 2 （hy2） 一起正常工作。
> 
> Try disabling GSO too

I made a mistake, but now closing gso has indeed resolved the issue, is this due to a firmware problem?

LGA1150, 2025-02-11:
> I made a mistake, but now closing gso has indeed resolved the issue, is this due to a firmware problem?

Try disabling GSO but keep GRO on, is the issue still present?

NogKCcn, 2025-02-11:
> > I made a mistake, but now closing gso has indeed resolved the issue, is this due to a firmware problem?
> 
> Try disabling GSO but keep GRO on, is the issue still present?

experimental:
  quic-go-disable-gso: true
I just put this code in the yaml configuration file and solved the problem by just shutting down the gso.

wherss1, 2025-02-15:
我希望的目的是，像23.05一样，hy2节点导入就可以用，您现在找到方法了吗？？

LGA1150, 2025-02-17:
@wherss1 
Try reverting the 2 commits mentioned above (delete the 4 added patches) and rebuild OpenWrt.

LGA1150, 2025-02-17:
@NogKCcn 
Please reopen the issue if there is indeed a bug in linux UDP GSO. Disabling GSO is only a workaround, not a fix.

wherss1, 2025-02-17:
> 尝试还原上述 2 个提交（删除添加的 4 个补丁）并重建 OpenWrt。
请问是那4个补丁

NogKCcn, 2025-02-20:
> [@NogKCcn](https://github.com/NogKCcn) Please reopen the issue if there is indeed a bug in linux UDP GSO. Disabling GSO is only a workaround, not a fix.

Ture,but it not a only way.I find many workaround.
1.Put this code in yaml
experimental:
quic-go-disable-gso: true
2.The core do not update quic-go to 0.42.0 or high
3.Do not use fw4,use fw3

NogKCcn, 2025-02-22:
> 我希望的目的是，像23.05一样，hy2节点导入就可以用，您现在找到方法了吗？？

openclash已经添加了禁用gso选项，打开就行了[image]

Kryo123456, 2025-03-12:
> > 我希望的目的是，像23.05一样，hy2节点导入就可以用，您现在找到方法了吗？？
> 
> openclash已经添加了禁用gso选项，打开就行了[image]

打开这个选项是不是会影响性能啊？

LGA1150, 2026-01-21:
> 这个bug 现在还有吗？还需要禁用gso 吗？

Try 25.12-rc

LGA1150, 2026-10-19:
With LLM, I'm able to locate the cause.

Commit 79fce424046fed521026df7c7f838441fb46add8 backported torvalds/linux@10154dbded6d6a2fecaebdfda206609de0f121a9 from 6.11. However, that commit requires another fix torvalds/linux@d96016a764f6aa5c7528c3d3f9cb472ef7266951, which was not yet backported. As a result, UDP single-datagram GSO packets sent to PPPoE don't have their checksums calculated, and are dropped by peer.

The fix is simple: backport that fix to 24.10, or move to 25.12.
@nbd168