openwrt/openwrt #17856: x86_64 / 24.10.0 boot fails (bootloop)

LegendaryYbarro2005, 2025-02-05:
### Describe the bug

The router crashes when booting. This cause when it can't load the rootfs. Watchdog reboots the machine.

The text shows the screen:
> Starting 'OpenWrt'

At this point, the system hangs and reboots the machine.

In addition, failsafe mode doesn't work and router is eventually reboots the machine.

Booting the older version like OpenWrt 24.10.0-rc2 or 23.05.5 boots work fine.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

x86/64

### Device

x86_64 virtual machine

### Image kind

Official downloaded image

### Steps to reproduce

Download a OpenWrt 24.10.0 from official image or custom images from OpenWrt Firmware selector.

### Actual behaviour

Kernel crashes at some point and result of boot loop: Looks like crash. The router is eventually rebooted. Looks like watchdog to me.

### Expected behaviour

Booting the OpenWrt 24.10.0 should works fine without bootloop instead of kernel crash happening.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

aimacintyre, 2025-02-05:
What virtual machine environment?

LegendaryYbarro2005, 2025-02-05:
> What virtual machine environment?
A virtual machine (VM) is the virtualization or emulation of a computer system. I using VirtualBox for testing the OpenWrt before using it on a production router.

LegendaryYbarro2005, 2025-02-05:
Downloaded from official downloaded image (`https://downloads.openwrt.org/releases/24.10.0/targets/x86/generic/`) and used as EFI image. Unfortunately it doesn't boot and got the UEFI shell when starting the VM. I tried typing exit and booting from other devices which keeps throws out. See video for evidence:
https://github.com/user-attachments/assets/3c594d2c-5fdd-431f-b0bf-cff6530480d9

pkgadd, 2025-02-05:
> Downloaded from official downloaded image (https://downloads.openwrt.org/releases/24.10.0/targets/x86/generic/) and used as EFI image.

Unless you really know what you're doing (and why), you need to use the x86_64 combined-efi images and not generic (i386):
https://downloads.openwrt.org/releases/24.10.0/targets/x86/64/

(generic is i386, which also must be booted by an i386 UEFI, while there were some second generation Atom boards with this, it's very uncommon and unlikely to be emulated by virtualbox).

LegendaryYbarro2005, 2025-02-18:
Sorry for late response. I have did this and bootloop/kernel panic issue disappeared.

Settings -> System -> Processor assign more than 2 processor cores (I had quad core running AMD A8-7680).

Run again in OpenWrt 24.10.0 (both 32bit and 64bit) and then it works again but I don't really understand why it doesn't work in one processor core which cause bootloop or kernel panic during booting.