openwrt/openwrt #17858: <Synology DS213j> unable flash image via sysupgrade

Z10N0110, 2025-02-05:
### Describe the bug

Unable to flash image via `sysupgrade` command as follow the steps described in commit 3f7637b.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Synology DS213j

### Image kind

Official downloaded image

### Steps to reproduce

Follow the steps described in commit 3f7637b.

### Actual behaviour

The UART will output mtd writing error messages as below after executing the `sysupgrade`, and device cannot be boot as expected.

```shell
Could not open mtd device: image
Can't open device for writing!
cat: write error: Broken pipe
s[  126.847664] reboot: Restarting system
```

### Expected behaviour

-

### Additional info

The total UART output:

``` shell
Terminal ready
?��������������������������������������������������������������������{������������������������������������������������������������������������������������������������������������������������������������{������������������{��������������������������������������������������������������������������������������������
BootROM 1.08
Booting from SPI flash
High speed PHY - Version: 2.1.2 (COM-PHY-V20)
Update PEX Device ID 0x67100
High speed PHY - Ended Successfully
                                   DDR3 Training Sequence - Ver 4.5.0
DDR3 Training Sequence - Ended Successfully
Status = MV_OK
BootROM: Image checksum verification PASSED

 __   __                      _ _
|  \/  | __ _ _ ____   _____| | |
| |\/| |/ _` | '__\ \ / / _ \ | |
| |  | | (_| | |   \ V /  __/ | |
|_|  |_|\__,_|_|    \_/ \___|_|_|
         _   _     ____              _
        | | | |   | __ )  ___   ___ | |_
        | | | |___|  _ \ / _ \ / _ \| __|
        | |_| |___| |_) | (_) | (_) | |_
         \___/    |____/ \___/ \___/ \__|
[... 398 lines cut ...]
       SATA0 (Lane 2)
       SATA1 (Lane 3)
       SGMII0 Phy module (Lane 1)

Synology Model: DS213j
Fan Status: Good

Net:   egiga0 [PRIME]
Press Ctrl+C to abort autoboot in 3 second
Wrong Image Format for bootm command
ERROR: can't get kernel image!
Marvell>> printenv bootcmd
bootcmd=bootm f4110000
Marvell>>
```

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

musashino205, 2025-02-06:
The following commit seems to have moved the `virtual_flash` node to the wrong location.

https://github.com/openwrt/openwrt/commit/2260ec4fb7a4689482b6d55b1ffcb50dfe7c6fcb

cc: @dangowrt

Z10N0110, 2025-02-15:
So any updates on this?