openwrt/openwrt #17865: Mesh and AP not working together on Openwrt 24.10 on Xiaomi 4a Gigabit Edition.

fernandodiacenco, 2025-02-05:
### Describe the bug

Router is XIaomi 4a Gigabit Edition Openwrt is version 24.10.0

To reproduce:
Create a 802.11s mesh on 2.4ghz, it works.
Create an AP on the same radio on 2.4ghz, now neither will come up.
Remove 802.11s and uses only ap, works.

This works fine on Openwrt 23.05.5, it stopped working on 24.10.0

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

ramips/mt7621

### Device

Xiaomi Mi Router 4A Gigabit Edition

### Image kind

Official downloaded image

### Steps to reproduce

To reproduce:
Create a 802.11s mesh on 2.4ghz, it works.
Create an AP on the same radio on 2.4ghz, now neither will come up.
Remove 802.11s and uses only ap, works.

### Actual behaviour

Neither mesh or ap comes up unless one or the other is removed from config.

### Expected behaviour

Both working as they do in version 23.x

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-02-06:
As we figured out on forum:
https://forum.openwrt.org/t/mesh-and-ap-on-the-same-radio-stopped-working-on-24-10-0/223821/8

Functionality b4oke between -rc5 and -rc6

Check history between releases and try to bisect building middle versions until you find evil patch.
https://github.com/openwrt/openwrt/commits/openwrt-24.10/

dspagnol, 2025-02-07:
I'm having the same problem in the following hardwares:

- GL.iNet GL-MT6000 (mediatek-filogic)
- TP-Link Archer C7 v4 (ath79-generic)

Downgrading to 24.10.0-rc5 solved the problem on both routers.

brada4, 2025-02-07:
Seems this is only change r1 to r2?
https://github.com/openwrt/openwrt/commit/767339066e9fc3a2232e9d2881469797cbf7ba8b

dspagnol, 2025-02-08:
I double checked the documentation, replaced the wpad-basic-mbedtls with wpad-mesh-mbedtls and it's working on 24.10. My bad.

```
BusyBox v1.36.1 (2025-02-03 23:09:37 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 24.10.0, r28427-6df0e3d02a
 -----------------------------------------------------
root@bucanero:~# opkg list-installed | grep wpad
wpad-mesh-mbedtls - 2024.09.15~5ace39b0-r2
```

fernandodiacenco, 2025-02-08:
I tried exactly the same as dsspagnol but I still have a problem on Xiaomi, when I revert to RC5 default image which has 2024.09.15~5ace39b0-r1 it works normally.

> BusyBox v1.36.1 (2025-02-04 19:21:11 UTC) built-in shell (ash)
> 
>   _______                     ________        __
>  |       |.-----.-----.-----.|  |  |  |.----.|  |_
>  |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
>  |_______||   __|_____|__|__||________||__|  |____|
>           |__| W I R E L E S S   F R E E D O M
>  -----------------------------------------------------
>  OpenWrt 24.10.0, r28427-6df0e3d02a
>  -----------------------------------------------------
> root@Wifi-Escritorio:~# opkg list-installed | grep wpad
> wpad-mesh-mbedtls - 2024.09.15~5ace39b0-r2

Neustradamus, 2025-02-23:
To follow this ticket.

ddmesh, 2025-06-09:
Freifunk Dresden uses 80211s meshing with a lot of routers. since v24.10.0-rc6 meshing is INCOMPATIBLE.
Consider two groups of routers. 
Group A - contains routers with firmware based on openwrt upto v22 (we don't have v23).
Group B - contains routers with firmware starting from v24.10.0-rc6 (still not released).

Meshing on 2.4GHz:
 - All Routers within Group A can mesh with each other
 - All Routers within Group B can mesh with each other
 BUT: NO MESHING beetween Group A and Group B anymore (wifi settings are same on all routers)

Meshing on 5Ghz:
 - All routers can mesh with each other

Since a lot of "old" routers MUST stick to openwrt 22 (because of increasing flash/ram demands)
we can not update those to openwrt 24. And because new router are only supported/introduced for openwrt 24. We have a compatiblity issue.

I have not tested it, but I assume that meshing between openwrt 24 and other vendors (fritzbox,or so) will also break. 

Tested: different routers from target: 
 - ramips-mt7621-generic
 - mediatek-filogic (cudy wr3000) which has a lot of wifi instabilities

Also: ramips and mediatek show a lot of "Crash warnings" in syslog every few minutes since openwrt 22. All releated to 80211s.

My impression is, that wifi gets worse and worse and firmware gets more and more untested on real devices before any release.