openwrt/openwrt #17868: GL-MT6000 channel auto change from 44 to 48

ziiiiiiiii, 2025-02-06:
### Describe the bug

I use Openwrt 24.10 and there’s a bug to fix:
I select channel 44 (5Ghz) but Openwrt auto change to channel 48.
I re-flash firmware but this problem still happens.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

mediatek/filogic

### Device

GL.iNet GL-MT6000

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-02-06:
There is no "problem", since wifi4 main 20mhz channel is selected automatically. Use 20MHz channels to disable standard behavior.

ziiiiiiiii, 2025-02-06:
> There is no "problem", since wifi4 main 20mhz channel is selected automatically. Use 20MHz channels to disable standard behavior.

You mean use 20mhz instead of 160mhz on 5hgz on wifi 6 router?

brada4, 2025-02-06:
Since you see a problem in swapping beacon channel you have no other choice [/sarcasm]

ziiiiiiiii, 2025-02-06:
> Since you see a problem in swapping beacon channel you have no other choice [/sarcasm]

On Openwrt 23.05.5 I use wifi 5ghz channel 44 normally but on 24.10 it doesn't work so I wrote this to report problem.

brada4, 2025-02-06:
24.10 just reports back active channel. Zeerooo to fix.