openwrt/openwrt #17871: Sysupgrade bricking router -> Linksys MX5300

austonpramodh, 2025-02-06:
### Describe the bug

Sysupgrade from 24.10.0 rc-7 -> 24.10.0 bricks the router.

I can recover it back due to the router being dual partition. Just reboot thrice, and it would blink red light fast and reboot again.

unfortunately, I can test out a lot since I don't have uart plugged in. 

### OpenWrt version

r28417-daef29c75d

### OpenWrt release

24.10.0-rc7

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Linksys MX5300

### Image kind

Official downloaded image

### Expected behaviour

Sysupgrade from 24.10.0 RC7 to 24.10.0

### Additional info

However, I was able to get 24.10.0 through the factory image using the instructions here http://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=70fd815e57dc98cef88828faa5ce71717540e99e and it works when flashed with factory image.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

samara1532, 2025-02-06:
-> 24.10.0 bricks the router.   this is not the first router model on which bricks

24.10.0 successful release  (no) .

testuser7, 2025-02-06:
@austonpramodh Please check if you have the same problem: https://github.com/openwrt/openwrt/issues/17681

closed 2025-10-19 by hauke