openwrt/openwrt #17873: LAN port on the NanoPi R4S 4GB Standard Edition stops working after a reboot!

shalom-epta, 2025-02-06:
### Describe the bug

LAN port on the NanoPi R4S 4GB Standard Edition stops working after a reboot!
changing the SD card or the power supply did not give any result! after a reboot, the lan port falls off

### OpenWrt version

OpenWrt 24.10.0 r28427-6df0e3d02a / LuCI openwrt-24.10 branch 25.014.55016~7046a1c

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

rockchip/armv8

### Device

NanoPi R4S 4GB Standard Edition

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

shalom-epta, 2025-02-06:
> Invalid Version reported. `OpenWrt 24.10.0 r28427-6df0e3d02a / LuCI openwrt-24.10 branch 25.014.55016~7046a1c` Is this from a clean repository?

https://firmware-selector.openwrt.org/?version=24.10.0&target=rockchip%2Farmv8&id=friendlyarm_nanopi-r4s

shalom-epta, 2025-02-06:
самая ублюдская поддержка! знают о проблеме, но исправлять не хотят!

shalom-epta, 2025-02-08:
> For the R4S 4GB version without eMMC, this issue has been resolved as of version 24.10.0. Additionally, please refrain from blaming the developers.

The issue is not resolved, starting from the release candidates! I just checked, flashed 24.10.0, immediately restarted, and got a LAN down!

shalom-epta, 2025-02-08:
> > > For the R4S 4GB version without eMMC, this issue has been resolved as of version 24.10.0. Additionally, please refrain from blaming the developers.
> > 
> > 
> > The issue is not resolved, starting from the release candidates! I just checked, flashed 24.10.0, immediately restarted, and got a LAN down!
> 
> Invalid Version reported. OpenWrt 24.10.0 r28427-6df0e3d02a / LuCI openwrt-24.10 branch 25.014.55016~7046a1c Is this from a clean repository?

About 10-15 minutes ago, I downloaded the firmware here https://firmware-selector.openwrt.org/?version=24.10.0&target=rockchip%2Farmv8&id=friendlyarm_nanopi-r4s for the FriendlyARM NanoPi R4S 4GB LPDDR4 24.10.0 - on it, the LAN port drops out just like it did on the release candidates! Even on the OpenWRT forum, they wrote that no changes were made for the FriendlyARM NanoPi R4S 4GB LPDDR4!
What fix are you talking about? I don't know, IMHO these are your fabrications! The problem still exists! Since no changes have been made!!!

shalom-epta, 2025-02-08:
> Я снова протестировал все, включая PPPOE и IPv6, и все работает отлично. Я также перезапустил несколько раз, и порт локальной сети функционирует без каких -либо проблем. Пожалуйста, проверьте свой сетевой кабель.

I'm repeating this again, for the alternative-minded!
The problem is in the firmware! Stupid advice like "change the cable", "change the SD card", etc., keep to yourself!
On the OpenWRT forum, several people have reported this issue, and others have said it works fine for them!
They also mentioned that they are aware of the problem, and they won't fix it! So, we have a $100 router that doesn't work beyond 23.05.5!

shalom-epta, 2025-02-08:
> No log = no solution.

Need a log? Give me UART! They don't sell it here!

shalom-epta, 2025-02-08:
> Buy it on Amazon/eBay

They don't exist in Russia.

shalom-epta, 2025-02-08:
> Aliexpress

it's easier to buy another router, where the support is much more adequate, and fixes problems, rather than sending a fuck.

shalom-epta, 2025-02-08:
Order a UART, wait a few months for it to arrive! wait for the developers to fix the problem! or should I go buy another router? I think the question is obvious!

sqrwf, 2025-02-08:
> it's easier to buy another router, where the support is much more adequate

You should demand a full refund on what you paid for OpenWrt and the attached service level agreement.

shalom-epta, 2025-02-08:
> > it's easier to buy another router, where the support is much more adequate
> 
> You should demand a full refund on what you paid for OpenWrt and the attached service level agreement.

тебе следует пойти нахуй!

sqrwf, 2025-02-08:
> > You should demand a full refund on what you paid for OpenWrt and the attached service level agreement.
> 
> тебе следует пойти нахуй!

I'm terribly sorry for being unclear, I didn't mean you should make demands on _me_. For refunds on your OpenWrt support contract please contact the OpenWrt service center you made the contract with. You will find the contact data on your invoices.