openwrt/openwrt #17881: Kirkwood/Check Point L-50w - No WiFi after upgrading to 24.10.0 from 23.05.5

Prazola, 2025-02-06:
### Describe the bug

After upgrading from 23.05.5 to 24.10.0, radio is not detected and it's impossibile to enable anything wireless related.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

kirkwood/generic

### Device

Check Point L-50

### Image kind

Official downloaded image

### Steps to reproduce

Upgrade from 23.05.5 to 24.10.0 on a Check Point L-50w (with WiFi module)

### Actual behaviour

After upgrading from 23.05.5, radio is not detected and it's impossibile to enable anything wireless related.

### Expected behaviour

Radio should be detected and enabled.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

chainqq, 2025-02-12:
I experienced this issue after upgrading on Friday.

Going back to 23.05.5 and importing my settings from backup resolved the issue, allowing me to continue using Wi-Fi.  Ethernet appeared to work under 24.10.0, however.

I did not do any further testing, only using 24.10.0 for a few minutes before going back to 23.05.5.

ferrashoo, 2025-03-17:
Tried the update 24.10 on my L-50, removed the kmod-ath, kmod-ath9k driver(s) and restarted the device after the upgrade, no success. Attempt to add the ap0 after reinstalling the atheros drivers but it could not find the device. Updated to latest snapshot, same issue. Hope this helps resolve the issue soon.

Prazola, 2025-04-16:
Issue still present in 24.10.1

Prazola, 2025-08-30:
> > Same issue with 24.10.2
> 
> Are both 2.4 and 5ghz radios nonfunctional? or just 5ghz?
> 
> Same issue? [#19088](https://github.com/openwrt/openwrt/issues/19088)

Only 2.4GHz is supported by this device.

Prazola, 2025-10-22:
Issue persists with 24.10.4.

Prazola, 2026-04-21:
Confirmed still present int 25.12.X