openwrt/openwrt #17887: No Wireless for TP-Link Archer C6 (EU) in release 24.10

ashr123, 2025-02-06:
### Describe the bug

This is what I see after the upgrade:

[image]

In 23.05 it worked just fine, how can I solve it?

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer C6 v2 (EU/RU/JP)

### Image kind

Official downloaded image

### Steps to reproduce

Do sysugprade from release 23.05 to 24.10

### Actual behaviour

[image]

No wireless "device"

### Expected behaviour

Wireless need to work just fine

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

p0k33m0n, 2025-02-07:
I had the same thing on my router Netgear WNDR4300, but after a few reboots it started working and is still working (may it stay that way!). 24.10 seems a bit too soon released.

ashr123, 2025-02-07:
@p0k33m0n did you rebooted your router since it started working?

p0k33m0n, 2025-02-07:
As I wrote: several times, after around 2-3 it started to work.

ashr123, 2025-02-07:
@p0k33m0n in addition, I've tryied to reset the configurations to default, in that case I saw in the wireless section only 2.4GHz and not 5GHz, I think in my case it's more serious problem

Shine-, 2025-02-07:
https://openwrt.org/releases/24.10/notes-24.10.0#known_issues

#14541

p0k33m0n, 2025-02-07:
> https://openwrt.org/releases/24.10/notes-24.10.0#known_issues
> 
> [#14541](https://github.com/openwrt/openwrt/issues/14541)

My router is on Atheros, I don't think it would be an accident.

Shine-, 2025-02-07:
@p0k33m0n - you shouldn't hijack other issues. Not this one and not other ones you already hijacked. Your device is **ath9k**, not ath10k. This issue is related to **ath10k only** and as already mentioned above, a duplicate to #14541, and it's a known issue as described in the release notes for 24.10.

p0k33m0n, 2025-02-07:
> you shouldn't hijack other issues.

I had exactly the same situation during the first few launches of the router, so don't tell mo what to connect to what. I consider the 24.10 release to be rushed and that's it.

sp4rkiop, 2025-06-15:
hey, just to add up. I have tried installing latest 24.10.0 / 24.10.1 on TP-Link C6U, and on both the device fails to boot during 1st  few attempts, have to restart it several times plugging unplugging to make it boot. I had uart to see the issue but saw logs related to kernel/sector i don't remember. I can share any details required if anyone steps in.

ashr123, 2025-06-15:
@sp4rkiop of course, share shat you saw

closed 2025-08-11 by ashr123