openwrt/openwrt #17888: Problems with WiFi CPU recognition and Settings Hidden - Belkin RT3200/Linksys E8450

gabriel120pro, 2025-02-07:
### Describe the bug

After installing the new major update 24.10, the device, without maintaining the configurations of the previous compilation, does not detect the CPUs indicated for Wi-Fi and its configurations are not detected, the problem is "solved" when the radio of the respective CPU is enabled, showing its name, however, the configurations are limited, in the case of selecting Maximum transmit power it is limited to a range lower than the corresponding one, if the Wi-Fi is configured and kept enabled, the display of this is repaired resource, showing the real one.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Linksys E8450 (UBI)

### Image kind

Official downloaded image

### Steps to reproduce

Described in the bug description more thoroughly.
Install 24.10.0, Go to Network > Wireless

### Actual behaviour

Interface shows Generic unknown on 2.4 and 5Ghz CPUs

[image]

### Expected behaviour

It should show MT7622e and Mt7915e

[image]

### Additional info

Configuration problems

[image]

The real config

[image]

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-02-07:
Hope you have pre-upgrade backup, if not make one now. Try to reset device to defaults and report back whether wifi returns.

gabriel120pro, 2025-02-08:
> Hope you have pre-upgrade backup, if not make one now. Try to reset device to defaults and report back whether wifi returns.

I have a backup copy of the boot files created by the same factory flash to ubi that is provided on the official openwrt page of this router, of the configurations, no, but I think it is not necessary since it is a new version of openwrt, the problem persists after a factory reset

RolandoMagico, 2025-02-10:
Duplicate of https://github.com/openwrt/openwrt/issues/14207

Will be fixed with https://github.com/openwrt/openwrt/pull/17821