openwrt/openwrt #17891: Bridged AP not working on Netgear R7800 (IPQ806X) with wan port added to bridge in release 24.10

egc112, 2025-02-07:
### Describe the bug

Setup as a [Dumb/Bridged  AP](https://openwrt.org/docs/guide-user/network/wifi/wifiextenders/bridgedap) WAN port added to the br-lan bridge.
No access from LAN ports to the router only from the WAN port

### OpenWrt version

r28433-8cd50a6aec

### OpenWrt release

24.10-SNAPSHOT

### OpenWrt target/subtarget

ipq806x/generic

### Device

Netgear Nighthawk X4S R7800

### Image kind

Self-built image

### Steps to reproduce

Setup as a [Dumb/Bridged  AP](https://openwrt.org/docs/guide-user/network/wifi/wifiextenders/bridgedap)

I did a quick test, my router was on its own subnet 192.168.5.0/24 But I now have made it a Dumb AP, deleted everything from the wan and  added the wan port to br-lan, set the IP address in the main lan (192.168.0.1/24) and added gateway and dns and disabled the firewall:
```
config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'lan1'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'lan4'
	list ports 'wan'
	option ip6assign '64'
	option ip6hint '5'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.0.5'
	option netmask '255.255.255.0'
	option ip6assign '63'
	option gateway '192.168.0.1'
	list dns '192.168.0.1'
```

### Actual behaviour

Rebooted the router and my workstation which is on LAN port 3  had internet. The WAN port still had the cable to the main router.

What is not working is that from the LAN ports I cannot connect/ping the router at 192.168.0.5, but I can from the WAN port

https://forum.openwrt.org/t/openwrt-24-10-0-first-stable-release/223845/85?u=egc 

### Expected behaviour

Both WAN and LAN ports would work as if on the same switch, from LAN ports I can connect to the router

### Additional info

Not sure what to make of this, I know this router has a dual CPU port to the switch but not sure how to debug this, I know my way around swconfig but how to do this for DSA?

https://forum.openwrt.org/t/openwrt-24-10-0-first-stable-release/223845/85?u=egc 

[... 293 lines cut ...]
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_unet-cli=y
CONFIG_PACKAGE_unetd=y
CONFIG_PACKAGE_vxlan=y
CONFIG_PACKAGE_wg-installer-client=y
CONFIG_PACKAGE_wget-ssl=y
CONFIG_PACKAGE_wireguard-companion=y
CONFIG_PACKAGE_wireguard-tools=y
CONFIG_PACKAGE_wireguard-watchdog=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_wpad-openssl=y
CONFIG_PACKAGE_wsdd2=y
CONFIG_PACKAGE_xtables-nft=y
CONFIG_PACKAGE_zlib=y
CONFIG_PCRE2_JIT_ENABLED=y
CONFIG_PREINITOPT=y
CONFIG_TARGET_PREINIT_TIMEOUT=5
egc@egc-VirtualBox:/linuxdata/openwrt$
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

rwl408, 2025-02-08:
My comment is from the experience to configure a Netgear XR500 (a R7800 minus the eSATA) as a Dumb AP. Someone offered a kludge [https://forum.openwrt.org/t/openwrt-24-10-0-rc7-seventh-release-candidate/223006/70](url) to walk around the problem. The solution modified the file /etc/config/netowrk with an added vlan layer:

config device            
        option name 'br-lan'
        option type 'bridge'
        option ipv6 '0'  
        list ports 'lan1' 
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'  
        list ports 'wan'        
                                               
config interface 'lan'     
        option device 'br-lan.1'
        option proto 'static'
        option ipaddr '192.168.1.129'
        option netmask '255.255.255.0'
        option gateway '192.168.1.1'
        option broadcast '192.168.1.255'
        option delegate '0'
                                               
config bridge-vlan            
        option device 'br-lan'
        option vlan '1'       
        list ports 'lan1:u*'
        list ports 'lan2:u*'
        list ports 'lan3:u*'
        list ports 'lan4:u*'
        list ports 'wan:u*'

The solution worked and had tested with OpenVPN and WireGuard. But after I installed the packages for Docker (luci-app-dockerman dockerd docker), I lost all internet connection (both wired and wifi). The Docker installation created a new bridge and a new interface:

config interface 'docker'
	option device 'docker0'
	option proto 'none'
	option auto '0'

config device
	option type 'bridge'
	option name 'docker0'

So I am stuck again.

Ramon00, 2025-02-08:
Maybe try adding eth0 and eth1 to the bridge (like it was on 23.05)? I think all 7 are on the same HW , 2 bridge ports are connected to the CPU, 4 to the LAN ports 1 to the WAN port. In the previous version default was that eth0 was tagged on VLAN2. WAN was untagged on vlan 2, eth 1 was tagged on VLAN1, all the lan ports were untagged on vlan1. Im guessing it has got to do something with that.

rkboni, 2025-02-08:
Same behavior on a Linksys EA8500.  Router was connected to rest of LAN via `lan1` initially, with just `lan[1-4]` ports in the `br-lan` (and with `br-lan` configured as a DHCP client), and everything worked, internet access via wireless was fine.

Added `wan` to the `br-lan` (without moving cable), saved and applied, and the router dropped off the network (presumably DHCP failed on `br-lan`).  Moved cable from `lan1` to `wan` port and router is back online.

Config snippets: 
```
config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'
        list ports 'wan'

[...]
config interface 'lan'
        option device 'br-lan'
        option proto 'dhcp'
        option ip6assign '60'
```

and:
```
config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'sae'
        option key '....not-my-password..,'
        option ocv '0'

config wifi-iface 'wifinet2'
        option device 'radio0'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'sae'
        option key '...not-my-password...'
        option ocv '0'
        option network 'lan'
```

egc112, 2025-02-09:
> Maybe try adding eth0 and eth1 to the bridge (like it was on 23.05)? I think all 7 are on the same HW , 2 bridge ports are connected to the CPU, 4 to the LAN ports 1 to the WAN port. In the previous [...]

Could well be related, as it has two cpu ports, tagging should not be needed at all I think?

kkudielka, 2025-02-09:
I think I see the same problem on a TP-Link Archer C2600 (ipq806x). Presently I am running 24.10.0-rc7, but that's not a big deal.

- Everything works as expected in "accesspoint" mode, as long as I do **not** add the 'wan' device to the 'br-lan' bridge.
- There are no 'wan' interfaces in /etc/config/network, so the 'wan' device is not managed by netifd anymore.

And then, a trial at the lowest possible level:
- `ìp link set wan up` ... still okay.
- `ip link set wan master br-lan` ... I immediately lose connectivity on the LAN ports.

kkudielka, 2025-02-09:
Okay, some success using the real iproute2:
```
opkg install ip-full
ip link set wan type dsa conduit eth1
ip link set wan master br-lan
ip link set wan up
```
And then I have LAN connection on all ports.
So, it looks like all ports on the switch must have the same conduit... Which is not the idea of DSA.

Ramon00, 2025-02-09:
> Could well be related, as it has two cpu ports, tagging should not be needed at all I think?

CPU ports need to be tagged I think.  I am not sure if both CPU ports are interacting on the same VLANs, maybe that will create a loop?

egc112, 2025-02-09:
I also have an R7800 running  DDWRT and the cpu ports are not tagged.
If you have one cpu port then tagging is necessary with two it does not at least not on my ddwrt router

dewi-ny-je, 2025-03-27:
Is this outdated info?
https://openwrt.org/toh/netgear/r7800#network_interfaces

kiss81, 2025-03-30:
still can't add the WAN port to the "dumb ap":
[LuCI openwrt-24.10 branch (25.037.68331~42f464c)](https://github.com/openwrt/luci) / KONG 24 (r28521-f3a210b742)

dewi-ny-je, 2025-04-01:
> Could well be related, as it has two cpu ports, tagging should not be needed at all I think?

Not needed, but maybe working to avoid the bug? I can try to add both cpu ports to the bridge and see. Otherwise good to know VLANs make it work.

dewi-ny-je, 2025-04-01:
For the time being, my findings: the devices connected to the LAN, wifi, and the router itself (web, SSH) are reachable only from WAN, not from LAN+wifi.
So, from my desktop on LANx I cannot reach any other LANy device, and I cannot reach any wifi device. And the same from a wifi device: I can reach the rest of the home network connected to WAN, but nothing connected to LAN of the same router.

It's basically isolating the clients on LAN+wifi. To be clear: the option to isolate clients on wifi is off.

dewi-ny-je, 2025-04-02:
For info, in case other people not expert on the topic wondered: using the VLAN workaround the CPU has no additional load while performing Gigabit transfers, so no loss of performance should take place.

dewi-ny-je, 2025-04-02:
https://github.com/openwrt/openwrt/issues/17891#issuecomment-2645798952

dewi-ny-je, 2025-06-20:
This is repository is about stock openwrt and not hynman. 
Stock openwrt is known to have said bug at least until beginning of 2025.

If you have no issues with hynman then better find out which change he made helps in this regard.

The guides you mentioned are well known but the dumpAP guide gives different instructions

dewi-ny-je, 2025-06-20:
I can confirm that my router is doing fine after following https://github.com/openwrt/openwrt/issues/17891#issuecomment-2645798952 but apparently it doesn't work for everyone?

kkudielka, 2025-06-21:
> ```
> opkg install ip-full
> ip link set wan type dsa conduit eth1
> ip link set wan master br-lan
> ip link set wan up
> ```
> 
> And then I have LAN connection on all ports. So, it looks like all ports on the switch must have the same conduit... Which is not the idea of DSA.

Let me elaborate on that, quoting the kernel's [DSA Documentation](https://docs.kernel.org/networking/dsa/configuration.html):

> Affinity of user ports to CPU ports
> 
> Typically, DSA switches are attached to the host via a single Ethernet interface, but in cases where the switch chip is discrete, the hardware design may permit the use of 2 or more ports connected to the host, for an increase in termination throughput.
> 
> DSA can make use of multiple CPU ports in two ways. First, it is possible to statically assign the termination traffic associated with a certain user port to be processed by a certain CPU port. This way, user space can implement custom policies of static load balancing between user ports, by spreading the affinities according to the available CPU ports.

So, having switch ports with different conduits (CPU ports) on a single bridge, seems to be a valid use case for DSA (static load balancing). To me, the present behaviour looks like a bug of the qca8k DSA driver. Making the conduit equal for all ports on the bridge is just a work-around.

kiss81, 2025-07-31:
> [@Ansuel](https://github.com/Ansuel), any ideas?

Would be nice to have the WAN port available on these "old" devices so we can still use them as dumb AP's / network switches. 
Which is the best workaround if this can't be fixed? Go back to 23.05...?

motolav, 2025-08-16:
@kiss81 you don't need the wan port to be able to use ipq806x devices as dumb APs its only an inconvenience if you need the extra port

jeback1, 2025-08-16:
In my use case, I need the extra port on my tgr1900 AP, which has only two ports. I have my network configured with vlans for lan, guest, and iot. The ap is connected via ethernet cable to a managed switch. I use the ap to provide wifi access to the lan and guest vlans and the second ethernet port for an ip camera on the iot vlan.

dewi-ny-je, 2025-08-16:
Whether we need it or not, this ticket is about the bug.
It would be nice to see it fixed.

Floriszz, 2025-12-04:
After trying many times and losing access to the R7800 and had to factory reset it. I took the time to watch the first video on the page below from OneMarkFifty from minute 12:12 till the end. And managed to do it all through Luci in DSA👍 Works like a charm now!
https://openwrt.org/docs/guide-user/network/wifi/wifiextenders/bridgedap#optionaladd_wan_port_to_the_lan_bridge