openwrt/openwrt #17894: IPV4 and IPV6 paramaters not stored consistently in network and dhcp config file

peterb99, 2025-02-07:
### Describe the bug

Various parameters for the LAN interface for IPV4 and IPV6 are not stored consistently and create a difficult to understand config file

Example: Specifying dns preferred servers to use and set the dhcp option to advertise that server list to clients

For IPV4:

in network file ;

```
config interface 'lan'                         
        option device 'br-lan'                 
        option proto 'static'                  
        option ipaddr '192.168.0.1'            
        option netmask '255.255.255.0'         
        option ip6assign '64'                  
        option ip6ifaceid '::99'      
        list ip6class 'wan6'          
        list ip6class 'local'         
        option delegate '0'           
        list dns '208.67.222.222'     
        list dns '208.67.220.220'     
        list dns '8.26.56.26'         
        list dns '8.20.247.20'        
        list dns '62.6.38.125'        
        list dns '8.8.8.8'            
        option ip6hint '5'   
```

in dhcp config file;

```
config dhcp 'lan'                                                               
        option interface 'lan'                                                  
        option start '100'                                                      
        option limit '150'                                                      
        option leasetime '12h'                                                  
        option force '1'                                                        
        option dhcpv4 'server'                                                  
        option ra_useleasetime '1'                                              
        option dhcpv6 'server'                                                  
        option ra 'server'                                                      
        list ra_flags 'managed-config'                                          
        list ra_flags 'other-config'                                            
        option ra_maxinterval '1200'                                            
        option ra_lifetime '3000'                                               
        list dhcp_option '3,192.168.0.1'                                        
        list dhcp_option '4,192.168.0.1'                                        
        list dhcp_option '6,192.168.0.1'                                        
        list dhcp_option '15,Lan'                                               
        list dhcp_option '42,192.168.0.1'                                       
        list dhcp_option '44,192.168.0.1'                                       
        list dns '2a02:1788:8ff::1'                                             
        list dns '2a02:1788:8ff::2'                                             
        list dns '2620:119:53::53'                                              
        list dns '2620:119:35::35'                                              
        list dns '2606:4700:4700::1111'                                         
        list ntp '0.openwrt.pool.ntp.org'                                       
        list ntp '1.openwrt.pool.ntp.org'     
```        

So for ipv4 the preferred dns list is in the network file , for ipv6 in the dhcp file
Is it not better to have both lists in one place naming them separate like 'list ipv4_dns' and 'ipv6_dns'?
Could this be done generally, keeping all ipv options for a certain feature/function together in one place with ipv4 and ipv6 prefix where applicable?
Also keep the Luci interface consistent give the user interface windows for ipv4 and ipv6 same layout and same names, currently it is very confusing to find the relevant sections for each, again use the local specified list of nameservers to use as example.
The dhcp options to specify to use the local router dns list ( option 6 ) . It is not clear if this is true for both ipv4 and ipv6. For ntp setting ( option 4) there is a different entry in the dhcp ,  list ntp '0.openwrt.pool.ntp.org'   .

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

mediatek/filogic

### Device

Openwrt One

### Image kind

Official downloaded image

### Steps to reproduce

just look at the various config files

### Actual behaviour

more consistent use of Luci user interface and locations of parameters/options in config files

### Expected behaviour

No mount issues

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

jow-, 2025-02-24:
These are unrelated settings. The dns list in the interface settings specifies the DNS server OpenWrt uses itself as upstream. The dns list in the DHCP settings specifies the DNS servers advertised to clients (bypassing the OpenWrt DNS cache).

Both accept IPv4 and IPv6 addresses.

jow-, 2025-02-25:
> should all that info not be under interface -> LAN ?

No, the info under `interface -> LAN` specifies upstream DNS servers for OpenWrt itself, and it should usually not be used at all since that info is usually obtained automatically from wan via DHCP, DHCPv6, PPPoE, slaac or comparable protocols.

In router mode there is usually no need to manually set any DNS related details in Interface -> LAN.

peterb99, 2025-02-25:
Indeed on the main page, but under the tab dhcp server, all setting are for the local dhcp server, 

[image]

[image]

peterb99, 2025-02-27:
my point also is that info for dhcp setup is split between interfaces and dhcp/dns section.

peterb99, 2025-03-30:
> These are unrelated settings. The dns list in the interface settings specifies the DNS server OpenWrt uses itself as upstream. The dns list in the DHCP settings specifies the DNS servers advertised to clients (bypassing the OpenWrt DNS cache).
> 
> Both accept IPv4 and IPv6 addresses.

If both accept ipv4 and ipv6 addresses, should luci not be adapted, as both shown separate in ipv4 and ipv6 sections. The ones in ipv6 under 

[image]
ipv4 under 

[image]