openwrt/openwrt #17895: mt7621 + mt7915e wifi not working and kernel panic

lsoltero, 2025-02-07:
### Describe the bug

On OpenWRT 24.10 configured on an mt7621 SOC using mt7915e WiFi PCIe module 
1. WiFi will only work on cold boot.  
2. on warm boot the kernel will panic and cause a reboot and eventually will startup but the WiFi will not work generating driver errors.

The issue is caused by the mt7621 driver not waiting long enough after resetting the wifi module.  The following patch fixes the issue

`Index: linux-6.6.73/drivers/pci/controller/pcie-mt7621.c
===================================================================
--- linux-6.6.73.orig/drivers/pci/controller/pcie-mt7621.c
+++ linux-6.6.73/drivers/pci/controller/pcie-mt7621.c
@@ -60,7 +60,7 @@
 #define PCIE_PORT_LINKUP		BIT(0)
 #define PCIE_PORT_CNT			3
 
-#define INIT_PORTS_DELAY_MS		100
+#define INIT_PORTS_DELAY_MS		500
 #define PERST_DELAY_MS			100
 
 /**
@@ -319,7 +319,13 @@ static void mt7621_pcie_reset_assert(str
 	list_for_each_entry(port, &pcie->ports, list) {
 		/ PCIe RC reset assert /
 		mt7621_control_assert(port);
+	}
 
+	// https://github.com/openwrt/openwrt/issues/12359
+	// https://github.com/openwrt/mt76/issues/644
+	mdelay(PERST_DELAY_MS);
+ 
+	list_for_each_entry(port, &pcie->ports, list) {
 		/ PCIe EP reset assert /
 		mt7621_rst_gpio_pcie_assert(port);
 	}`

### OpenWrt version

v24.10.0-5-gef122aa73d

### OpenWrt release

24.10

### OpenWrt target/subtarget

mt7621

### Device

mt7915e

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

lsoltero, 2025-02-07:
```
`Index: linux-6.6.73/drivers/pci/controller/pcie-mt7621.c
===================================================================
--- linux-6.6.73.orig/drivers/pci/controller/pcie-mt7621.c
+++ linux-6.6.73/drivers/pci/controller/pcie-mt7621.c
@@ -60,7 +60,7 @@
 #define PCIE_PORT_LINKUP		BIT(0)
 #define PCIE_PORT_CNT			3
 
-#define INIT_PORTS_DELAY_MS		100
+#define INIT_PORTS_DELAY_MS		500
 #define PERST_DELAY_MS			100
 
 /**
@@ -319,7 +319,13 @@ static void mt7621_pcie_reset_assert(str
 	list_for_each_entry(port, &pcie->ports, list) {
 		/ PCIe RC reset assert /
 		mt7621_control_assert(port);
+	}
 
+	// https://github.com/openwrt/openwrt/issues/12359
+	// https://github.com/openwrt/mt76/issues/644
+	mdelay(PERST_DELAY_MS);
+ 
+	list_for_each_entry(port, &pcie->ports, list) {
 		/ PCIe EP reset assert /
 		mt7621_rst_gpio_pcie_assert(port);
 	}`
```

brada4, 2025-02-07:
@nbd168 is the author, you can submit your patch directly upstream linux-wireless.

lsoltero, 2025-02-08:
it has had the above patch applied and committed.  

here is the commit prior to applying the patch.

commit 1fad1b4965dc6f4e5f4ba7b9605987f443a4c276 (tag: v24.10.0)

BTW..this issue has been present since OpenWRT 22.03.