openwrt/openwrt #17897: IPQ5018 addition commit 34d917265 breaks ipq807x/DL-WRX36 boot (due to NAND driver changes?)

hnyman, 2025-02-07:
### Describe the bug

cc @robimarko @georgemoussalem @Ansuel 

It seems that the addition of ipq5018 support by 34d917265 prevents the successful boot of ipq807x/DL-WRX36.

With the changes made by 34d917265 , it seems that the nand detection fails somehow, and then rootfs is not recognized.

```
[    0.896823] nand: ECC (step, strength) = (512, 8) does not fit in OOB
[    0.896836] qcom-nandc 79b0000.nand-controller: No valid ECC settings possible
[    0.910996] bam-dma-engine 7984000.dma-controller: Cannot free busy channel
[    0.918070] qcom-nandc: probe of 79b0000.nand-controller failed with error -28
...
[    1.207390] UBI error: cannot open mtd rootfs, er▒
```

Data points about regression:
* r28737-084618f8db boots ok  (buildbot snapshot build)
* r28762-39750798f7   ok (self-built)

* r28763-34d9172655 + PR17884  fails  (self-built)
* r28775-98d7f4202f   fails (self-built)
(There is no later buildbot build available, yet)

Looking at the really large ipq5018 patchset, in addition of pure ipq5018 changes it also contains generic NAND driver changes in patches 401-421...
Those would be my first suspect for the regression culprits.

Discussion also in: 
* https://forum.openwrt.org/t/dynalink-dl-wrx36-askey-rt5010w-ipq8072a-technical-discussion/110454/3049
* https://github.com/openwrt/openwrt/pull/17884#issuecomment-2641011514

Failing boot:  (r28763-34d9172655 + PR17884)
```
[    0.883951] loop: module loaded
[    0.885369] nand: device found, Manufacturer ID: 0x2c, Chip ID: 0xaa
[    0.885909] nand: Micron NAND 256MiB 1,8V 8-bit
[    0.892499] nand: 256 MiB, SLC, erase size: 128 KiB, page size: 2048, OOB size: 64
[    0.896823] nand: ECC (step, strength) = (512, 8) does not fit in OOB
[    0.896836] qcom-nandc 79b0000.nand-controller: No valid ECC settings possible
[    0.910996] bam-dma-engine 7984000.dma-controller: Cannot free busy channel
[    0.918070] qcom-nandc: probe of 79b0000.nand-controller failed with error -28
[    0.925735] spmi spmi-0: PMIC arbiter version v2 (0x20010000)
[    0.985746] i2c_dev: i2c /dev entries driver
[    0.992186] sdhci: Secure Digital Host Controller Interface driver
[    0.992225] sdhci: Copyright(c) Pierre Ossman
[    0.997305] sdhci-pltfm: SDHCI platform and OF driver helper
[    1.003549] remoteproc remoteproc0: releasing cd00000.q6v5_wcss
[    1.011216] NET: Registered PF_INET6 protocol family
[    1.014642] Segment Routing with IPv6
[    1.018310] In-situ OAM (IOAM) with IPv6
[    1.021906] NET: Registered PF_PACKET protocol family
[    1.026081] 8021q: 802.1Q VLAN Support v1.8
[    1.062567] qcom,cpr4-apss-regulator b018000.cpr4-ctrl: CPR valid fuse count: 4
[    1.062926] cpr4_ipq807x_apss_read_fuse_data: apc_corner: speed bin = 0
...
[    1.167768] cpr4_ipq807x_apss_calculate_target_quotients: apc_corner: fused   STURBO: quot[ 7]=1191, quot_offset[ 7]= 175
[    1.178953] cpr3_regulator_init_ctrl: apc: Default CPR mode = closed-loop
[    1.182041] cpufreq: cpufreq_online: CPU0: Running at unlisted initial frequency: 800000 KHz, changing to: 1017600 KHz
[    1.198045] remoteproc remoteproc0: cd00000.q6v5_wcss is available
[    1.207390] UBI error: cannot open mtd rootfs, er▒[    1.218846] Waiting for root device /dev/ubiblock0_1...
```

Log of successful boot:
```
Fri Feb  7 17:18:23 2025 kern.info kernel: [    1.151440] nand: device found, Manufacturer ID: 0x2c, Chip ID: 0xaa
Fri Feb  7 17:18:23 2025 kern.info kernel: [    1.166319] nand: Micron MT29F2G08ABBGAH4
Fri Feb  7 17:18:23 2025 kern.info kernel: [    1.172716] nand: 256 MiB, SLC, erase size: 128 KiB, page size: 2048, OOB size: 128
Fri Feb  7 17:18:23 2025 kern.notice kernel: [    1.177142] 25 qcomsmem partitions found on MTD device qcom_nand.0
Fri Feb  7 17:18:23 2025 kern.notice kernel: [    1.184296] Creating 25 MTD partitions on "qcom_nand.0":
Fri Feb  7 17:18:23 2025 kern.notice kernel: [    1.190346] 0x000000000000-0x000000100000 : "0:sbl1"
...
Fri Feb  7 17:18:23 2025 kern.notice kernel: [    1.482062] 0x00000fa80000-0x00000fd00000 : "certificate"
Fri Feb  7 17:18:23 2025 kern.info kernel: [    1.499025] cpufreq: cpufreq_online: CPU0: Running at unlisted initial frequency: 800000 KHz, changing to: 1017600 KHz
Fri Feb  7 17:18:23 2025 kern.info kernel: [    1.500827] remoteproc remoteproc0: cd00000.q6v5_wcss is available
Fri Feb  7 17:18:23 2025 kern.notice kernel: [    1.509017] ubi0: attaching mtd18
Fri Feb  7 17:18:23 2025 kern.notice kernel: [    1.946209] ubi0: scanning is finished
Fri Feb  7 17:18:23 2025 kern.notice kernel: [    1.952570] ubi0: attached mtd18 (name "rootfs", size 97 MiB)
Fri Feb  7 17:18:23 2025 kern.notice kernel: [    1.952608] ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
Fri Feb  7 17:18:23 2025 kern.notice kernel: [    1.957313] ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
Fri Feb  7 17:18:23 2025 kern.notice kernel: [    1.964083] ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
Fri Feb  7 17:18:23 2025 kern.notice kernel: [    1.970930] ubi0: good PEBs: 776, bad PEBs: 0, corrupted PEBs: 0
Fri Feb  7 17:18:23 2025 kern.notice kernel: [    1.977706] ubi0: user volume: 3, internal volumes: 1, max. volumes count: 128
Fri Feb  7 17:18:23 2025 kern.notice kernel: [    1.983952] ubi0: max/mean erase counter: 112/75, WL threshold: 4096, image sequence number: 634945484
Fri Feb  7 17:18:23 2025 kern.notice kernel: [    1.990991] ubi0: available PEBs: 0, total reserved PEBs: 776, PEBs reserved for bad PEB handling: 40
Fri Feb  7 17:18:23 2025 kern.notice kernel: [    2.000286] ubi0: background thread "ubi_bgt0d" started, PID 630
Fri Feb  7 17:18:23 2025 kern.info kernel: [    2.000948] block ubiblock0_1: created from ubi0:1(rootfs)
Fri Feb  7 17:18:23 2025 kern.info kernel: [    2.015741] clk: Disabling unused clocks
Fri Feb  7 17:18:23 2025 kern.info kernel: [    2.023710] VFS: Mounted root (squashfs filesystem) readonly on device 254:0.
```

### OpenWrt version

r28763-34d9172655

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

DL-WRX36

### Image kind

Self-built image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

hnyman, 2025-02-07:
I removed the NAND driver related patches 401-431 added by the ipq5018 commit, and DL-WRX36 happily boots again, with the correct flash chip identified and with the correct OOB size.
```
[    0.873074] loop: module loaded
[    0.874478] nand: device found, Manufacturer ID: 0x2c, Chip ID: 0xaa
[    0.875031] nand: Micron MT29F2G08ABBGAH4
[    0.881620] nand: 256 MiB, SLC, erase size: 128 KiB, page size: 2048, OOB size: 128
```

robimarko, 2025-02-07:
Ugh, so sadly something broke in the upstream driver changes.

Ansuel, 2025-02-07:
@hnyman i wonder... can you try to use the ipq95xx patch?

robimarko, 2025-02-07:
@Ansuel Doesnt that one only target SNAND driver and not the generic one?

Ansuel, 2025-02-07:
@robimarko AFAIK they generalized the SNAND driver for the SPI driver so it affect ALSO the previous SNAND driver

Ansuel, 2025-02-07:
yep ipq95xx have the latest one that should be v14 i guess?

robimarko, 2025-02-07:
Yes, v14 was the one that was merged (At least the NANDC part, SNAND was not merged) but its still broken.

To reproduce, just boot initramfs on any IPQ807x, IPQ60xx board, probably even IPQ40xx would be broken if this was in generic patches.

Ansuel, 2025-02-07:
Ok time to get hands dirty and repro :D I will tackle this bad boy

robimarko, 2025-02-07:
Craziest part is that it detects the NAND but then:
`nand: No NAND device found`

Ansuel, 2025-02-07:
Compiling image....

robimarko, 2025-02-07:
Sure, I intented to put it in generic as well since part was merged upstream and ipq957x uses the same as well

robimarko, 2025-02-07:
BTW, its `qcom_nand_host_init_and_register` that is failling.

Ansuel, 2025-02-07:
I bet it's the FILED_PREP patch :D

robimarko, 2025-02-07:
Could easily be it

Ansuel, 2025-02-07:
the other patch are quite clean that is the only one where they actually change stuff... or maybe they decided to silently change stuff when they made the "lib" patch ?

Ansuel, 2025-02-07:
Well for sure it's not good... cpu_to_le32 is still a NOP on our system but it's a regression anyway...

robimarko, 2025-02-07:
Its gotta be the `mtd: rawnand: qcom: use FIELD_PREP and GENMASK` since the conversion to a common lib still works.

Ansuel, 2025-02-07:
If only my pc didn't decided to explode right now I could repro it... ahhhhhh

robimarko, 2025-02-07:
BTW, I dont think they believe in warnings at QCA:
```
In file included from ./include/linux/string.h:294,
                 from ./include/linux/bitmap.h:11,
                 from ./include/linux/cpumask.h:12,
                 from ./include/linux/smp.h:13,
                 from ./include/linux/lockdep.h:14,
                 from ./include/linux/mutex.h:17,
                 from ./include/linux/notifier.h:14,
                 from ./include/linux/clk.h:14,
                 from drivers/mtd/nand/qpic_common.c:6:
In function 'fortify_memset_chk',
    inlined from 'qcom_clear_bam_transaction' at drivers/mtd/nand/qpic_common.c:88:2:
./include/linux/fortify-string.h:507:25: warning: call to '__write_overflow_field' declared with attribute warning: detected write beyond size of field (1st parameter); maybe use struct_group()? [-Wattribute-warning]
  507 |                         __write_overflow_field(p_size_field, size);
```

Which we then have this patch for:
https://github.com/openwrt/openwrt/blob/main/target/linux/qualcommax/patches-6.6/0408-spi-spi-qpic-fix-compilation-issues.patch

And I am not to sure about it

robimarko, 2025-02-07:
And yeah, its `mtd: rawnand: qcom: cleanup qcom_nandc driver`

Ansuel, 2025-02-07:
but you said it wasn't?

robimarko, 2025-02-07:
Ugh, not the cleanup one, but `mtd: rawnand: qcom: use FIELD_PREP and GENMASK`, damn copy /paste

Ansuel, 2025-02-07:
field prep on BIT... i can already see the problem....

Ansuel, 2025-02-07:
Hope i'm not late i can finally join the party...

hnyman, 2025-02-07:
They are defining the mask separately for all other bitfields except for ECC_PARITY_SIZE_BYTES_RS.
There mask replaces the the fields start bit definition.

```
-#define	ECC_PARITY_SIZE_BYTES_RS	19
+#define	ECC_PARITY_SIZE_BYTES_RS	GENMASK(22, 19) except 
```

Ansuel, 2025-02-07:
oh god i think i found it....

Ansuel, 2025-02-07:
Yep preparing a patch... it's another case of.... not tested the other variant :(

robimarko, 2025-02-07:
Ah FFS, QCA is constantly somehow outdoing themselves by not testing stuff at all.

Ansuel, 2025-02-07:
didn't want to say that explicitly but yhea ahahahaa

georgemoussalem, 2025-02-08:
Just woke up to this (am probably on the other side of the globe than you guys)..  Wow, what a nasty one. Another case of not testing at qcom. Will do some testing later today with your patches on ipq5018 and report back.

commit 4b6886d9 (2025-02-08): generic: fix Qcom SNAND driver and move to backports directory

Add patch to fix Qcom SNAND driver and move the SNAND patches to
backports directory as they are shared between qualcommax and qualcommbe
target.

Fixes: #17897

Link: https://github.com/openwrt/openwrt/pull/17900
PR #17900 (2025-02-08): generic: fix Qcom SNAND driver and move to backports directory
Add patch to fix Qcom SNAND driver and move the SNAND patches to backports directory as they are shared between qualcommax and qualcommbe target.

Fixes: #17897

---

@robimarko
@hnyman can you also test? Works on my linksys ipq60xx device
Also I already flagged the fix as backport as it's really straightforward
closed 2025-02-08 by openwrt-bot