openwrt/openwrt #17899: Cannot build 24.10.0 (r28427-6df0e3d02a)  for ZTE MF287Plus (ipq40xx/generic) on firmware-selector.openwrt.org

Lyapsus, 2025-02-07:
### Describe the bug

```
Error: Impossible package selection
STDERR

Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/ipq40xx/generic/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/openwrt_core
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/ipq40xx/generic/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/arm_cortex-a7_neon-vfpv4/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/openwrt_base
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/arm_cortex-a7_neon-vfpv4/base/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/ipq40xx/generic/kmods/6.6.73-1-60aeaf7e722ca0f86e06f61157755da3/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/openwrt_kmods
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/ipq40xx/generic/kmods/6.6.73-1-60aeaf7e722ca0f86e06f61157755da3/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/arm_cortex-a7_neon-vfpv4/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/openwrt_luci
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/arm_cortex-a7_neon-vfpv4/luci/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/arm_cortex-a7_neon-vfpv4/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/openwrt_packages
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/arm_cortex-a7_neon-vfpv4/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/arm_cortex-a7_neon-vfpv4/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/openwrt_routing
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/arm_cortex-a7_neon-vfpv4/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/arm_cortex-a7_neon-vfpv4/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/openwrt_telephony
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/arm_cortex-a7_neon-vfpv4/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_pkg: Checksum or size mismatch for package rpcd-mod-luci. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package luci.
make[2]: *** [Makefile:234: package_install] Error 255
make[1]: *** [Makefile:171: _call_manifest] Error 2
make: *** [Makefile:349: manifest] Error 2
```

### OpenWrt version

24.10.0

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

ipq40xx/generic

### Device

ZTE MF287Plus

### Image kind

Self-built image

### Steps to reproduce

Request build instead of downloading ready-made one, no need to edit packages

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

cybern0id, 2025-02-07:
I've also just encountered this (except for AVM Fritz!box 7530 - same chipset).

Lyapsus, 2025-02-07:
Well, now it is successful for some reason. @cybern0id, might worth it to try again

cybern0id, 2025-02-07:
Indeed! Thanks for the heads up. Works for me now also.

closed 2025-02-08 by Lyapsus