openwrt/openwrt #17901: [24.10.0] dropbear no longer offers ed25519

philfry, 2025-02-07:
### Describe the bug

I just upgraded from 23.05.5 to 24.10.0 and I found that the shipped dropbear no longer supports ed25519:
```bash
# dropbear -V
Dropbear v2024.86

# dropbear -h
Dropbear server v2024.86 https://matt.ucc.asn.au/dropbear/dropbear.html
Usage: dropbear [options]
-b bannerfile	Display the contents of bannerfile before user login
		(default: none)
-r keyfile      Specify hostkeys (repeatable)
		defaults: 
		- rsa /etc/dropbear/dropbear_rsa_host_key
-R		Create hostkeys as required
[…]
```

while on 23.05.5:
```bash
# dropbear -V
Dropbear v2022.82

# dropbear -h
Dropbear server v2022.82 https://matt.ucc.asn.au/dropbear/dropbear.html
Usage: dropbear [options]
-b bannerfile	Display the contents of bannerfile before user login
		(default: none)
-r keyfile      Specify hostkeys (repeatable)
		defaults: 
		- rsa /etc/dropbear/dropbear_rsa_host_key
		- ed25519 /etc/dropbear/dropbear_ed25519_host_key
-R		Create hostkeys as required
[…]
```

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

ramips/mt76x8

### Device

TP-Link RE200 v2

### Image kind

Official downloaded image

### Steps to reproduce

run `dropbear -h` or `dropbearkey -h`

### Actual behaviour

only rsa key is supported

### Expected behaviour

rsa and ed25519 keys are supported

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

srobies, 2025-02-08:
Seems to be intended according to #17510.

philfry, 2025-02-08:
Thanks, I missed that one.
Unfortunately, openssh-server is not an option because the device in question is too low on space…

muxator, 2025-03-07:
Same problem here, `ramips/mt76x8`, `mipsel_24kc`, Gl.iNet GL-MT300N v2.

It's a shame, since ed25519 was working on 23.05, and regaining ssh access involved bringing up a serial console.
I'll try a custom build.

Neustradamus, 2025-04-08:
To follow this ticket.

fermino, 2025-04-15:
Would it be appropiate to add a warning in the dropbear LuCI page?

orazioedoardo, 2025-05-18:
Note that this also applies to the host key, which means you may need to connect using `ssh -o HostKeyAlgorithms=ssh-rsa ...` and you may delete `/etc/dropbear/dropbear_ed25519_host_key` if you were upgrading from a previous version.

npenin, 2025-10-08:
I have just realized my GL.iNet GL-MT6000 does not have it either despite its 8GiB of storage and 1GiB of RAM. Is it really considered a low mem device ?

closed 2025-10-19 by hauke

marek22k, 2025-11-05:
I have the same problem. Am I correct in understanding that Dropbear would have become too large (for these small devices) with support for RSA and ED25519, and that is why the decision was made to drop ED25519?
Wouldn't it make more sense to remove RSA instead of ED25519? Or was this not done for traditional reasons? (RSA is older than ED25519).

marek22k, 2025-11-09:
RSA is not an option for me. I want a key with a key length >= 4096, and that takes several seconds to log in. That's unacceptable with [Ansible](https://github.com/zeerayne/ansible-openwrt).

mtelka, 2025-11-09:
Here is my workaround (run before `make image`):
```
cd packages
wget https://downloads.openwrt.org/releases/24.10.4/targets/lantiq/xrx200/packages/dropbear_2024.86-r1_mips_24kc.ipk
cd ..
```
As you can easily check the difference is just ca 6kB (when compared to https://downloads.openwrt.org/releases/24.10.4/targets/lantiq/xrx200_legacy/packages/dropbear_2024.86-r1_mips_24kc.ipk).