openwrt/openwrt #17910: MT7621 bridged LAN - eth0 cannot be added for VLAN configuration

rsterz, 2025-02-08:
### Describe the bug

If adding eth0 on MT7621 devices (tested with Netgear R6220 and Cudy WR1300v2) to a bridged LAN (eg. lan1-4, wan) the bridged ports go permanently offline after rebooting the device. This happens with release 24.10.0 .The same configuration worked on 23.05.5 without problems.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

ramips/mt7621

### Device

Cudy WR1300 v2

### Image kind

Official downloaded image

### Steps to reproduce

Add eth0 to network configuration and reboot device afterwards.

```
config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'eth0'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'
        list ports 'wan'
```

### Actual behaviour

All LAN ports and WAN port are offline after reboot. LEDs for ports are not flashing and device is not reachable via network on any port.

### Expected behaviour

eth0 should be added to the bridged lan in order to be able to configure VLANs.

### Additional info

The ethernet ports only go offline after reboot. Reloading the configuration at runtime doesn't take the port offline immediately.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

LynxChaus, 2025-02-08:
eth0 on MT7621 devices - is DSA-master device. You must not touch it at all.

rsterz, 2025-02-08:
I used eth0 (CPU) in 23.05.5 as described in the wiki for VLAN configuration and it worked as documented:
https://openwrt.org/docs/guide-user/network/vlan/switch_configuration

How should a VLAN setup (I use ID 1 untagged and ID 100 tagged) look like with 24.10.0 please?

This is my config running 23.05.5:

/etc/config/network (only relevant parts):

```
config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'eth0'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'
        list ports 'wan'

config bridge-vlan
        option device 'br-lan'
        option vlan '1'
        list ports 'eth0:t'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'
        list ports 'wan'

config bridge-vlan
        option device 'br-lan'
        option vlan '100'
        list ports 'eth0:t'
        list ports 'wan:t'

config interface 'lan'
        option proto 'static'
        option netmask '255.255.255.0'
        option ip6assign '60'
        option gateway '192.168.178.1'
        list dns '192.168.178.1'
        option ipaddr '192.168.178.4'
        option device 'br-lan.1'

config interface 'languest'
        option proto 'static'
        option device 'br-lan.100'
        option ipaddr '192.168.179.4'
        option netmask '255.255.255.0'
        option defaultroute '0'
```

rsterz, 2025-02-08:
I removed eth0 from my config now. And VLANs are still working. Seems to be my mistake since 2018 to use eth0 as part of the bridge. Maybe I missunderstood the concept of eth0.

Upgrade to 24.10.0 without eth0 works and the VLANs are working smoothly. Thank you for the hint.

closed 2025-02-08 by rsterz