openwrt/openwrt #17911: Bricked ASUS TUF AX4200 with OpenWrt 24.10.0

andreabravetti, 2025-02-08:
### Describe the bug

Don't if it is the right place to report it but OpenWrt 24.10.0 stable bricks the ASUS TUF AX4200.

Tried many times with different method but after 24.10.0 sysupgrade the router is bricked.

Even ASUS Firmware Restoration fails and the only way to recover is asus_tuf-ax4200-ubi-cleaner.trx.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

mediatek/filogic

### Device

ax4200

### Image kind

Official downloaded image

### Steps to reproduce

Install 24.10.0, from stock or from 23.05.5, with any possible method, and it is bricked until ubi cleaner trx in rescue mode.

### Actual behaviour

bricked.

dead on the network, it seems in reboot loop.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

andreabravetti, 2025-02-08:
Please note: the 24.10.0 release for the ax4200 should be removed ASAP from [downloads.openwrt.org](https://downloads.openwrt.org/releases/24.10.0/targets/mediatek/filogic/openwrt-24.10.0-mediatek-filogic-asus_tuf-ax4200-squashfs-sysupgrade.bin)

namiltd, 2025-02-08:
A serial console log would be very helpful in solving this problem.

CharlesMengCA, 2025-02-08:
https://forum.openwrt.org/t/asus-ax4200-bootloop-after-upgraded-from-23-05-4-to-snapshot-build/205261

andreabravetti, 2025-02-09:
> A serial console log would be very helpful in solving this problem.

```
root@sibilla:~# tio -b 115200 /dev/ttyUSB0
[10:13:36.903] tio 3.8
[10:13:36.903] Press ctrl-t q to quit
[10:13:36.910] Connected to /dev/ttyUSB0

F0: 102B 0000
FA: 1040 0000
FA: 1040 0000 [0200]
F9: 0000 0000
V0: 0000 0000 [0001]
00: 0000 0000
BP: 2400 0041 [0000]
G0: 1190 0000
EC: 0000 0000 [1000]
T0: 0000 0209 [010F]
Jump to BL

NOTICE:  BL2: v2.7(release):
NOTICE:  BL2: Built : 11:12:22, May  8 2024
NOTICE:  WDT status: 0x0
NOTICE:  ---- cold boot
NOTICE:  WDT: disabled
NOTICE:  CPU: MT7986 (2000MHz)
NOTICE:  EMI: Using DDR3 settings
NOTICE:  EMI: Detected DRAM size: 512MB
NOTICE:  EMI: complex R/W mem test passed
NOTICE:  SPI_NAND parses attributes from parameter page.
NOTICE:  SPI_NAND Detected ID 0xef
NOTICE:  Page size 2048, Block size 131072, size 268435456
NOTICE:  BL2: Booting BL31
NOTICE:  BL31: v2.7(release):
NOTICE:  BL31: Built : 11:12:28, May  8 2024

U-Boot 2022.10 (May 08 2024 - 13:49:58 +0800)
TUF-AX4200 bootloader version: 2.0.0.5

CPU:   MediaTek MT7986
Model: mt7986-rfb
DRAM:  512 MiB
Core:  50 devices, 18 uclasses, devicetree: separate
MMC:   mmc@11230000: 0
Loading Environment from MTD... 
SPINAND raw ID: EF AA 22 
spi-nand: spi_nand spi_nand@1: Winbond W25N02KV SPI NAND was found.
spi-nand: spi_nand spi_nand@1: 256 MiB, block size: 128 KiB, page size: 2048, OOB size: 128, ECC strength 8/512, custom ECC status.
*** Warning - bad CRC, using default environment

In:    serial@11002000
Out:   serial@11002000
Err:   serial@11002000
ASUS TUF-AX4200 gpio/led init
Net:   
Warning: ethernet1@15100000 (eth0) using random MAC address - e6:69:91:d8:fc:0b
eth0: ethernet1@15100000
Warning: ethernet0@15100000 (eth1) using random MAC address - 4e:65:d0:6b:8b:ff
, eth1: ethernet0@15100000
ubi0: attaching mtd4
ubi0: scanning is finished
ubi0: attached mtd4 (name "UBI_DEV", size 252 MiB)
ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
ubi0: good PEBs: 2016, bad PEBs: 0, corrupted PEBs: 0
ubi0: user volume: 8, internal volumes: 1, max. volumes count: 128
ubi0: max/mean erase counter: 7/4, WL threshold: 4096, image sequence number: 1714910448
ubi0: available PEBs: 2, total reserved PEBs: 2014, PEBs reserved for bad PEB handling: 40
UBI:   vol_id  reserved_pebs  alignment  data_pad  vol_type  usable_leb_size  used_ebs  used_bytes  last_eb_bytes  corrupted  upd_marker  name_len  name
[... 237 lines cut ...]
[    1.656427] clk: Disabling unused clocks
[    1.660975] /dev/root: Can't open blockdev
[    1.665065] VFS: Cannot open root device "" or unknown-block(0,0): error -6
[    1.672011] Please append a correct "root=" boot option; here are the available partitions:
[    1.680352] 1f00            4096 mtdblock0 
[    1.680356]  (driver?)
[    1.686866] 1f01          258048 mtdblock1 
[    1.686868]  (driver?)
[    1.693380] List of all bdev filesystems:
[    1.697371]  ext3
[    1.697372]  ext2
[    1.699284]  ext4
[    1.701198]  squashfs
[    1.703109]  f2fs
[    1.705366] 
[    1.708754] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
[    1.716992] SMP: stopping secondary CPUs
[    1.720899] Kernel Offset: disabled
[    1.724370] CPU features: 0x0,00000000,00000000,1000400b
[    1.729663] Memory Limit: none
[    1.734109] pstore: backend (ramoops) writing error (-28)
[    1.739489] Rebooting in 1 seconds..

ENDLESS LOOP, RESTART HERE......
```

namiltd, 2025-02-09:
Could you please provide the console log for version 23.05.5?

andreabravetti, 2025-02-09:
> https://forum.openwrt.org/t/asus-ax4200-bootloop-after-upgraded-from-23-05-4-to-snapshot-build/205261

Thank you, it seems to be the same problem.

What is not clear to me is if I need some uboot upgrade before installing 24.10.0, however I can't spend time on it now.

For now I will stay on 23.05.5 that works flawlessly.

andreabravetti, 2025-02-09:
> Could you please provide additional the console log for version 23.05.5?

Yep.

This is after cleaning it with ```asus_tuf-ax4200-ubi-cleaner.trx``` uploaded with ``` ASUS Firmware Restoration```, and then installing ```openwrt-23_tuf-ax4200-initramfs.trx``` and ```openwrt-23.05.5-mediatek-filogic-asus_tuf-ax4200-squashfs-sysupgrade.bin```, giving in the end a working 23.05.5.

```
root@sibilla:~# tio -b 115200 /dev/ttyUSB0
[10:47:45.769] tio 3.8
[10:47:45.769] Press ctrl-t q to quit
[10:47:45.777] Connected to /dev/ttyUSB0

F0: 102B 0000
FA: 1040 0000
FA: 1040 0000 [0200]
F9: 0000 0000
V0: 0000 0000 [0001]
00: 0000 0000
BP: 2400 0041 [0000]
G0: 1190 0000
EC: 0000 0000 [1000]
T0: 0000 0209 [010F]
Jump to BL

NOTICE:  BL2: v2.7(release):
NOTICE:  BL2: Built : 11:12:22, May  8 2024
NOTICE:  WDT status: 0x0
NOTICE:  ---- cold boot
NOTICE:  WDT: disabled
NOTICE:  CPU: MT7986 (2000MHz)
NOTICE:  EMI: Using DDR3 settings
NOTICE:  EMI: Detected DRAM size: 512MB
NOTICE:  EMI: complex R/W mem test passed
NOTICE:  SPI_NAND parses attributes from parameter page.
NOTICE:  SPI_NAND Detected ID 0xef
NOTICE:  Page size 2048, Block size 131072, size 268435456
NOTICE:  BL2: Booting BL31
NOTICE:  BL31: v2.7(release):
NOTICE:  BL31: Built : 11:12:28, May  8 2024

U-Boot 2022.10 (May 08 2024 - 13:49:58 +0800)
TUF-AX4200 bootloader version: 2.0.0.5

CPU:   MediaTek MT7986
Model: mt7986-rfb
DRAM:  512 MiB
Core:  50 devices, 18 uclasses, devicetree: separate
MMC:   mmc@11230000: 0
Loading Environment from MTD... 
SPINAND raw ID: EF AA 22 
spi-nand: spi_nand spi_nand@1: Winbond W25N02KV SPI NAND was found.
spi-nand: spi_nand spi_nand@1: 256 MiB, block size: 128 KiB, page size: 2048, OOB size: 128, ECC strength 8/512, custom ECC status.
*** Warning - bad CRC, using default environment

In:    serial@11002000
Out:   serial@11002000
Err:   serial@11002000
ASUS TUF-AX4200 gpio/led init
Net:   
Warning: ethernet1@15100000 (eth0) using random MAC address - 12:ab:e3:a8:09:25
eth0: ethernet1@15100000
Warning: ethernet0@15100000 (eth1) using random MAC address - 42:d5:e8:86:01:12
, eth1: ethernet0@15100000
ubi0: attaching mtd4
ubi0: scanning is finished
ubi0: attached mtd4 (name "UBI_DEV", size 252 MiB)
ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
ubi0: good PEBs: 2016, bad PEBs: 0, corrupted PEBs: 0
[... 326 lines cut ...]
[   18.274877] mt7530-mdio mdio-bus:1f lan1: configuring for phy/gmii link mode
[   18.285124] br-lan: port 1(lan1) entered blocking state
[   18.290417] br-lan: port 1(lan1) entered disabled state
[   18.297189] device lan1 entered promiscuous mode
[   18.313186] mt7530-mdio mdio-bus:1f lan2: configuring for phy/gmii link mode
[   18.323083] br-lan: port 2(lan2) entered blocking state
[   18.328392] br-lan: port 2(lan2) entered disabled state
[   18.335325] device lan2 entered promiscuous mode
[   18.346800] mt7530-mdio mdio-bus:1f lan3: configuring for phy/gmii link mode
[   18.357109] br-lan: port 3(lan3) entered blocking state
[   18.362343] br-lan: port 3(lan3) entered disabled state
[   18.369514] device lan3 entered promiscuous mode
[   18.380492] mt7530-mdio mdio-bus:1f lan4: configuring for phy/gmii link mode
[   18.390336] br-lan: port 4(lan4) entered blocking state
[   18.395568] br-lan: port 4(lan4) entered disabled state
[   18.402872] device lan4 entered promiscuous mode
[   18.427491] mtk_soc_eth 15100000.ethernet eth1: PHY [mdio-bus:06] driver [Maxlinear Ethernet GPY211C] (irq=POLL)
[   18.437698] mtk_soc_eth 15100000.ethernet eth1: configuring for phy/2500base-x link mode
[   21.371341] mt7530-mdio mdio-bus:1f lan1: Link is Up - 1Gbps/Full - flow control rx/tx
[   21.371360] br-lan: port 1(lan1) entered blocking state
[   21.384468] br-lan: port 1(lan1) entered forwarding state
[   21.390311] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready

# AT THIS POINT I CAN LOGIN TO THE ROUTER....
```

andreabravetti, 2025-02-09:
In the end it seems to me it was tested only with oem bootloader version **1.0.0.1** and NOT with version **2.0.0.5**.

If someone else confirms this is the case I can update the title to refer to the specific bootloader version.

andreabravetti, 2025-02-10:
> So if you want a fresh, corrected version, download it from here: https://github.com/openwrt-xiaomi/builder These firmwares are builded from the original OpenWrt sources (plus patches) on GitHub servers. There is no point in waiting for months (or years) for fixes in the official OpenWrt repository.

Hi @remittor, thank you for your work on this topic, I will give it a try.

Yes, but I can also patch and build it my self now, and my board is quite old so I should be able to safely downgrade the bootloader too, this is not the problem I would like to highlight.

What I would like to highlight is that the official build is fundamentally flawed and the number of (soft)bricked devices is growing fast, for sure, given the nature of the problem.

Is it possible for a well-known problem like this to be ignored for months?

hauke, 2025-02-10:
This commit adds a partition for the bootloader to remove, see 25c9ebc6d405bf1bcf83fa4a59b315df245be0c0

The `Bootloader` partition with capital `B` is not defined in the device tree. Maybe the boot loader is replacing the partition here and Linux is not detecting the UBI partition any more. 
```
[    0.916526] 0x000000000000-0x000000400000 : "Bootloader"
```

Could you please flash a initramfs FS partition and then check the dts properties in the `/sys/` file system. Compare the properties with what we actually defined in the device tree file. 

@dangowrt Could you help here please.

dangowrt, 2025-02-10:
Sadly the only good solution I can come up with is to prevent the bootloader from wiping the MTD partitions, ie. carry a downstream hack patch:
https://git.openwrt.org/?p=openwrt/staging/dangole.git;a=commit;h=420894a2538eb834d084eb2bc1cbf95778238014

And replace the compatible `spi-nand` in affected devices' DTS with `u-boot-dont-touch-spi-nand`.

I'll prepare a PR for testing this approach.

andreabravetti, 2025-02-11:
@dangowrt @hauke thank you for your work on this topic.

Unfortunately it's currently the only router I have and I need it to work. I can do any test to help fix this, but only from late Friday evening, after 6pm, Europe/Rome time, and during the weekend. In the meantime, if you have a list of images to test, procedures or code to compile in advance to be ready for the weekend, feel free to write to me.

andreabravetti, 2025-02-15:
@dangowrt I compiled openwrt from the **v24.10.0** tag with the addition of your patch and the configuration taken from [config.buildinfo](https://downloads.openwrt.org/releases/24.10.0/targets/mediatek/filogic/config.buildinfo). Installed the sysupgrade image via Luci and it works! No bootloop and everything seems to work fine.

andrey-le, 2025-02-15:
Same result for me.
Here is two UART logs - for pure 24.10 and patched one.

[ax4200-pure 24.10.txt](https://github.com/user-attachments/files/18809920/ax4200-pure.24.10.txt)
[ax4200-patched kernet 24.10.txt](https://github.com/user-attachments/files/18809919/ax4200-patched.kernet.24.10.txt)

sylarm27, 2025-02-19:
Hi all,
Is it possible to get/download this new working build?
Or in the next release 24.10.1 for example it will included?
I'm afraid I cannot build myself a release :(
Thanks

andreabravetti, 2025-02-20:
> Hi all, Is it possible to get/download this new working build? Or in the next release 24.10.1 for example it will included? I'm afraid I cannot build myself a release :( Thanks

Hi @sylarm27, I'm not involved in any release process but I think I can answer you:

The latest [SNAPSHOT](https://firmware-selector.openwrt.org/?version=SNAPSHOT) release should work but you will have some other patches and it may fail for other reasons.

If you can't compile it yourself the best thing to do is wait for 24.10.1 which hopefully could be out next month.

sylarm27, 2025-02-20:
> Hi [@sylarm27](https://github.com/sylarm27), I'm not involved in any release process but I think I can answer you:
>
> The latest [SNAPSHOT](https://firmware-selector.openwrt.org/?version=SNAPSHOT) [...]

Thank you.
We hope that release " 24.10.1" will be smoother.
I will make you one more question, do you know if "Attended Sysupgrade" will/should work to jump from 23.5.5 to 24.10.x?

andreabravetti, 2025-02-23:
> Thank you. We hope that release " 24.10.1" will be smoother. I will make you one more question, do you know if "Attended Sysupgrade" will/should work to jump from 23.5.5 to 24.10.x?

@sylarm27  please read here https://forum.openwrt.org/t/openwrt-24-10-0-first-stable-release/223845

sylarm27, 2025-02-23:
> [@sylarm27](https://github.com/sylarm27) please read here https://forum.openwrt.org/t/openwrt-24-10-0-first-stable-release/223845

Means attended sysupg is not supported...

Nyslay, 2025-02-23:
Hello, I also brocked this router via installing current version of openWRT. Where can I find asus_tuf-ax4200-ubi-cleaner.trx to debrick it.

maartenlangeveld, 2025-02-25:
Just bricked my ASUS 4200 TUF AX4200 by upgrade from 23.05 to 24.10. Discovering the problem has been known for two weeks (!)

Why for god's sake has the update not been take off the openwrt site??

Very rrresponsible behaviour from the devs, imo. Sorry to say but I am really pissed. Absolutely not because things go wrong, that happens, but because the update is still there !

andreabravetti, 2025-02-25:
> Just bricked my ASUS 4200 TUF AX4200 by upgrade from 23.05 to 24.10. Discovering the problem has been known for two weeks (!)

@maartenlangeveld sorry to hear this but don't worry: it can be fixed.

> Very irresponsible behaviour from the devs, imo. Sorry to say but I am really pissed. Absolutely not because things go wrong, that happens, but because the update is still there !

I agree, and I wrote it [here](https://github.com/openwrt/openwrt/issues/17911#issuecomment-2645816909)

@dangowrt is there any way to remove 24.10.0 image from [downloads.openwrt.org](https://downloads.openwrt.org/releases/24.10.0/targets/mediatek/filogic/openwrt-24.10.0-mediatek-filogic-asus_tuf-ax4200-squashfs-sysupgrade.bin), and also previous affected snapshots? Do you know who can do that?

maartenlangeveld, 2025-02-25:
> [@maartenlangeveld](https://github.com/maartenlangeveld) sorry to hear this but don't worry: it can be fixed.
>
>
> I agree, and I wrote it [...]

You did right, but for some reason it's still out there. Unbeleivable. Luckily I have an old TP-Link C7 as a spare/backup router which I now have temporarily installed and connected.

andrey-le, 2025-02-25:
As well bootloop warning information are to be published there:

https://openwrt.org/toh/asus/tuf-ax6000
https://openwrt.org/toh/asus/tuf-ax6000
https://openwrt.org/toh/asus/rt-ax59u

andrey-le, 2025-02-25:
> You did right, but for some reason it's still out there. Unbeleivable. Luckily I have an old TP-Link C7 as a spare/backup router which I now have temporarily installed and connected.

You can rescue your router without opening.
https://forum.openwrt.org/t/asus-ax4200-bootloop-after-upgraded-from-23-05-4-to-snapshot-build/205261

maartenlangeveld, 2025-02-25:
> You can rescue your router without opening. https://forum.openwrt.org/t/asus-ax4200-bootloop-after-upgraded-from-23-05-4-to-snapshot-build/205261

Thanks, wil try now.

sylarm27, 2025-02-26:
Hello,

I suppose the bugs are being corrected, I found these images from yesterday(25 Fev) under development "location":
https://downloads.openwrt.org/snapshots/targets/mediatek/filogic/openwrt-mediatek-filogic-asus_tuf-ax4200-squashfs-sysupgrade.bin
Someone have tested?
Regards

andreabravetti, 2025-02-26:
> I suppose the bugs are being corrected, I found these images from yesterday(25 Fev) under development "location": https://downloads.openwrt.org/snapshots/targets/mediatek/filogic/openwrt-mediatek-filogic-asus_tuf-ax4200-squashfs-sysupgrade.bin Someone have tested? Regards

The bug is fixed in SNAPSHOT.

Don't know if this issue should be closed as resolved or if is better to wait for a 24.10.1 release.

sylarm27, 2025-02-26:
> The bug is fixed in SNAPSHOT.
>
> Don't know if this issue should be closed as resolved or if is better to wait for a 24.10.1 release.

I don't know but.... wait for new release

maartenlangeveld, 2025-02-26:
> The bug is fixed in SNAPSHOT.
>
> Don't know if this issue should be closed as resolved or if is better to wait for a 24.10.1 release.

No it's NOT. Snapshot still bricks my TUF AX4200. Just tried.
And the bricked 24.10 version is still available for download (not even a warning).

UNBELEIVABLE. 

Is it too much too pull off the faulty kernel or to warn people??

maartenlangeveld, 2025-02-27:
@andreabravetti 
Will you help me? Can you tell me which snapshot build (perhaps download link?) you exactly used? 
Which OpenWrt kernel file and which sysupgrade file did you use?

I have many times restored my AX4200 to default and uploaded OpenWrt via the web interface but sometimes after uploading the kernel but always after uploading the sysupgrade file (via OpenWrt webgui) the machine gets bricked.

Thanks,
Maarten

andreabravetti, 2025-02-27:
@maartenlangeveld I whould like to help but I'm not an OpenWrt developer or expert of any kind.

Any snapshot after 4387de9 **should** work, and "should" mean that the fixes from c48afca and 4387de9 are in, but I do not use prebuilt snapshots and I don't know if any other changes break something, also I do not know which configuration is used for snapshots.

Right now I'm using a self built image from branch openwrt-24.10 up to 3abbc15 and it works fine. Before I used a self built image from tag v24.10.0 with the patch from dangowrt (c48afca and 4387de9) and it also worked fine. Honestly if you can't compile it yourself I would recommend using 23.05.5 until the stable 24.10.1 is released.

andreabravetti, 2025-02-27:
I'm closing this bug because it was fixed bye @dangowrt with c48afca and 4387de9.

For documentation, I added a note here:

https://openwrt.org/toh/asus/tuf-ax4200
https://openwrt.org/toh/asus/tuf-ax6000
https://openwrt.org/toh/asus/rt-ax59u

The note is:

**Please note that OpenWrt 24.10.0 works only on devices with U-Boot 2022.04-rc1.**
**If you have a recent U-Boot >= 2022.10 installing OpenWrt 24.10.0 will brick your device.**
**If unsure DO NOT INSTALL OpenWrt 24.10.0 and wait for OpenWrt 24.10.1.**

closed 2025-02-27 by andreabravetti

maartenlangeveld, 2025-02-27:
> [@maartenlangeveld](https://github.com/maartenlangeveld) I whould like to help but I'm not an OpenWrt developer or expert of any kind.
> 
> Any snapshot after [4387de9](https://github.com/openwrt/openwrt/commit/4387de94454913aa79bb88971eabcf1bfca78141) **should** work, and "should" mean that the fixes from [c48afca](https://github.com/openwrt/openwrt/commit/c48afca32ca875c4176b4c41be94335e61556783) and [4387de9](https://github.com/openwrt/openwrt/commit/4387de94454913aa79bb88971eabcf1bfca78141) are in, but I do not use prebuilt snapshots and I don't know if any other changes break something, also I do not know which configuration is used for snapshots.
> 
> Right now I'm using a self built image from branch openwrt-24.10 up to [3abbc15](https://github.com/openwrt/openwrt/commit/3abbc154546042af5efe6d299e3e40133dfb3d7f) and it works fine. Before I used a self built image from tag v24.10.0 manually applying the patch from dangowrt ([c48afca](https://github.com/openwrt/openwrt/commit/c48afca32ca875c4176b4c41be94335e61556783) and [4387de9](https://github.com/openwrt/openwrt/commit/4387de94454913aa79bb88971eabcf1bfca78141)) and it also worked fine. Honestly if you can't compile it yourself I would recommend using 23.05.5 until the stable 24.10.1 is released.

Thanks a lot! I will wait till the next release. I do not compile.
Kind regards, Maarten

lorand-horvath, 2025-04-14:
24.10.1 is finally building...

nasm27, 2025-04-14:
hello,
Is 24.10.1 available or not? Because at attended sysupgrade I saw the release but once trying to build it, if fails...  (Unsupported version: 24.10.1). We just have to wait a little bit more?

lorand-horvath, 2025-04-14:
@nasm27 Wait a few more hours (tomorrow to be sure), the firmware-selector returns the same error.

nasm27, 2025-04-15:
> [@nasm27](https://github.com/nasm27) Wait a few more hours (tomorrow to be sure), the firmware-selector returns the same error.

Or a few more days.... :)

Encymos, 2025-05-03:
May, 3th, just bricked my 4200 / stable 23.05 opewrt with current official 24.10 release
I guess it's too early
or i am really unfortunate)

andreabravetti, 2025-05-03:
> May, 3th, just bricked my 4200 / stable 23.05 opewrt with current official 24.10 release I guess it's too early or i am really unfortunate)

@Encymos which 24.10? 24.10.0 or 24.10.1?

merrick1901, 2025-05-07:
Unfortunately I've also just bricked my device by trying to flash 24.10.0. @andreabravetti You mention being able to unbrick the device using the rescue tool, but I can't seem to get my router to boot into rescue mode. I get a continuous blinking pattern which consists of the first led, the middle four and the second two respectively. Any advice?

Encymos, 2025-05-07:
@andreabravetti Just checked, looks like it was 24.10.0 , on official openwrt router page it is listed as a current release. (https://openwrt.org/toh/asus/tuf-ax4200)
I managed to reflash it to remittor 24.10.1 version though, works fine
Sorry for panic)

Encymos, 2025-05-07:
> Unfortunately I've also just bricked my device by trying to flash 24.10.0. [@andreabravetti](https://github.com/andreabravetti) You mention being able to unbrick the device using the rescue tool, [...]
@merrick1901 As long as i did it recently, i'll try to help - i had same flashing pattern. Just press reset button for around 10 seconds with a toothpick (it's a small hole under wps button) while it flashing like that. Then your flashing pattern should change for slow power led blinking, in that stage recovery tool is able to recognize the router. So next step is flashing ubi cleaner, and after that i directly flashed correct version of openwrt (10.1)

merrick1901, 2025-05-08:
@Encymos Much obliged :). It seems it was a classic PEBKAC issue and I mistook the wps button for the reset button 🤦