openwrt/openwrt #17913: sysupgrade: Not always preserving all files listed in /etc/sysupgrade.conf

VorpalBlade, 2025-02-08:
### Describe the bug

When upgrading from 23.05.5 to 24.10 all files were preserved (upgrading from pre-built to pre-built image). I then went ahead and used `owut` to consolidate installed packages from overlay into the base image, and a bunch of files were not kept.

My `/etc/sysupgrade.conf`:

```
## This file contains files and directories that should
## be preserved during an upgrade.

# /etc/example.conf
# /etc/openvpn/

/etc/.git
/etc/.gitignore
/root/.config
/root/.gitconfig
/root/bin
/root/.profile
```

Only the default files (`/etc/config`, etc) and the files in `/root/bin` were preserved on this 24.10 -> 24.10 upgrade. I mentioned this in [the owut forum thread](https://forum.openwrt.org/t/owut-openwrt-upgrade-tool/200035/536?u=vorpal), and the owut author [indicated](https://forum.openwrt.org/t/owut-openwrt-upgrade-tool/200035/538?u=vorpal) that this would be a sysupgrade issue rather than owut issue. Additionally they said:

>Now that you mention it, I think that happened to me, too. I did an -rc7 to .0 upgrade and when I later edited a file my `/root/.vimrc` was gone. This is all `sysupgrade`'s thing, but when I do `sysupgrade -l` it lists out exactly what I expect, including `/root/.vimrc`. Oh, and I see that `.config` got restore, so ???

So it seems I'm possibly not the only one running into this.

I do not think I ran into the files to preserve being too large to fit into RAM, the GL-MT6000 is beefy with 1 GB RAM. Additionally the sum of `/etc` and `/root` is just about 840 KB. So... no that isn't it.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

mediatek/filogic

### Device

GL.iNet GL-MT6000

### Image kind

Official downloaded image

### Steps to reproduce

I have not done extensive testing to indicate how reliable this is to reproduce, I rather suspect it is not reliably reproducible.

1. Have some files and folders listed in `/etc/sysupgrade.conf`, the more the merrier (probably). Managing `/etc` in git and copying the `.git` folder across might be a good way to test, there should be a bunch of files there. Maybe they need to be "dotfiles" as they were in my case, I don't know.
2. Perform a sysupgrade keeping configuration.

### Actual behaviour

Only a subset of the configuration files you specified were actually preserved.

### Expected behaviour

It should preserve all files and directories that I listed.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

bkbartk, 2025-02-14:
I think I have the same

filippocarletti, 2025-02-23:
I tested owut to upgrade from 24.10.0-rc2 to 24.10.0.
I had 4 lines in `/etc/sysupgrade.conf`.
Some files went lost during upgrade: the last files listed by `sysupgrade -l`.
The list of lost files contained both standard files:
```
/etc/sysctl.conf
/etc/sysupgrade.conf
/etc/uhttpd.crt
/etc/uhttpd.key
```
and my files from my additions to sysupgrade.conf:
```
/etc/usb-mode.json
/etc/vnstat.conf
/etc/vnstat.db
```
The first missing file is listed at line 80 of `sysupgrade -l` (bytes 1882).

bkbartk, 2025-02-25:
the label has target/mediatek
And I noticed it only fails on 1 of my 3 updated OpenWrt devices, the one containing a mediatek chip. And it also fails in the default backup.

VorpalBlade, 2025-02-25:
I have since done two owut upgrades, both have not had this issue. So it is quite random. And I don't know if it is rare or not yet. 

@bkbartk It might be mediatek specific, maybe not.

Please, those of you who post you have this issue: tell us what router and target you have. The more info you provide the better for homing in on the cause. @filippocarletti @hexsen92

filippocarletti, 2025-02-25:
I tested x86/64 on a PC Engines APU.

bkbartk, 2025-02-25:
some more info:
```
root@OpenWrt:~# ubus call system board
{
        "kernel": "6.6.73",
        "hostname": "OpenWrt",
        "system": "ARMv8 Processor rev 4",
        "model": "Zyxel EX5601-T0 ubootmod",
        "board_name": "zyxel,ex5601-t0-ubootmod",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "24.10.0",
                "revision": "r28427-6df0e3d02a",
                "target": "mediatek/filogic",
                "description": "OpenWrt 24.10.0 r28427-6df0e3d02a",
                "builddate": "1738624177"
        }
}
root@OpenWrt:~#

```

hexsen929, 2025-02-27:
[image]

pve x86_64

hexsen929, 2025-03-01:
I have compiled OpenWrt from scratch, and version 24.10 has been tested without any issues.

VorpalBlade, 2025-03-01:
> I have compiled OpenWrt from scratch, and version 24.10 has been tested without any issues.

How many tests? This seems to be a rare issue. I had only one out of 5 sysupgrades fail so far. And at least four different people have had the issue so far.

MiguVT, 2025-05-04:
Im using a fork of openwrt and is working fine and no issues on that fork, i updated like 10 times, i think it can be specific issues with some models (the fork is exactly for arm-based models, and not all only some like ax3600)

maybe i was just lucky or got patched after 2 months, if its working well now, close this issue can help people to know its just update to latest.

VorpalBlade, 2025-08-09:
This happened again to me. I recently updated with owut (from and to 24.10.1) and some files under `/etc/.git` were missing, causing git to consider the repository invalid.

I did a new clone from the server and move the `.git` from that into place of the old broken directory, which fixed it. But that shouldn't be needed.