openwrt/openwrt #17914: Problem back in 24.10: Several different UBNT devices not working, file system seems write protected #12882

dnordenberg, 2025-02-08:
### Describe the bug

Problem seems to be back in 24.10. Several different UBNT devices not working in 23.05rc1, file system seems write protected #12882

`
[    0.000000] Linux version 6.6.73 (builder@buildhost) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r28427-6df0e3d02a) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 Mon Feb  3 23:09:37 2025
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
[    0.000000] MIPS: machine is Ubiquiti NanoBeam AC Gen1 (XC)
[    0.000000] SoC: Qualcomm Atheros QCA9558 ver 1 rev 0
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 119968K/131072K available (6819K kernel code, 604K rwdata, 900K rodata, 1204K init, 233K bss, 11104K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] CPU clock: 720.000 MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 5309056796 ns
[    0.000001] sched_clock: 32 bits at 360MHz, resolution 2ns, wraps every 5965232126ns
[    0.008390] Calibrating delay loop... 358.80 BogoMIPS (lpj=1794048)
[    0.075023] pid_max: default: 32768 minimum: 301
[    0.090147] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.097947] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.115105] RCU Tasks Trace: Setting shift to 0 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=1.
[    0.130038] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.140566] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.150485] pinctrl core: initialized pinctrl subsystem
[    0.159369] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.166361] thermal_sys: Registered thermal governor 'step_wise'
[    0.168203] /ahb/apb: Fixed dependency cycle(s) with /ahb/apb/interrupt-controller@18060010
[    0.195928] clocksource: Switched to clocksource MIPS
[    0.212356] NET: Registered PF_INET protocol family
[    0.217839] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.227068] tcp_listen_portaddr_hash hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.236142] Table-perturb hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.244395] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.252568] TCP bind hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.260115] TCP: Hash tables configured (established 1024 bind 1024)
[    0.267612] MPTCP token hash table entries: 512 (order: 0, 6144 bytes, linear)
[    0.275548] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.282572] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.291004] NET: Registered PF_UNIX/PF_LOCAL protocol family
[    0.297166] PCI: CLS 0 bytes, default 32
[    0.306228] workingset: timestamp_bits=14 max_order=15 bucket_order=1
[    0.314800] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.321065] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[    0.339283] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 251)
[    0.350755] pinctrl-single 1804002c.pinmux: 544 pins, size 68
[    0.358257] Serial: 8250/16550 driver, 16 ports, IRQ sharing enabled
[    0.370463] printk: console [ttyS0] disabled
[    0.375477] 18020000.uart: ttyS0 at MMIO 0x18020000 (irq = 11, base_baud = 2500000) is a 16550A
[    0.384837] printk: console [ttyS0] enabled
[    0.393749] printk: bootconsole [early0] disabled
[    0.438259] spi-nor spi0.0: mx25l12805d (16384 Kbytes)
[    0.443586] 5 fixed-partitions partitions found on MTD device spi0.0
[... 343 lines cut ...]
### Image kind

Official downloaded image

### Steps to reproduce

Change something, reboot, changes lost.

### Actual behaviour

Changes lost

### Expected behaviour

Changes saved.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

stargieg, 2025-02-14:
I have the same Problem
OpenWrt version: r28427-6df0e3d02a
OpenWrt release: 24.10.0
OpenWrt target/subtarget: ath79/generic
Device: Ubiquiti LiteBeam AC Gen2

sysupgrade -n not working any more.
No upgrade or downgrade posable

```
Tue Feb  4 00:09:55 2025 kern.debug kernel: [   25.876556] ath: EEPROM indicates default country code should be used
Tue Feb  4 00:09:55 2025 kern.debug kernel: [   25.876565] ath: doing EEPROM country->regdmn map search
Tue Feb  4 00:09:55 2025 kern.debug kernel: [   25.876586] ath: country maps to regdmn code: 0x3a
Tue Feb  4 00:09:55 2025 kern.debug kernel: [   25.876596] ath: Country alpha2 being used: US
Tue Feb  4 00:09:55 2025 kern.debug kernel: [   25.876606] ath: Regpair used: 0x3a
Tue Feb  4 00:09:55 2025 kern.debug kernel: [   25.955780] ath: EEPROM regdomain: 0x0
Tue Feb  4 00:09:55 2025 kern.debug kernel: [   25.955814] ath: EEPROM indicates default country code should be used
Tue Feb  4 00:09:55 2025 kern.debug kernel: [   25.955823] ath: doing EEPROM country->regdmn map search
Tue Feb  4 00:09:55 2025 kern.debug kernel: [   25.955844] ath: country maps to regdmn code: 0x3a
Tue Feb  4 00:09:55 2025 kern.debug kernel: [   25.955855] ath: Country alpha2 being used: US
Tue Feb  4 00:09:55 2025 kern.debug kernel: [   25.955865] ath: Regpair used: 0x3a
Tue Feb  4 00:09:55 2025 kern.debug kernel: [   25.970813] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
Tue Feb  4 00:09:55 2025 kern.info kernel: [   25.972265] ieee80211 phy1: Atheros AR9340 Rev:2 mem=0xd7588089, irq=14
Tue Feb  4 00:09:55 2025 user.info kernel: [   26.011123] kmodloader: done loading kernel modules from /etc/modules.d/*
Tue Feb  4 00:09:59 2025 daemon.info dnsmasq[1]: started, version 2.90 cachesize 1000
Tue Feb  4 00:09:59 2025 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-nftset no-auth no-cryptohash no-DNSSEC no-ID loop-detect inotify dumpfile
Tue Feb  4 00:09:59 2025 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Tue Feb  4 00:09:59 2025 daemon.info dnsmasq[1]: using nameserver 2003:a:36d:2410::1#53
Tue Feb  4 00:09:59 2025 daemon.info dnsmasq[1]: using nameserver 2003:a:36d:2401::1#53
Tue Feb  4 00:09:59 2025 daemon.info dnsmasq[1]: using nameserver 2003:a:36d:2400::1#53
Tue Feb  4 00:09:59 2025 daemon.info dnsmasq[1]: using nameserver 2003:a:36d:2400::1#53
Tue Feb  4 00:09:59 2025 daemon.info dnsmasq[1]: using nameserver 2606:4700:4700::64#53
Tue Feb  4 00:09:59 2025 daemon.info dnsmasq[1]: using nameserver 2606:4700:4700::6400#53
Tue Feb  4 00:09:59 2025 daemon.info dnsmasq[1]: using nameserver 2001:4860:4860::64#53
Tue Feb  4 00:09:59 2025 daemon.info dnsmasq[1]: using nameserver 2001:4860:4860::6464#53
Tue Feb  4 00:09:59 2025 daemon.info dnsmasq[1]: using only locally-known addresses for test
Tue Feb  4 00:09:59 2025 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Tue Feb  4 00:09:59 2025 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Tue Feb  4 00:09:59 2025 daemon.info dnsmasq[1]: using only locally-known addresses for local
Tue Feb  4 00:09:59 2025 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Tue Feb  4 00:09:59 2025 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Tue Feb  4 00:09:59 2025 daemon.info dnsmasq[1]: using only locally-known addresses for olsr
Tue Feb  4 00:09:59 2025 daemon.warn dnsmasq[1]: no servers found in /tmp/resolv.conf.d/resolv.conf.auto, will retry
Tue Feb  4 00:09:59 2025 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Tue Feb  4 00:09:59 2025 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 1 names
Tue Feb  4 00:10:00 2025 authpriv.info dropbear[1651]: Not backgrounding
Tue Feb  4 00:10:03 2025 daemon.notice wpa_supplicant[1706]: Successfully initialized wpa_supplicant
Tue Feb  4 00:10:04 2025 user.notice : Added device handler type: bonding
Tue Feb  4 00:10:04 2025 user.notice : Added device handler type: 8021ad
Tue Feb  4 00:10:04 2025 user.notice : Added device handler type: 8021q
Tue Feb  4 00:10:04 2025 user.notice : Added device handler type: macvlan
Tue Feb  4 00:10:04 2025 user.notice : Added device handler type: veth
Tue Feb  4 00:10:04 2025 user.notice : Added device handler type: bridge
Tue Feb  4 00:10:04 2025 user.notice : Added device handler type: Network device
Tue Feb  4 00:10:04 2025 user.notice : Added device handler type: tunnel
Tue Feb  4 00:10:08 2025 cron.err crond[1996]: crond (busybox 1.36.1) started, log level 5
Tue Feb  4 00:10:13 2025 daemon.notice netifd: Interface 'fflandhcp' is enabled
Tue Feb  4 00:10:13 2025 daemon.notice netifd: Interface 'fflandhcp' is setting up now
Tue Feb  4 00:10:13 2025 daemon.notice netifd: Interface 'fflandhcprt' is enabled
Tue Feb  4 00:10:13 2025 daemon.notice netifd: Interface 'fflandhcprt' is setting up now
Tue Feb  4 00:10:13 2025 daemon.notice netifd: Interface 'fflandhcprt' is now up
Tue Feb  4 00:10:13 2025 daemon.notice netifd: Interface 'fflandhcp' is now up
Tue Feb  4 00:10:13 2025 kern.info kernel: [   52.001372] br-fflandhcp: port 1(dummy0) entered blocking state
Tue Feb  4 00:10:13 2025 kern.info kernel: [   52.007421] br-fflandhcp: port 1(dummy0) entered disabled state
Tue Feb  4 00:10:13 2025 kern.info kernel: [   52.013560] dummy0: entered allmulticast mode
Tue Feb  4 00:10:13 2025 kern.info kernel: [   52.018356] dummy0: entered promiscuous mode
[... cut ...]
[   64.731051] jffs2: Newly-erased block contained word 0x19852003 at offset 0x00030000
[   64.753188] jffs2: Newly-erased block contained word 0x19852003 at offset 0x00020000
[   64.783062] jffs2: Newly-erased block contained word 0x19852003 at offset 0x00010000
[   64.801072] jffs2: Newly-erased block contained word 0xdeadc0de at offset 0x00000000
[   64.817147] done.
[   64.819157] jffs2: notice: (2708) jffs2_build_xattr_subsystem: complete building xattr subsystem, 0 of xdatum (0 unchecked, 0 orphan) and 0 of xref (0 dead, 0 orphan) found.
```

grizlyadams, 2025-02-25:
I have the same problem too and I can't get any tftp image to stick - I push the image (factory) and the device appears to take the image (no checksum error) but when it reboots its back on 24.10. I can't flash from LuCi or tftp.

nikito7, 2025-02-25:
> I have the same problem too and I can't get any tftp image to stick - I push the image (factory) and the device appears to take the image (no checksum error) but when it reboots its back on 24.10. I can't flash from LuCi or tftp.

tftp it's not easy. You need the correct timming to start flash.

[image]

dnordenberg, 2025-02-25:
Tftp should definitely work since it does not depend on anything previously loaded on the device, it is a pure boot loader thing.

grizlyadams, 2025-02-25:
> tftp it's not easy. You need the correct timming to start flash.
>
> [image]

I have 8.7.4 UBNT firmware installed now - it was either because I tftp'd the f/w twice in a a row without allowing it to reboot a second time after flashing (because that second reboot would just take it to 24.10 again) or it was because i use windows 11 cli tftp instead of Pumpkin.

Now it will not allow flashing to openwrt via the web ui in UBNT 8.7.4. and the patching and running fwupdate (in 8.7.4) just errors with a "invalid file". So, more websearching and advice is to downgrade tftp to UBNT v8.7.0 and once flash reboot is done, then re-flash the same f/w. Web IU give an initialization error. So, (in my past, sparse notes I tftp installed openwrt-21.02.7 factory )  - that didn't work. tftp doesn't report successful transfer and it comes back up with 8.7.0 installed (with the same web UI initialization error)

so now I'm stuck with UBNT 8.7.4 on there and the binary patch jsut fails with "Invalid version 'WA.ar934x.v8.5.3-42.OpenWrt-r24012-d8dd03c46f'

Why oh why did I believe openwrt wiki and update the firmware🙄 Nice way to lose a couple of days of my life 😖🤬

nikito7, 2025-02-25:
Tftp. Only use openwrt 23.x factory firmware to fix issue!

grizlyadams, 2025-02-25:
Ok, I got it sorted in a pretty convoluted way - well not so convoluted once it was boiled down to a few steps, but the journey there was pretty convoluted!

These guys at AREDN [also](https://www.arednmesh.org) have firmware for the Litebeam AC Gen2 and they are also experiencing all of the same pain as us who are trying to install OpenWRT on ours. BUT their firmware has the patch to deal with the 8.7.4 UBNT firmware that we can't downgrade from. So I thought I would give their firmware a try to get off UBNT firmware then see if I could then upgrade via their web UI to an OpenWRT version and I am very, VERY happy to report that it worked - I uploaded OpenWrt 23.05.4, restored a backup of my config and bish-bash-bosh - its working!!!  

What a bloody palava!!!

Anyway, if you're stuck trying to fix your Litebeam AC Gen2 after the "stable" 24.10 release has goosed it, do the following:-

1. Download and install UBNT 8.7.4 via tftp

2. after the tftp reboot, use the web UI to re-upload the same firmware and reboot.

3. download the ARENT firmware using their firmware selector here [https://downloads.arednmesh.org/afs/www/](url)

4. then connect via SSH and follow the instructions here [https://openwrt.org/toh/ubiquiti/common#wa_v874xc_v874 ](url) to patch the fwupdate.real binary and copy your f/w from step 3. to the antenna. Let it upload and reboot

5. Then set your NIC you used for the tftp and SSH to DCHP - you will receive a 10.x.x.x ipaddress from the Litebeam

6. log on to the ARENT web ui using the 10.x.x.x address. and set your node name and passowrd as per the instructions on their page.

7. Go to the firmware part of their web UI once you're logged in (note - its not obvious in the UI that parts are clickable  - they are)

8. upload your chosen, working OpenWRT (i used 23.05.4)

9. after the upload/reboot - connect to the 10.x.x.x ip address again and either restore your config or re-set it all up.

10. reboot, pour red wine and breathe a sigh of relief....

zendesk-thittesdorf, 2025-03-03:
@grizlyadams got me down the right path with my Nanostation 5AC.  Had to overcome a few additional issues though as ARENT was complaining that the OpenWRT 23.05.4 image wasn't valid and an attempt to force update left the device unbootable.

So here's my steps:

1. Install Ubiquiti firmware 8.7.4 via tftp
2. Install Ubiquiti firmware 8.7.4 from Ubiquiti Web UI
3. Install ARENT firmware using the standard Ubiquiti SSH install method
4. Install Ubiquiti firmware 8.7.4 via tftp
5. Install OpenWRT firmware 23.05.4 via standard Ubiquiti SSH install method
6. ???
7. Profit

grizlyadams, 2025-03-11:
@zendesk-thittesdorf - weird that it let you install 23.05.4 after 8.7.4 tftp and also weird that ARENT complained about 23.05.4, but glad you got it sorted. 💪

m6w6, 2025-03-14:
Canconfirm https://github.com/openwrt/openwrt/issues/17914#issuecomment-2695481960 worked! 👍 🥳

nivex, 2025-05-19:
Just ran headlong into this upgrading my Nanostation AC Loco to 24.10.1. Going to have to do the tftp dance to get this thing back to 21.02 since that was the only thing that worked on it. Though having already been through this with 23.05 I'm thinking I'll write off using OpenWRT on it ever again.

nfontan, 2025-06-06:
> I skipped the AREDN route, just reflashed the 8.5.7:
> 
> 1. boot to tftp mode by holding the reset button while powering the device.
> 2. give you machine an ip like 192.168.1.2
> 3. tftp 192.168.20.1
> 4. type binary
> 5. type put FIRMWARE FILE
> 6. Wait a minute, device reboots
> 7. Go to the webbrowser and go to 192.168.20.1
> 8. Set a password
> 9. ssh user@192.168.20.1
> 10. hexdump -Cv /bin/ubntbox | sed 's/14 40 fe fd/00 00 00 00/g' | hexdump -R > /tmp/fwupdate.real
> 11. chmod +x /tmp/fwupdate.real
> 12. wget the factory openwrt image into /tmp
> 13. /tmp/fwupdate.real -m FIRMWARE FILE
> 
> [image]
> 
> No more ram based overlay at 100% disk usage.

This process worked for me, with just one difference: I had to do it with firmware 8.7.4, and step 10 is slightly different.
hexdump -Cv /bin/ubntbox | sed 's/14 40 fe 27/00 00 00 00/g' | hexdump -R > /tmp/fwupdate.real
chmod +x /tmp/fwupdate.real
https://openwrt.org/toh/ubiquiti/common

I also generated an image with the fix and posted it to my repository in case anyone needs it.
https://github.com/nfontan/openwrt-loco5ac

hauke, 2025-09-02:
The fix was integrated in Linux kernel [6.12.44](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit?h=v6.12.44&id=264e99d), [6.6.103](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit?h=v6.6.103&id=b9e9afc) merged into main in b92bab633fe8cf3c3a72345c08d1ab6a8dd0abd2 and 1c92e468d544db98018c55d5d18e41fc1a3af20b and openwrt-24.10 in c676281e7e26c532b53908f84c59786fde189a76.

closed 2025-09-02 by hauke

sandervandegeijn, 2025-09-04:
Hm tried to tftp that to my m2 400:

```
Sending DATA packet 12924 of 512 bytes
Waiting 5 seconds for packet
Received 4 bytes in a ACK packet
Sending DATA packet 12925 of 512 bytes
Waiting 5 seconds for packet
Received 4 bytes in a ACK packet
Sending DATA packet 12926 of 512 bytes
Waiting 5 seconds for packet
Received 4 bytes in a ACK packet
Sending DATA packet 12927 of 512 bytes
Waiting 5 seconds for packet
Received 4 bytes in a ACK packet
Sending DATA packet 12928 of 512 bytes
Waiting 5 seconds for packet
Received 4 bytes in a ACK packet
Sending DATA packet 12929 of 334 bytes
Waiting 5 seconds for packet
Got ERROR packet: Firmware check failed
Sent 6619136 bytes during 13.8 seconds in 12928 blocks
```

The current released install bin did work.

sandervandegeijn, 2025-09-04:
nevermind got it, should have used the factory image over tftp again. Flashed ok and it does save my new root password without errors or loss of data. Great!