openwrt/openwrt #17915: Hardware flow offloading and conntrack bug breaking long-lived connections

mark8223, 2025-02-08:
### Describe the bug

TL;DR: Wireguard peer behind OpenWrt firewall with "Hardware flow offloading" enabled stops forwarding data after some time, because OpenWrt seemingly forgets the conntrack entry for its connection. 

Full story (this might be long, but I gathered a lot of data that I hope will help):

With hardware flow offloading turned on and a WireGuard client in the local LAN connecting to a WireGuard server on the internet, the tunnel stops forwarding packets after some time.

I know, I know, they are both simply peers, but for the purpose of this description it helps calling them client and server.

The WireGuard server on the internet has:
- public IP address `x.x.x.x`
- port `3578`

The WireGuard client in the local LAN has:
- private IP address `192.168.81.11`
- public IP address `y.y.y.y`
- source port `38720`
- MAC address `aa:aa:aa:aa:aa:aa`

Looking with tcpdump at the relevant interfaces when this reproduces, I can see outgoing traffic (the WireGuard handshake TX - 148 bytes) reaching the server, the server responds with the handshake RX - 92 bytes, which arrives at the WAN interface of the local OpenWrt router, which responds with: 

```
10:03:29.395418 PPPoE  [ses 0x16ca] IP y.y.y.y > x.x.x.x: ICMP y.y.y.y udp port 38720 unreachable, length 128
```

To me, it looks like the outgoing path is forwarding data correctly, but the incoming path is dead and the firewall responds as if there's no NAT hole punched and the port is closed.

After investigating more, it looks like when this is happening,
(1) the conntrack entry for this connection is not present at all
(2) in the ppe0 file, the outgoing path is BND (binded), but the incoming path is UNB (unbinded)

```
# cat /proc/net/nf_conntrack | grep 38720

```

```
# cat /sys/kernel/debug/ppe0/entries | grep 38720
00e06 BND IPv4 5T orig=192.168.81.11:38720->x.x.x.x:3578 new=z.z.z.z:38720->x.x.x.x:3578 eth=34:60:f9:8d:f8:69->40:a6:b7:7c:fc:a0 etype=6488 vlan=0,0 ib1=61481436 ib2=007ff051 packets=0 bytes=0
02de2 UNB IPv4 5T orig=x.x.x.x:3578->z.z.z.z:38720 new=x.x.x.x:3578->192.168.81.11:38720 eth=34:60:f9:8d:f8:68->aa:aa:aa:aa:aa:aa etype=0400 vlan=0,0 ib1=50000136 ib2=007ff035 packets=0 bytes=0
```

How I see this working ideally is that the conntrack entry should never get stale, because there are keepalives sent by the client every 25 seconds. Then, when this reproduces, there are TX handshakes every 5 seconds because the tunnel tries to re-establish connection, which should recreate the conntrack entry even if it had gone stale.

I was initially really conflicted to see outgoing traffic reaching the server without any conntrack entry, however I found an explanation for that. I think these packets hit the outgoing path in the ppe0 file (which is still BND), thus bypassing the kernel, and the conntrack entry never gets recreated so that an incoming path can be recreated.

To me, it looks like some kind of synchronization problem between the kernel's conntrack table and the ppe0 offloading engine's table, which end up diverging at some point.

### OpenWrt version

23.05.4, 24.10.0

### OpenWrt release

23.05.4, 24.10.0

### OpenWrt target/subtarget

ramips/mt7621

### Device

TP-Link Archer C6U v1

### Image kind

Official downloaded image

### Steps to reproduce
[... 226 lines cut ...]

Conntrack entry disappears indefinitely even though there is traffic continuously flowing in both directions, which should help to either keep it from expiring, or recreate it once it expires.

Some time after this happens, the incoming path gets UNB in the ppe0 offloading engine and incoming traffic stops flowing indefinitely for this connection.

### Expected behaviour

Conntrack entry should be present in the conntrack table, and both directions should be BND in the ppe0 offloading engine.

### Additional info

No additional packages, no fancy configurations, no VLANs, etc.
WAN connection getting one publicly routable IPv4 and a /56 IPv6, using PPPoE.
Just the official image for either 23.05.4 or 24.10.0 (same behaviour between them in this regard, I waited for the first stable version of 24.10 just to upgrade and test it too, but no luck).
Official 23.05.5 image fails booting approximately 1 out of 10 times (which is an entirely different mysterious and unsolved problem), but this is the reason I kept 23.05.4.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-02-08:
Could you make connection event recording using
```
conntrack -E -o extended
```

brada4, 2025-02-08:
17 is not timeout, it is `ct protocol udp`

brada4, 2025-02-09:
Aldo is it sw hw or both offloads?

mark8223, 2025-02-09:
> 17 is not timeout, it is ct protocol udp

I might have been unclear, I did not say that is the timeout, I said you can find the timer (when it appears), to the right of the "17" column, and to the left of the "src=" column.
You can try to search for `17 149 src=` in this page to find exactly when it appears.

> Aldo is it sw hw or both offloads?

From my experiments, this bug is only present with "Hardware flow offloading", and not with "Software flow offloading" or no offloading at all.

> Could you make connection event recording using conntrack -E -o extended

Sure.

When trying to reproduce this again, I found an even simpler way to reproduce this bug.

**You can literally apply any unrelated firewall rule change, in this case I used `Network > Firewall > Traffic Rules > uncheck Allow-MLD > Save & Apply`, and the tracked connection will lose the [HW_OFFLOAD] tag, will gain an expiry timer, and will actually expire and stay expired indefinitely as long as there is traffic flowing through the connection that will take the paths created by ppe0.**

Small recap of what will happen next: The connection will keep working for some time, taking the paths created by ppe0. Then, as in the initial description of the bug, when the incoming path in ppe0 eventually gets unbinded after some time, the connection will be broken indefinitely. There will be no conntrack entry, the outgoing path will be facilitated by the ppe0 outgoing path, and the incoming path will just be rejected by the firewall with ICMP port unreachable.

Full walkthrough below:

```
# while true; do date; (cat /proc/net/nf_conntrack | grep 38720); (cat /sys/kernel/debug/ppe0/entries | grep 38720); echo; sleep 1; done

17:14:20
```
No connections are made at this point. In the following frame I'll start the WireGuard tunnel which is continuously forwarding some data.

```
17:14:21
ipv4     2 udp      17 src=192.168.81.11 dst=x.x.x.x sport=38720 dport=3578 packets=27 bytes=3592 src=x.x.x.x dst=y.y.y.y sport=3578 dport=38720 packets=25 bytes=7448 [HW_OFFLOAD] mark=0 zone=0 use=3
0252a UNB IPv4 5T orig=x.x.x.x:3578->y.y.y.y:38720 new=42.3.40.128:64206->240.35.0.12:45068 eth=00:00:00:00:00:00->00:00:00:00:00:00 etype=0000 vlan=0,0 ib1=50001ae8 ib2=497f6a55 packets=0 bytes=0
03c98 UNB IPv4 5T orig=192.168.81.11:38720->x.x.x.x:3578 new=0.0.0.0:0->0.0.0.0:0 eth=00:00:00:00:00:00->00:00:00:00:00:00 etype=0000 vlan=0,0 ib1=50001de8 ib2=00000000 packets=0 bytes=0
```

At this point, `conntrack -E -o extended` said:
```
    [NEW] ipv4     2 udp      17 60 src=192.168.81.11 dst=x.x.x.x sport=38720 dport=3578 [UNREPLIED] src=x.x.x.x dst=y.y.y.y sport=3578 dport=38720
 [UPDATE] ipv4     2 udp      17 86400 src=192.168.81.11 dst=x.x.x.x sport=38720 dport=3578 src=x.x.x.x dst=y.y.y.y sport=3578 dport=38720 [OFFLOAD]
```

In the next frame, connection is tracked as [HW_OFFLOAD] with no expiry timer, and both directions are binded. This is the usual steady state that we're aiming for.

```
17:14:22
ipv4     2 udp      17 src=192.168.81.11 dst=x.x.x.x sport=38720 dport=3578 packets=31 bytes=4600 src=x.x.x.x dst=y.y.y.y sport=3578 dport=38720 packets=31 bytes=11904 [HW_OFFLOAD] mark=0 zone=0 use=3
0252a BND IPv4 5T orig=x.x.x.x:3578->y.y.y.y:38720 new=x.x.x.x:3578->192.168.81.11:38720 eth=34:60:f9:8d:f8:68->aa:aa:aa:aa:aa:aa etype=0400 vlan=0,0 ib1=614108e9 ib2=007ff035 packets=0 bytes=0
03c98 BND IPv4 5T orig=192.168.81.11:38720->x.x.x.x:3578 new=y.y.y.y:38720->x.x.x.x:3578 eth=34:60:f9:8d:f8:69->40:a6:b7:7c:e3:50 etype=6488 vlan=0,0 ib1=614808e9 ib2=007ff051 packets=0 bytes=0
```

Starting from the next frame, I'll apply any unrelated firewall rule change (in this case Network > Firewall > Traffic Rules > Uncheck Allow-MLD > Save & Apply), and you'll see the tracked connection losing the [HW_OFFLOAD] tag and gaining an expiry timer starting with 149, all while `conntrack -E -o extended` did not output anything new.
Traffic will take the paths created by ppe0 and thus not refresh the timer, so the conntrack entry will just expire.

```
17:14:30
ipv4     2 udp      17 src=192.168.81.11 dst=x.x.x.x sport=38720 dport=3578 packets=31 bytes=4600 src=x.x.x.x dst=y.y.y.y sport=3578 dport=38720 packets=31 bytes=11904 [HW_OFFLOAD] mark=0 zone=0 use=3
0252a BND IPv4 5T orig=x.x.x.x:3578->y.y.y.y:38720 new=x.x.x.x:3578->192.168.81.11:38720 eth=34:60:f9:8d:f8:68->aa:aa:aa:aa:aa:aa etype=0400 vlan=0,0 ib1=614108ef ib2=007ff035 packets=0 bytes=0
03c98 BND IPv4 5T orig=192.168.81.11:38720->x.x.x.x:3578 new=y.y.y.y:38720->x.x.x.x:3578 eth=34:60:f9:8d:f8:69->40:a6:b7:7c:e3:50 etype=6488 vlan=0,0 ib1=614808ef ib2=007ff051 packets=0 bytes=0

17:14:31
ipv4     2 udp      17 149 src=192.168.81.11 dst=x.x.x.x sport=38720 dport=3578 packets=31 bytes=4600 src=x.x.x.x dst=y.y.y.y sport=3578 dport=38720 packets=31 bytes=11904 mark=0 zone=0 use=2
0252a BND IPv4 5T orig=x.x.x.x:3578->y.y.y.y:38720 new=x.x.x.x:3578->192.168.81.11:38720 eth=34:60:f9:8d:f8:68->aa:aa:aa:aa:aa:aa etype=0400 vlan=0,0 ib1=614108f1 ib2=007ff035 packets=0 bytes=0
03c98 BND IPv4 5T orig=192.168.81.11:38720->x.x.x.x:3578 new=y.y.y.y:38720->x.x.x.x:3578 eth=34:60:f9:8d:f8:69->40:a6:b7:7c:e3:50 etype=6488 vlan=0,0 ib1=614808f1 ib2=007ff051 packets=0 bytes=0

17:14:32
ipv4     2 udp      17 148 src=192.168.81.11 dst=x.x.x.x sport=38720 dport=3578 packets=31 bytes=4600 src=x.x.x.x dst=y.y.y.y sport=3578 dport=38720 packets=31 bytes=11904 mark=0 zone=0 use=2
[... cut ...]
17:14:21     [NEW] ipv4     2 udp      17 60 src=192.168.81.11 dst=x.x.x.x sport=38720 dport=3578 [UNREPLIED] src=x.x.x.x dst=y.y.y.y sport=3578 dport=38720
17:14:21  [UPDATE] ipv4     2 udp      17 86400 src=192.168.81.11 dst=x.x.x.x sport=38720 dport=3578 src=x.x.x.x dst=y.y.y.y sport=3578 dport=38720 [OFFLOAD]
17:17:02 [DESTROY] ipv4     2 udp      17 src=192.168.81.11 dst=x.x.x.x sport=38720 dport=3578 packets=31 bytes=4600 src=x.x.x.x dst=y.y.y.y sport=3578 dport=38720 packets=31 bytes=11904
```

brada4, 2025-02-09:
Got it. Lets make hw offload  fire on and off more often. 
sysctl net.netfilter.nf_flowtable_udp_timeout=5
repeat `conntrack -E -o extended`
should be relpied+offload (sleep 5s) offload->(replied) (some packet) replied+offload ... and so on
1/ is it really 5s and no more
2/ are packet counters updated

brada4, 2025-02-09:
Linked issue: yes, could be this, maybe I did not see past offload being expelled when >mtu packet is pushed into tunnel and offload is terminated.

danpawlik, 2025-02-09:
@mark8223 I got opposite: few times on the forum I mentioned issue related to the wifi calling /vowifi that it is starting working after some time, not immediately like when you have just sw offload enabled. I thought it is related to BPi-R4... And maybe it is and this is just another case.

brada4, 2025-02-09:
@danpawlik there is a documented limitation that hw offload keeps stale states for one of 30s timeouts.
REF: https://www.kernel.org/doc/html/v5.15/networking/nf_flowtable.html#limitations
24.10 makes control more precise - sw offload stops at software bridges where forwarding is enabled keeping ARP and FIB examined as without offload, hw offload checkbox even not supported takes the freedom to bypass those 2.

brada4, 2025-02-09:
Does not magically beam to mainline, but yes, you have bery little control here.

RomanovDanila, 2025-02-10:
Shouldn't this help?

```
 static void nf_flow_offload_gc_step(struct nf_flowtable *flow_table,
 				    struct flow_offload *flow, void *data)
 {
+	bool teardown = test_bit(NF_FLOW_TEARDOWN, &flow->flags);
+
 	if (nf_flow_has_expired(flow) ||
 	    nf_ct_is_dying(flow->ct) ||
 	    nf_flow_custom_gc(flow_table, flow))
 		flow_offload_teardown(flow);
+	else if (!teardown)
+		nf_flow_table_extend_ct_timeout(flow->ct);
```
https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/net/netfilter/nf_flow_table_core.c?h=next-20250210&id=03428ca5cee9f0792edc996c06ce4514816af1fb

brada4, 2025-02-10:
Not so round timer points to that ... But that is about sw offload where timer is extended on next packet and kind of does not show loud.

brada4, 2025-02-10:
The flow is continuously valid, i am just chasing ghosts which state transition breaks the truth.

mark8223, 2025-02-10:
> Got it. Lets make hw offload fire on and off more often. `sysctl net.netfilter.nf_flowtable_udp_timeout=5` (dont save ;-) ) repeat `conntrack -E -o extended` should be relpied+offload (sleep 5s) offload->(replied) (some packet) replied+offload ... and so on 1/ is it really 5s and no more 2/ are packet counters updated EDIT: and counters set to default ie not shown, or close to that at every event.

I applied the sysctl and re-ran the scenario from my last comment and it played out exactly the same, so I won't attach any logs.

It should be very easily reproducible for anyone with minimal setup. Just fire off a WireGuard tunnel, stream something through it, apply any firewall change and watch the router forget the connection in a few minutes.

**Unfortuantely, today I found the conntrack entry missing a few hours after a clean boot** (after which I did not apply any settings to the router; it was just a clean, fresh boot, but somehow the entry still expired). So it does happen organically as well, not only with my reproduction steps.

I've been thinking about what can actually be done here, since you can look at this from 2 perspectives:

1. Why does the connection lose the [HW_OFFLOAD] tag and then expire? It shouldn't do that, the packets are still flowing in both directions. Or, maybe, let's say it should, then:
2. When the connection expires, it should also bring down the ppe entries, because then the next packets would actually hit the kernel and recreate the tracked connection, which should also repopulate the ppe entries. This sounds a lot like the feature @romanovj linked above.

I am not aware of the mechanics of how this stuff is handled internally, but I hope I provided enough context and easy reproduction steps that maybe someone more experienced can pick this up and make progress.

brada4, 2025-02-10:
Thanks for info
1/ hw states do not return  to slowpath on timer.
2/ destroying hw flowtable does not return state to conntrack slowpath properly.

@Ansuel ?

brada4, 2025-02-10:
Workaround: obviously sw offload

mark8223, 2025-02-12:
Just in case we needed more proof, it happened again randomly today after ~36 hours of uptime on a fresh boot never having touched anything:

```
root@OpenWrt:~# cat /proc/net/nf_conntrack | grep 38720

```

```
root@OpenWrt:~# cat /sys/kernel/debug/ppe0/entries | grep 38720
00e06 BND IPv4 5T orig=192.168.81.11:38720->x.x.x.x:3578 new=y.y.y.y:38720->x.x.x.x:3578 eth=34:60:f9:8d:f8:69->40:a6:b7:7c:f2:50 etype=6488 vlan=0,0 ib1=61481e81 ib2=007ff051 packets=0 bytes=0
03b74 UNB IPv4 5T orig=x.x.x.x:3578->y.y.y.y:38720 new=x.x.x.x:3578->192.168.81.11:38720 eth=34:60:f9:8d:f8:68->aa:aa:aa:aa:aa:aa etype=0400 vlan=0,0 ib1=50000181 ib2=007ff035 packets=0 bytes=0
```

I'm switching back to only software flow offloading until we need any more data points.

@romanovj 
>An interesting patch and comment.

That sounds relevant. Unfortunately I see the discussion happened back in November 2022 and did not materialise into anything... Maybe it was just overlooked?

Serzh-L, 2025-03-26:
I have the same problem. My device is an ASUS RT-AC85P on mt7621 with hw_offload. I noticed that when WireGuard established a connection, but there is no traffic inside the tunnel, only a "handshake" (10 seconds), then in the Connection -> Status I see an increase in the number of transmitted packets. If there is enough traffic in the tunnel, the number of packets freezes. When the exchange stops inside the tunnel, the number of transmitted packets starts to grow again after some times.
If I establish a connection and there is enough traffic for the number of packets to "freeze" and turn off WireGuard, then after a while I will see a "magical" increase in the number of packets.
If I restart WireGuard, the first connection will have a counter for 150 seconds (I know it's actually 180, but the command cat/proc/net/nf_conntrack | grep starts showing the counter only from 146-150).
It all looks like a problem.
P.S.: I have other tunnels, on mt7621 itself. They have a timer of 180 seconds (the connection is defined as net.netfilter.nf_conntrack_udp_timeout_stream=180) is always displayed.

Serzh-L, 2025-03-27:
I found a fairly quick way to reproduce this bug 100%.
1. Install the Wireguard tunnel via OpenWRT with hw_offload.
2. Launch ping (ping {ip_address_over_the_tunnel} -t -l 65500) through the tunnel to create constant traffic. I ran 14 copies in parallel to generate more than 10 Mb/sec of traffic. It turned out that the recording from /proc/net/nf_conntrack disappeared quite quickly. After a while, the ping disappeared. At first there were two entries in /sys/kernel/debug/pe 0/entries, but then there was only one entry of outgoing traffic marked BND. The incoming recording has disappeared.

brada4, 2025-03-27:
Hardware offload grips only udp stream under wireguard on other  machine to somewhere on the internet.

Serzh-L, 2025-04-01:
OK, I did some experiments. I got three possible different entries in /proc/net/nf_conntrack. When you connect for the first time, the [UNREPLIED] label appears. Subsequent connections do not have this label and do not have a timer. In the third option, I managed to notice a timer for 148 (maybe 150) seconds. But I've never seen the [HW_OFFLOAD] label disappear.
What I changed:
1. Routing.
2. Install updates.
But if I don't change the firewall and routing settings (in fact, I haven't changed any settings at all), then everything has been working stably for 3 days now with traffic of about 500GB. I checked on OpenWRT 24.10.0 (r28427-6df0e3d02a) with the latest updates on 25.087.31282~f0a70d0. I think this is the closest thing to the true problem.
It is possible that after making changes to the firewall or routing, you will need to restart the running tunnels. Maybe set up watchcat (luci-app-watchcat) or use a manual script.
@mark8223, have you ever noticed something like this?

brada4, 2025-04-01:
Offload maintains own timer and in sysctl-able 30s drops a packet back to normal slowpath.

Serzh-L, 2025-04-02:
I think I can test the operation of the 30 second interval by increasing the number of packets every 30 seconds. If I load the tunnel at about 10-15 mbps, then I don't see any changes for more than 10-20 minutes (maybe 30m), but everything works fine. I can confirm 100% that this bug will occur if you select "Disable" any route from the "Routing" menu. I can fix it by restarting the tunnel itself on another device, or by restarting mt7621.
@brada4, Thank you for not giving up on this problem.

brada4, 2025-04-02:
I cant get it to fail on me

Serzh-L, 2025-04-02:
If you need to do any testing, I'll be happy to help.

Serzh-L, 2025-04-04:
@romanovj, I'm really sorry, but I don't know what to do about it. As I understand it, only two parameters change. If you can tell me how to make these changes using ssh, then I can do it.
I found a way around this problem. After each change to the routing or firewall settings, press the "Reload" button on the "19 firewall" in System -> Startup. The way I think it solves this problem.

mark8223, 2025-04-05:
@Serzh-L This bug happens from a clean cold boot as well, without any runtime changes, so that doesn't really apply.

Serzh-L, 2025-04-06:
@mark8223 Yes, you're absolutely right. It's just that I've seen different packet magnification behavior. I once saw an increase in the number of packets every second. It worked for more than 3 days until I rebooted mt7621 myself. Unfortunately, I couldn't repeat it. It seems that my conclusions are premature and I have not yet been able to establish an algorithm for creating a stable connection.

Serzh-L, 2025-05-13:
@mark8223, I upgraded openwrt to 24.10.1 and got a stable connection for over 8 days. But after making the changes, the problem returned. I rebooted the router. After the reboot, the new connection has been working for more than 10 days. The bug may have been partially fixed. Does it work the same way in your case?

ECquadrat, 2025-06-06:
I have also the issue here, that Wireguard is disabled after some minutes on my "AVM FRITZ!Box 3370 Rev. 2 (Micron NAND)" router with OpenWrt 24.10.1 running.

The Wireguard setting runs smoothly on firmware 23.05.5 without any issues. After flashing 24.10.1 and updating all packages, Wireguard is also starting, I can see on the "Status/WireGuard" tab in luci that two Peers are registered and WireGuard clients are fully working when connected. But after some minutes the entries in the "WireGuard Status" page are disappearing and it's stated "no such device". After switching to the luci interfaces tab and clicking restart on the wg0 WireGuard device, the items in the status page reappearing and WireGuard is working for another couple of minutes before it's not working again.

Switching back to firmware 23.05.5, WireGuard is running smoothly again without any issues.

brada4, 2025-07-20:
24.10 soft offload permits all roaming as the offload terminates on br-lan, not physical/wifi ports.

oxwivi, 2025-07-21:
After upgrade to 24.10, I tried again with the offloads and still couldn't get WireGuard to survive roaming until I disabled offload entirely. If soft offload is supposed to permit all, there's a bug there as well.

brada4, 2025-07-21:
Please check (start of) `fw4 print` for device set in flowtable definition - roaming happens on the side of "bridge" but not where physical interface is noted and arp and fdb get bypassed by offload.

oxwivi, 2025-08-14:
> You can disconnect client via web ui and it will reconnect to other AP with same name under its own understandings, or reconnect back after 5+s

I thought `fw4 print` would be a log of conntrack somehow. Anyhow, I've confirmed that this happens on three different devices, RT3200/Linksys E8540, WAX206 and WR3000s. They are all Mediatek devices though, and FWIW my service uses PPPoE.

Anyhow, the result of `fw4 print`:
```
# fw4 print
table inet fw4
flush table inet fw4
delete flowtable inet fw4 ft

table inet fw4 {
        #
        # Flowtable
        #

        flowtable ft {
                hook ingress priority 0;
                devices = { "lan1", "lan2", "lan3", "lan4", "pppoe-wan", "wan", "wl0-ap0", "wl0-ap1", "wl1-ap0", "wl1-ap1" };
                counter;
                flags offload;
        }

        #
        # Defines
        #

        define lan_devices = { "br-lan" }
        define lan_subnets = { 192.168.1.0/24, 2001:8f8:1839:e627::/64, fd42:5793:8834::/60 }

        define wan_devices = { "pppoe-wan", "wan" }
        define wan_subnets = { 83.110.134.79, fe80::9883:5846:105c:4ed6, 2001:8f8:1838:7af8::/64 }

        define rent_devices = { "lan4" }
        define rent_subnets = { 192.168.3.0/24 }

        #
        # User includes
        #

        include "/etc/nftables.d/*.nft"

        #
        # Filter rules
        #

        chain input {
                type filter hook input priority filter; policy drop;

                iif "lo" accept comment "!fw4: Accept traffic from loopback"

                ct state vmap { established : accept, related : accept } comment "!fw4: Handle inbound flows"
                tcp flags & (fin | syn | rst | ack) == syn jump syn_flood comment "!fw4: Rate limit TCP syn packets"
                iifname "br-lan" jump input_lan comment "!fw4: Handle lan IPv4/IPv6 input traffic"
                iifname { "pppoe-wan", "wan" } jump input_wan comment "!fw4: Handle wan IPv4/IPv6 input traffic"
                iifname "lan4" jump input_rent comment "!fw4: Handle rent IPv4/IPv6 input traffic"
                jump handle_reject
        }

        chain forward {
                type filter hook forward priority filter; policy drop;

                meta l4proto { tcp, udp } flow offload @ft;
                ct state vmap { established : accept, related : accept } comment "!fw4: Handle forwarded flows"
                iifname "br-lan" jump forward_lan comment "!fw4: Handle lan IPv4/IPv6 forward traffic"
                iifname { "pppoe-wan", "wan" } jump forward_wan comment "!fw4: Handle wan IPv4/IPv6 forward traffic"
                iifname "lan4" jump forward_rent comment "!fw4: Handle rent IPv4/IPv6 forward traffic"
                jump handle_reject
        }

[... 150 lines cut ...]
        #

        chain mangle_prerouting {
                type filter hook prerouting priority mangle; policy accept;
        }

        chain mangle_postrouting {
                type filter hook postrouting priority mangle; policy accept;
                oifname { "pppoe-wan", "wan" } tcp flags syn / syn,fin,rst tcp option maxseg size set rt mtu comment "!fw4: Zone wan IPv4/IPv6 egress MTU fixing"
        }

        chain mangle_input {
                type filter hook input priority mangle; policy accept;
        }

        chain mangle_output {
                type route hook output priority mangle; policy accept;
        }

        chain mangle_forward {
                type filter hook forward priority mangle; policy accept;
                iifname { "pppoe-wan", "wan" } tcp flags syn / syn,fin,rst tcp option maxseg size set rt mtu comment "!fw4: Zone wan IPv4/IPv6 ingress MTU fixing"
        }
}
```

brada4, 2026-01-18:
There are no material changes in fw4 since, prs are rotting for years.

tabascoz, 2026-01-20:
> There are no material changes in fw4 since, prs are rotting for years.

I think the bug is wrongly marked as invalid. it does affect 24.10.5

kprzybys, 2026-10-17:
Same bug here on **GL.iNet GL-MT6000 (MT7986), OpenWrt 25.12.5, kernel 6.12.94**, `flow_offloading=1`, `flow_offloading_hw=1`, `nf_flowtable_udp_timeout=30`. Workload: Tailscale (WireGuard) on a LAN host, UDP 41641 ↔ a VPS peer on UDP 41641. Some extra data points that may help:

**1. Orphaned one-way PPE entry with no conntrack entry at all, kept alive by traffic**

```
# grep 192.168.1.150:41641 /sys/kernel/debug/ppe0/bind
010c8 BND IPv4 5T orig=192.168.1.150:41641->P.P.P.P:41641 new=10.x.x.x:41641->P.P.P.P:41641 ... packets=57525
# grep P.P.P.P /sys/kernel/debug/ppe1/bind
(nothing – no reverse direction)
# conntrack -L -p udp | grep 192.168.1.150 | grep P.P.P.P
(nothing)
```

The `packets` counter of this entry grows 1:1 with packets sent by the LAN host (`57582 → 57591` for 5 pings + replies), so the hardware keeps forwarding the outbound half using a stale SNAT mapping (source port 41641). Replies arrive on WAN for `10.x.x.x:41641`, have no conntrack entry and are rejected in `input_wan`. Inbound-only direction works, so the tunnel is asymmetric (in our case Tailscale falls back to a relay).

**2. The orphan survives `conntrack -D`**

`conntrack -D -p udp --orig-src P.P.P.P` and `--orig-src 192.168.1.150 --orig-dst P.P.P.P` did not remove the PPE entry; it was still `BND` with a growing counter minutes later. Because the LAN host keeps sending (WireGuard keepalives), the entry never ages out.

**3. How it was triggered (for us)**

We enabled miniupnpd (NAT-PMP) on the router, so the LAN host got a new external port (41643) and new flows were set up for the same 5-tuple (`192.168.1.150:41641 ↔ P.P.P.P:41641`). Our guess (not verified step by step): the old outbound flow's conntrack entry went away, but its HW entry stayed bound. The new conntrack entry for the same tuple (peer → DNAT 41643 → LAN host) then stayed `[UNREPLIED]` forever, because the LAN host's packets never reach the CPU – they are consumed by the stale PPE entry:

```
udp 17 src=P.P.P.P dst=10.x.x.x sport=41641 dport=41643 [UNREPLIED] src=192.168.1.150 dst=P.P.P.P sport=41641 dport=41641 packets=0 [HW_OFFLOAD]
```

A second peer showed the identical pattern at the same time (`01894 BND ... orig=192.168.1.150:41641->S.S.S.S:41643`, no reverse entry, no conntrack).

**4. Workaround without disabling HW offload**

Stopping the UDP sender on the LAN host for ~2.5 minutes (`systemctl stop tailscaled`) let the PPE entries age out; after restarting, both flows were rebuilt correctly (proper SNAT + `[ASSURED]` DNAT entries) and direct connectivity came back.

So it looks like a conntrack entry (or flowtable flow) can be torn down while its HW PPE entry stays bound, and nothing reconciles them afterwards – `conntrack -D` doesn't reach it, and continuous traffic prevents HW aging. Happy to run further diagnostics on this device.

macaquinyho, 2026-11-03:
I hit this on an AX3000T (MT7981, 25.12.2) with a WireGuard tunnel and tracked it down.

Any `fw4 reload` triggers it, including the automatic ones from `/etc/hotplug.d/iface/20-firewall` (PPPoE reconnects, DHCP renewals). fw4 deletes and recreates the flowtable on each reload, and since upstream commit 13210fc63f35 ("netfilter: nf_tables: imbalance in flowtable binding") the driver is unbound before the old flows are removed, so the PPE never gets the delete. Conntrack drops `[HW_OFFLOAD]` and expires the entry while the PPE keeps forwarding. When the reply direction ages out of the PPE, replies hit the stack with no conntrack entry and get rejected. With high traffic the stale PPE entry gets re-bound and it heals by itself, with low traffic like WireGuard keepalives it stays broken.

It's quite easy to reproduce: offloaded UDP flow above 30 pps, `fw4 reload`, then watch `/proc/net/nf_conntrack` and `/sys/kernel/debug/ppe0/entries`. The conntrack entry gets a timer while both PPE entries stay `BND`. Removing the `delete flowtable` line from `ruleset.uc` makes it go away.

I sent a fix upstream: https://lore.kernel.org/netfilter-devel/20261103222243.445644-1-vallsfustearnau@gmail.com/

Tested on 6.12.74 and 7.4-rc1. Until it's merged, software offloading avoids the issue.