openwrt/openwrt #17917: [24.10] Netgear RAX120v2 only able to boot into initramfs

dkgof, 2025-02-09:
### Describe the bug

When i install the "web-ui-factory.img" it works, but im stuck in initramfs mode, any "sysupgrade.bin" i install puts my device into a reboot loop, where it only just boots up enough to be pingable 2-3 times, then crashes again and reboots. It never gets up and running.

Only way to use it, is to go into tftp mode and flash the web-ui-factory image again, and im back to initramfs.

### OpenWrt version

24.10.0

### OpenWrt release

r28427-6df0e3d02a

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Netgear RAX120v2

### Image kind

Official downloaded image

### Steps to reproduce

Install "web-ui-factory.img", then when booted into initramfs mode, install "sysupgrade.bin".

### Actual behaviour

The device keeps rebooting, before starting up completely.

### Expected behaviour

The system should boot correctly with the sysupgrade flash, since it works from initramfs.

Additional info

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

zsh, 2025-02-11:
I am able to reproduce this from a clean installation on the `24.10.0` release version, which it's not an issue in `24.10-rc2`

### Steps to reproduce
- In Netgear UI stock firmware, web upgrade using `openwrt-24.10.0-qualcommax-ipq807x-netgear_rax120v2-squashfs-web-ui-factory.img`
- RAX120v2 boots into LuCI without problem
- Upgrade using `openwrt-24.10.0-qualcommax-ipq807x-netgear_rax120v2-squashfs-sysupgrade.bin`
- RAX120v2 stuck in reboot loop.

dkgof, 2025-04-17:
Has anyone tried with the new 24.10.1?

Reginold-P, 2025-05-29:
No luck. I tried with 24.10.0, 24.10.1 and snapshot (28th May 2025). All keeps rebooting.

Although, Netgear fixed the WiFi reboot issue in their latest patch, [1.2.9.52](https://kb.netgear.com/000065822/RAX120-RAX120v2-Firmware-Version-1-2-9-52). It is a well known issue and It might be related.

xuhdev, 2025-12-24:
24.10.5 is also broken.

How to get the serial console logs?

neilvandyke, 2026-01-03:
RAX120v2 was working.  How do we get it to be maintained as a well-supported device?

Is there a developer who would work on RAX120v2 support, but doesn't own the device?

testuser7, 2026-01-16:
> How to get the serial console logs?

There is a detailed information how to connect serial on the wiki: https://openwrt.org/inbox/toh/netgear/netgear_rax120_nighthawk_ax12#serial

brobison75, 2026-11-10:
I just successfully installed 25.12.6 on a RAX120v2.  It previously had 24.10.5 (installed by flashing from the Netgear interface then doing a sysupgrade).  Steps were:

1. Start with a working official 24.10.5 installation.
2. TFTP-boot/install the 25.12.6 web-ui-factory.img, leaving the old rootfs intact.
3. From the 25.12.6 initramfs environment, run a clean sysupgrade -n using the matching 25.12.6 sysupgrade image.
4. Reboot successfully into a persistent installation.

(TFTP install was a `put` from a Debian system, device set to 192.168.1.1 and host set to 192.168.1.10.)

I'm not convinced that initial 24.10.5 is necessary, but I likely won't be trying it from stock Netgear firmware or current 25.12.X any time soon, so I'll just say that's how it worked for me.