openwrt/openwrt #17921: Cannot attach an virtio interface to openwrt KVM with arm architecture

songqing0353036, 2025-02-09:
### Describe the bug

when created an openwrt KVM and hotplug a virtio NIC to this VM:
$ virsh attach-interface vm0 --type bridge --source $eth --model virtio  --live --persistent

and there isn't any NIC in this VM, but when i restart this VM, the nic then can be founded.

while X86 image works well when hotplug a NIC.

### OpenWrt version

any

### OpenWrt release

any

### OpenWrt target/subtarget

any

### Device

armv8

### Image kind

Official downloaded image

### Actual behaviour

1. create a KVM virsh file
<domain type='kvm'>　　
  <name>vm0</name>  
  <memory unit='KiB'>2048576</memory>  
  <currentMemory unit='KiB'>2048576</currentMemory>  
  <vcpu placement='static'>2</vcpu> 
  <sysinfo type='smbios'>
    <system>
      <entry name='manufacturer'>xxxxx</entry>
      <entry name='product'>Cloud Elastic Compute</entry>
      <entry name='version'>18.2.2-1.el7</entry>
      <entry name='serial'>fe291179-2b8a-4fa4-add2-07e22ac44099</entry>
      <entry name='uuid'>fe291179-2b8a-4fa4-add2-07e22ac44099</entry>
      <entry name='family'>Virtual Machine</entry>
    </system>
  </sysinfo>
  <os>
    <type arch='aarch64' machine='virt-3.1'>hvm</type>
    <loader readonly='yes' type='pflash'>/usr/share/AAVMF/AAVMF_CODE.fd</loader>
    <nvram template='/usr/share/AAVMF/AAVMF_VARS.fd'>/var/lib/libvirt/qemu/nvram/50302317-9b64-44db_VARS.fd</nvram>
    <bootmenu enable='yes' timeout='10000'/>
    <smbios mode='sysinfo'/>
  </os>
  <features>
    <acpi/>
    <apic/>
  </features>
  <cpu mode='host-passthrough'/>
  <clock offset='localtime'/>
  <on_poweroff>destroy</on_poweroff>
  <on_reboot>restart</on_reboot>
  <on_crash>restart</on_crash>
  <devices>
    <emulator>/usr/libexec/qemu-kvm</emulator>
    <disk type='file' device='disk'>
      <driver name='qemu' type='raw'/>
      <source file='/var/log/kolla/libvirt/openwrt-24.10.0-rc5-armv8.img'/>
      <target dev='hda' bus='scsi'/>
    </disk>
    <interface type='network'>  
      <source network='default' bridge='virBR0'/>  
    </interface>
<input type='tablet' bus='usb'>
    </input>
    <input type='keyboard' bus='usb'>
    </input>
    <controller type='usb' index='0' model='qemu-xhci'>
    </controller>
    <graphics type='vnc' port='63205' autopart='no' listen='0.0.0.0' keymap='en-us'/>　　#5905为端口号
<controller type='scsi' model='virtio-scsi'>
    </controller>
<controller type='pci' index='0' model='pcie-root'/>
    <controller type='pci' index='1' model='pcie-root-port'>
    </controller>
    <controller type='pci' index='2' model='pcie-root-port'>
    </controller>
    <controller type='pci' index='3' model='pcie-root-port'>
    </controller>
    <controller type='pci' index='4' model='pcie-root-port'>
    </controller>
    <controller type='pci' index='5' model='pcie-root-port'>
    </controller>
    <controller type='pci' index='6' model='pcie-root-port'>
    </controller>
    <controller type='pci' index='7' model='pcie-root-port'>
    </controller>
    <controller type='pci' index='8' model='pcie-root-port'>
    </controller>
    <controller type='pci' index='9' model='pcie-root-port'>
    </controller>
    <controller type='pci' index='10' model='pcie-root-port'>
    </controller>
  </devices>
</domain>

2. Start a KVM
# virsh define vm0.xml  //导入虚拟机配置   create xml
# virsh start vm0  //开启vm0

3. hotplug a NIC
eth=virbr10
ip link add $eth type bridge; ip link set $eth up
virsh attach-interface vm0 --type bridge --source $eth --model virtio  --live --persistent

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-02-09:
Do you see new adapter in lspci? Dmesg?

songqing0353036, 2025-02-10:
> Do you see new adapter in lspci? Dmesg?
no pci device was found in this virtual mathine.

songqing0353036, 2025-02-10:
version: 22.06~24.10 
and the image was download from the official website.
https://downloads.openwrt.org/releases/24.10.0/targets/armsr/armv8/openwrt-24.10.0-armsr-armv8-generic-squashfs-rootfs.img.gz

the virtual mathine was launched by virsh command,  while Nic was inserted by 'virsh attach'

songqing0353036, 2025-02-10:
but it will recover when rebooted. or rescan pci device by 'echo 1 > /sys/bus/pci/rescan',  does any one know the reason? 
X86 image  works well on hot-pluged Nics

aparcar, 2025-06-30:
Could you please try the latest snapshot? I think this was fixed with Kernel 6.6.95

@dangowrt do you think the issue is related to what I discovered?