openwrt/openwrt #17923: R4S: rockchip-pcie f8000000.pcie: PCIe link training gen1 timeout!

Ramon00, 2025-02-09:
### Describe the bug

as reported on the forum: [link](https://forum.openwrt.org/t/openwrt-24-10-0-first-stable-release/223845/271)

After a reboot on 24.10.0  eth1 is unavailable due to a timeout:
`[    0.936433] rockchip-pcie f8000000.pcie: PCIe link training gen1 timeout!`

Seems like a fix is already available: [link](https://github.com/armbian/build/pull/6057)

Can this be cherry picked?

Maybe good to have a look at [this](https://github.com/armbian/build/issues/6655#issuecomment-2538489984) as well?

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

rockchip/armv8

### Device

FriendlyElec NanoPi R4S

### Image kind

Official downloaded image

### Steps to reproduce

Install the firmware. Reboot. gives the timeout
powercycle clears it.

### Actual behaviour

See provided links

### Expected behaviour

No timeout

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

wurzerj, 2025-02-11:
Adding @1715173329 and @blocktrron. Not sure if you have time to take a look at this. I’ll certainly try to do so on the weekend.

To add some context: Interestingly it seems not all devices are affected by this. While some users observe this behavior, I could not reproduce it on my devices.

RHBH, 2025-02-12:
Log available here

https://forum.openwrt.org/t/openwrt-24-10-0-first-stable-release/223845/347

RHBH, 2025-02-12:
> Adding [@1715173329](https://github.com/1715173329) and [@blocktrron](https://github.com/blocktrron). Not sure if you have time to take a look at this. I’ll certainly try to do so on the weekend. [...]

My R4S is not affected.

Bought several years ago using a Raspiberry Pi 15w power supply and Sandisk Extreme Pro 32GB.

hvh-blet, 2025-02-12:
nanopi r4s 4gb
reboot = lan down

After switching on:

root@OpenWrt:~# logread | grep -i 'network\|eth\|wan\|lan\|link'
Mon Feb  3 23:36:32 2025 kern.info kernel: [    0.000000] psci: probing for conduit method from DT.
Mon Feb  3 23:36:32 2025 kern.info kernel: [    0.046563] NET: Registered PF_NETLINK/PF_ROUTE protocol family
Mon Feb  3 23:36:32 2025 kern.info kernel: [    0.263809] rk_gmac-dwmac fe300000.ethernet: IRQ eth_wake_irq not found
Mon Feb  3 23:36:32 2025 kern.info kernel: [    0.264407] rk_gmac-dwmac fe300000.ethernet: IRQ eth_lpi not found
Mon Feb  3 23:36:32 2025 kern.info kernel: [    0.265058] rk_gmac-dwmac fe300000.ethernet: PTP uses main clock
Mon Feb  3 23:36:32 2025 kern.info kernel: [    0.345562] 8021q: 802.1Q VLAN Support v1.8
Mon Feb  3 23:36:32 2025 kern.info kernel: [    0.401607] rk_gmac-dwmac fe300000.ethernet: IRQ eth_wake_irq not found
Mon Feb  3 23:36:32 2025 kern.info kernel: [    0.402203] rk_gmac-dwmac fe300000.ethernet: IRQ eth_lpi not found
Mon Feb  3 23:36:32 2025 kern.info kernel: [    0.402901] rk_gmac-dwmac fe300000.ethernet: PTP uses main clock
Mon Feb  3 23:36:32 2025 kern.info kernel: [    0.415008] rk_gmac-dwmac fe300000.ethernet: IRQ eth_wake_irq not found
Mon Feb  3 23:36:32 2025 kern.info kernel: [    0.415614] rk_gmac-dwmac fe300000.ethernet: IRQ eth_lpi not found
Mon Feb  3 23:36:32 2025 kern.info kernel: [    0.416348] rk_gmac-dwmac fe300000.ethernet: PTP uses main clock
Mon Feb  3 23:36:32 2025 kern.info kernel: [    0.441153] rk_gmac-dwmac fe300000.ethernet: IRQ eth_wake_irq not found
Mon Feb  3 23:36:32 2025 kern.info kernel: [    0.441161] rk_gmac-dwmac fe300000.ethernet: IRQ eth_lpi not found
Mon Feb  3 23:36:32 2025 kern.info kernel: [    0.442819] rk_gmac-dwmac fe300000.ethernet: PTP uses main clock
Mon Feb  3 23:36:32 2025 kern.info kernel: [    0.443485] rk_gmac-dwmac fe300000.ethernet: clock input or output? (input).
Mon Feb  3 23:36:32 2025 kern.info kernel: [    0.444105] rk_gmac-dwmac fe300000.ethernet: TX delay(0x28).
Mon Feb  3 23:36:32 2025 kern.info kernel: [    0.444602] rk_gmac-dwmac fe300000.ethernet: RX delay(0x11).
Mon Feb  3 23:36:32 2025 kern.info kernel: [    0.445105] rk_gmac-dwmac fe300000.ethernet: integrated PHY? (no).
Mon Feb  3 23:36:32 2025 kern.info kernel: [    0.445666] rk_gmac-dwmac fe300000.ethernet: clock input from PHY
Mon Feb  3 23:36:32 2025 kern.info kernel: [    0.451209] rk_gmac-dwmac fe300000.ethernet: init for RGMII
Mon Feb  3 23:36:32 2025 kern.info kernel: [    0.452009] rk_gmac-dwmac fe300000.ethernet: User ID: 0x10, Synopsys ID: 0x35
Mon Feb  3 23:36:32 2025 kern.info kernel: [    0.452642] rk_gmac-dwmac fe300000.ethernet:      DWMAC1000
Mon Feb  3 23:36:32 2025 kern.info kernel: [    0.453102] rk_gmac-dwmac fe300000.ethernet: DMA HW capability register supported
Mon Feb  3 23:36:32 2025 kern.info kernel: [    0.453780] rk_gmac-dwmac fe300000.ethernet: RX Checksum Offload Engine supported
Mon Feb  3 23:36:32 2025 kern.info kernel: [    0.454436] rk_gmac-dwmac fe300000.ethernet: COE Type 2
Mon Feb  3 23:36:32 2025 kern.info kernel: [    0.454895] rk_gmac-dwmac fe300000.ethernet: TX Checksum insertion supported
Mon Feb  3 23:36:32 2025 kern.info kernel: [    0.455510] rk_gmac-dwmac fe300000.ethernet: Wake-Up On Lan supported
Mon Feb  3 23:36:32 2025 kern.info kernel: [    0.456074] rk_gmac-dwmac fe300000.ethernet: Normal descriptors
Mon Feb  3 23:36:32 2025 kern.info kernel: [    0.457389] rk_gmac-dwmac fe300000.ethernet: Ring mode enabled
Mon Feb  3 23:36:32 2025 kern.info kernel: [    0.457899] rk_gmac-dwmac fe300000.ethernet: Enable RX Mitigation via HW Watchdog Timer
Mon Feb  3 23:36:32 2025 kern.err kernel: [   17.653936] leds enp1s0-0::lan: Setting an LED's brightness failed (-524)
Mon Feb  3 23:36:32 2025 kern.err kernel: [   17.724357] leds enp1s0-1::lan: Setting an LED's brightness failed (-524)
Mon Feb  3 23:36:32 2025 kern.info kernel: [   17.764094] r8169 0000:01:00.0 eth1: RTL8168h/8111h, da:a0:67:80:58:d2, XID 541, IRQ 75
Mon Feb  3 23:36:32 2025 kern.info kernel: [   17.764833] r8169 0000:01:00.0 eth1: jumbo features [frames: 9194 bytes, tx checksumming: ko]
Mon Feb  3 23:36:32 2025 kern.err kernel: [   17.773908] leds enp1s0-2::lan: Setting an LED's brightness failed (-524)
Mon Feb  3 23:36:33 2025 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Mon Feb  3 23:36:34 2025 user.notice : Added device handler type: macvlan
Mon Feb  3 23:36:34 2025 user.notice : Added device handler type: veth
Mon Feb  3 23:36:34 2025 user.notice : Added device handler type: Network device
Mon Feb  3 23:36:34 2025 user.notice ucitrack: Setting up /etc/config/dhcp reload dependency on /etc/config/network
Mon Feb  3 23:36:34 2025 daemon.notice netifd: Interface 'wan' is enabled
Mon Feb  3 23:36:34 2025 daemon.notice netifd: Interface 'wan' is setting up now
Mon Feb  3 23:36:34 2025 kern.info kernel: [   21.043640] r8169 0000:01:00.0 eth1: Link is Down
Mon Feb  3 23:36:34 2025 kern.info kernel: [   21.044249] br-lan: port 1(eth1) entered blocking state
Mon Feb  3 23:36:34 2025 kern.info kernel: [   21.044734] br-lan: port 1(eth1) entered disabled state
Mon Feb  3 23:36:34 2025 kern.info kernel: [   21.045236] r8169 0000:01:00.0 eth1: entered allmulticast mode
Mon Feb  3 23:36:34 2025 kern.info kernel: [   21.046058] r8169 0000:01:00.0 eth1: entered promiscuous mode
Mon Feb  3 23:36:34 2025 kern.info kernel: [   21.048843] br-lan: port 1(eth1) entered blocking state
[... cut ...]
[   21.124685] rk_gmac-dwmac fe300000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211E Gigabit Ethernet] (irq=67)
[   21.134050] rk_gmac-dwmac fe300000.ethernet eth0: No Safety Features support found
[   21.134745] rk_gmac-dwmac fe300000.ethernet eth0: PTP not supported by HW
[   21.136434] rk_gmac-dwmac fe300000.ethernet eth0: configuring for phy/rgmii link mode
[   21.753890] br-lan: port 1(eth1) entered disabled state
[   22.777238] rk_gmac-dwmac fe300000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
root@OpenWrt:~#

hvh-blet, 2025-02-12:
after reboot:
root@OpenWrt:~# logread | grep -i 'network\|eth\|wan\|lan\|link'
Mon Feb  3 23:39:03 2025 kern.info kernel: [    0.000000] psci: probing for conduit method from DT.
Mon Feb  3 23:39:03 2025 kern.info kernel: [    0.046544] NET: Registered PF_NETLINK/PF_ROUTE protocol family
Mon Feb  3 23:39:03 2025 kern.info kernel: [    0.263893] rk_gmac-dwmac fe300000.ethernet: IRQ eth_wake_irq not found
Mon Feb  3 23:39:03 2025 kern.info kernel: [    0.264490] rk_gmac-dwmac fe300000.ethernet: IRQ eth_lpi not found
Mon Feb  3 23:39:03 2025 kern.info kernel: [    0.265140] rk_gmac-dwmac fe300000.ethernet: PTP uses main clock
Mon Feb  3 23:39:03 2025 kern.info kernel: [    0.345958] 8021q: 802.1Q VLAN Support v1.8
Mon Feb  3 23:39:03 2025 kern.info kernel: [    0.402270] rk_gmac-dwmac fe300000.ethernet: IRQ eth_wake_irq not found
Mon Feb  3 23:39:03 2025 kern.info kernel: [    0.402867] rk_gmac-dwmac fe300000.ethernet: IRQ eth_lpi not found
Mon Feb  3 23:39:03 2025 kern.info kernel: [    0.403613] rk_gmac-dwmac fe300000.ethernet: PTP uses main clock
Mon Feb  3 23:39:03 2025 kern.info kernel: [    0.416502] rk_gmac-dwmac fe300000.ethernet: IRQ eth_wake_irq not found
Mon Feb  3 23:39:03 2025 kern.info kernel: [    0.417107] rk_gmac-dwmac fe300000.ethernet: IRQ eth_lpi not found
Mon Feb  3 23:39:03 2025 kern.info kernel: [    0.417808] rk_gmac-dwmac fe300000.ethernet: PTP uses main clock
Mon Feb  3 23:39:03 2025 kern.info kernel: [    0.443239] rk_gmac-dwmac fe300000.ethernet: IRQ eth_wake_irq not found
Mon Feb  3 23:39:03 2025 kern.info kernel: [    0.443894] rk_gmac-dwmac fe300000.ethernet: IRQ eth_lpi not found
Mon Feb  3 23:39:03 2025 kern.info kernel: [    0.444603] rk_gmac-dwmac fe300000.ethernet: PTP uses main clock
Mon Feb  3 23:39:03 2025 kern.info kernel: [    0.445345] rk_gmac-dwmac fe300000.ethernet: clock input or output? (input).
Mon Feb  3 23:39:03 2025 kern.info kernel: [    0.445975] rk_gmac-dwmac fe300000.ethernet: TX delay(0x28).
Mon Feb  3 23:39:03 2025 kern.info kernel: [    0.446481] rk_gmac-dwmac fe300000.ethernet: RX delay(0x11).
Mon Feb  3 23:39:03 2025 kern.info kernel: [    0.446995] rk_gmac-dwmac fe300000.ethernet: integrated PHY? (no).
Mon Feb  3 23:39:03 2025 kern.info kernel: [    0.447584] rk_gmac-dwmac fe300000.ethernet: clock input from PHY
Mon Feb  3 23:39:03 2025 kern.info kernel: [    0.453136] rk_gmac-dwmac fe300000.ethernet: init for RGMII
Mon Feb  3 23:39:03 2025 kern.info kernel: [    0.454273] rk_gmac-dwmac fe300000.ethernet: User ID: 0x10, Synopsys ID: 0x35
Mon Feb  3 23:39:03 2025 kern.info kernel: [    0.455281] rk_gmac-dwmac fe300000.ethernet:      DWMAC1000
Mon Feb  3 23:39:03 2025 kern.info kernel: [    0.455749] rk_gmac-dwmac fe300000.ethernet: DMA HW capability register supported
Mon Feb  3 23:39:03 2025 kern.info kernel: [    0.456411] rk_gmac-dwmac fe300000.ethernet: RX Checksum Offload Engine supported
Mon Feb  3 23:39:03 2025 kern.info kernel: [    0.457073] rk_gmac-dwmac fe300000.ethernet: COE Type 2
Mon Feb  3 23:39:03 2025 kern.info kernel: [    0.457541] rk_gmac-dwmac fe300000.ethernet: TX Checksum insertion supported
Mon Feb  3 23:39:03 2025 kern.info kernel: [    0.458164] rk_gmac-dwmac fe300000.ethernet: Wake-Up On Lan supported
Mon Feb  3 23:39:03 2025 kern.info kernel: [    0.458734] rk_gmac-dwmac fe300000.ethernet: Normal descriptors
Mon Feb  3 23:39:03 2025 kern.info kernel: [    0.459259] rk_gmac-dwmac fe300000.ethernet: Ring mode enabled
Mon Feb  3 23:39:03 2025 kern.info kernel: [    0.459776] rk_gmac-dwmac fe300000.ethernet: Enable RX Mitigation via HW Watchdog Timer
Mon Feb  3 23:39:03 2025 kern.err kernel: [    1.086199] rockchip-pcie f8000000.pcie: PCIe link training gen1 timeout!
Mon Feb  3 23:39:04 2025 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Mon Feb  3 23:39:04 2025 user.notice : Added device handler type: macvlan
Mon Feb  3 23:39:04 2025 user.notice : Added device handler type: veth
Mon Feb  3 23:39:04 2025 user.notice : Added device handler type: Network device
Mon Feb  3 23:39:05 2025 kern.info kernel: [   20.503081] rk_gmac-dwmac fe300000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
Mon Feb  3 23:39:05 2025 kern.info kernel: [   20.564728] rk_gmac-dwmac fe300000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211E Gigabit Ethernet] (irq=67)
Mon Feb  3 23:39:05 2025 kern.info kernel: [   20.573916] rk_gmac-dwmac fe300000.ethernet eth0: No Safety Features support found
Mon Feb  3 23:39:05 2025 daemon.notice netifd: Interface 'lan' is enabled
Mon Feb  3 23:39:05 2025 daemon.notice netifd: Interface 'lan' is setting up now
Mon Feb  3 23:39:05 2025 daemon.notice netifd: Interface 'lan' is now up
Mon Feb  3 23:39:05 2025 kern.info kernel: [   20.574626] rk_gmac-dwmac fe300000.ethernet eth0: PTP not supported by HW
Mon Feb  3 23:39:05 2025 kern.info kernel: [   20.576631] rk_gmac-dwmac fe300000.ethernet eth0: configuring for phy/rgmii link mode
Mon Feb  3 23:39:05 2025 daemon.notice netifd: Network device 'lo' link is up
Mon Feb  3 23:39:05 2025 daemon.notice netifd: Interface 'loopback' has link connectivity
Mon Feb  3 23:39:05 2025 user.notice firewall: Reloading firewall due to ifup of lan (eth0)
Mon Feb  3 23:39:05 2025 user.notice ucitrack: Setting up /etc/config/dhcp reload dependency on /etc/config/network
Mon Feb  3 23:39:05 2025 user.notice ucitrack: Setting up /etc/config/network reload dependency on /etc/config/wireless
[... cut ...]
[   20.503081] rk_gmac-dwmac fe300000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   20.564728] rk_gmac-dwmac fe300000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211E Gigabit Ethernet] (irq=67)
[   20.573916] rk_gmac-dwmac fe300000.ethernet eth0: No Safety Features support found
[   20.574626] rk_gmac-dwmac fe300000.ethernet eth0: PTP not supported by HW
[   20.576631] rk_gmac-dwmac fe300000.ethernet eth0: configuring for phy/rgmii link mode
[   22.211067] rk_gmac-dwmac fe300000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
root@OpenWrt:~#

hvh-blet, 2025-02-12:
> using a Raspiberry Pi 15w power supply

I use it too!
microsd, sandisk 32 GB, netac 8 GB, 16 GB, kingston 64 GB

SuPeR-NoVa-UK, 2025-02-12:
I also have the same issue with the LAN disconnecting on boot, I tried FriendlyWRT out of curiosity and I’ve not encountered the issue once, Do you think the devs will get this fixed in the future for OpenWRT?

hvh-blet, 2025-02-13:
I really hope they fix it! I'm already thinking about buying a new R4S, but I'm afraid that it might have the same bug.

hvh-blet, 2025-02-13:
> So is anybody able to test the fix on openwrt? That would be step one to getting the patch included.

Fixed 24.10.0??? Does it even exist?

Ramon00, 2025-02-13:
Scroll all the up, click on the link with the fix. There is a patch. Not sure if it can be directly applied or not to openwrt

1715173329, 2025-02-13:
Please report this issue to [upstream mailing list](https://lore.kernel.org/linux-rockchip/): linux-rockchip@lists.infradead.org

wurzerj, 2025-02-15:
@Ramon00 Thanks for submitting it (https://lore.kernel.org/linux-rockchip/CAG-UM5EbD7eJPO7+rRnZsV4mPfeFbtGmz18+23P1ZJgdqpFdAg@mail.gmail.com/T/#u). I really hope a kernel developer picks it up.

Ramon00, 2025-02-19:
I am afraid this is not going to be picked up any time soon. Can we just include the patch in openwrt?

Ramon00, 2025-02-19:
or is the most feasible thing, just setting up a build environment myself?

wurzerj, 2025-02-19:
I’ll submit a pull request for OpenWrt as soon as I have time, hoping it will get merged. While it would be ideal to fix it directly in the Linux kernel, I am also afraid this might take ages.

Ramon00, 2025-02-23:
@wurzerj Thx!

imebeh, 2025-02-23:
same issue on R4SE, but I find the pcie device [10ec:8168] in the bootlog, try to install `kmod-r8168`, then fixed.
 ps: `kmod-r8169` is packed into openwrt 24.10

hvh-blet, 2025-02-23:
> install `kmod-r8168`

I've already tried this, it doesn't help!

Ramon00, 2025-02-23:
@imebeh  I think both support more or less the same HW, one is vendor maintained the other kernel maintained.  So one may be better than the other in specific cases.

imebeh, 2025-02-23:
> I've already tried this, it doesn't help!

yes, you are right, after I install the board into its case, this issue is back.

then I tried some combinations, and find out when the CPU temperature is upper 45°, reboot is safe, lan will work well🥲

Ramon00, 2025-02-23:
maybe if the temperature is high then it loses state at reboot then is fine? Mine is running at 25-26 degC typically, so that could explain it maybe

closed 2025-04-23 by Ramon00

dmsza, 2025-05-25:
**Why this PR has been closed?** There are **many** R4S users with this problem. It is not a hardware nor a temperature issue, it is a **software** defect that needs more investigation.

For example, I did many tests with the vendor's build (FriendlyWrt 24.01.1) and it works perfectly fine (eht1 is recognized fine, reboot works everytime). I suspect it might be related to the bootloader, since **I tested the OpenWrt image generated during the FriendlyWrt build, and has the same issues as the official OpenWrt image.** But FriendlyWrt build does not use the img generated by OpenWrt. Instead it gets all contents from `/build_dir/target-aarch64_generic_musl/root-rockchip` generated by OpenWrt build and packages it in a different image format with a different bootloader and kernel.

Again, I am not saying FriendlyWrt  is better, and honestly I do not want to use it. But right now it is the only way to have one of my R4S units operating reliably until this issue is fixed in OpenWrt.

I do not have enough knowledge to further investigate this problem, but I am willing to do any test and since I have two R4S units that have different behaviors with official OpenWrt.