openwrt/openwrt #17932: OpenWrt 24.10 on Raspberry CM4 with Expanding root partition and filesystem fails to boot

santi-wg, 2025-02-10:
### Describe the bug

Install OpenWrt 24.10 on my Raspberry pi CM4 boot sucess, but when I expand root partition and filesystem (https://openwrt.org/docs/guide-user/advanced/expand_root), my pi not boot (no dhcp ip)

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

bcm27xx/bcm2711

### Device

Raspberry Pi 4B/400/CM4 (64bit)

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

reedmideke, 2025-02-12:
I encountered this on a RPI 3B+

The underlying problem appears to be that resizing the partition with `parted` (as suggested on the wiki pages in the previous reply) replaces the disk GUID with an unpredicable value. Then when you reboot, cmdline.txt still has `root=PARTUUID=<old UUID>-<part number>` and the pi hangs on boot because it can't find the boot partition.

confirmed by running `lsblk -o +UUID,PARTUUID` before and after `parted /dev/mmcblk0 resizepart 2 8519679s`

I did not see this behavior with parted on 25.03.

Resizing the partition with sfdisk like `echo ',4GiB,,' | sfdisk --no-reread -N 2 /dev/mmcblk0` and then doing everything else as suggested on the wiki appears to avoid the problem. Note that sfdisk does not update the kernel view of the partition, so lsblk doesn't see the partition size change until you reboot.

jamesorlakin, 2025-04-05:
I too ran into this issue, even by using gParted on another machine to expand the partition a bit before first boot. Now of course this is backed by parted so I'm not surprised the behaviour is the same where the partition UUID seems to change. I had to edit `cmdline.txt` and `partuuid.txt` on the SD card to match the new value before the Pi would boot.

But what did surprise me is using the OpenWrt script _again_ to fully expand the partition worked fine. parted seemed to not change the partition UUID the second time around.

I'm at a bit of a loss as to _why_ this might be though. Could there be something subtly corrupt about the OpenWrt-created partition that parted regenerates as a one off when it interacts with it?

tedaz, 2025-04-24:
I'm encountering the same issue with the Raspberry Pi 4B. The root cause is that, starting with OpenWrt 23.05.05, the /boot/cmdline.txt file uses PARTUUID instead of the legacy /dev/mmcblk0p2. When you create a third partition on the TF card or expand the existing second partition (root partition), the UUID remains unchanged, but the PARTUUID changes, leading to a boot failure on the Raspberry Pi.
If you modify /boot/cmdline.txt to use UUID instead, the Raspberry Pi enters a boot loop and fails to start successfully.

I'm seeking a solution to use the buildroot to create the image without the PARTUUID feature.

namchuai, 2025-04-29:
I'm encounter this issue also. So frustrated. I thought I did something wrong and flashing over and over again..

I followed this guide to increase the size of my rootfs: https://openwrt.org/docs/guide-user/advanced/expand_root#automated

meehatpa, 2025-05-03:
I've noticed that `blkid` does not come installed in the base image. Before running `expand-root.sh` and rebooting, make sure you have installed `blkid`.

tedaz, 2025-07-16:
> https://github.com/rahulelex/resize-storage-on-openwrt-raspberry-pi
> 
> see this way

no need that complex, as that is just change the settings from PARTUUID to legacy /dev/mmcblk.

you can do the change by just modify the cobfig file.

the root resolution is change the source code to use legacy /dev/mmcblk.