openwrt/openwrt #17934: FriendlyARM NanoPi R4S not bootable

wolvth, 2025-02-10:
### Describe the bug

FriendlyARM NanoPi R4S 1GB not bootable，Only solid red led.
https://github.com/openwrt/openwrt/pull/3701#issue-771390517

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

rockchip/armv8

### Device

FriendlyElec NanoPi R4S 1GB

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

nshopik, 2025-02-10:
The R4S model with 1GB RAM and the R4SE with onboard 32GB eMMC are not officially supported however have custom builds available.
https://openwrt.org/toh/friendlyarm/nanopi_r4s_v1

wolvth, 2025-02-10:
> 具有 1GB RAM 的 R4S 型号和具有板载 32GB eMMC 的 R4SE 不受官方支持，但有可用的自定义版本。 https://openwrt.org/toh/friendlyarm/nanopi_r4s_v1

I know, but I want to use the official OP and hope to support it. The old version used to be fine, but I don't know why it's not working now.

Ramon00, 2025-02-11:
Then make it a request not a bug report