openwrt/openwrt #17936: reboot crash on 24.10.0@Linksys WHW03 V2

Jackie264, 2025-02-10:
### Describe the bug

I've found that this firmware version has a reboot and crash issue. Whether I use the 'reboot' command or the LuCI interface to perform a reboot, the system is always unable to connect after restarting. I can only unplug the power or use the power switch on the router to cold boot it to restore normal operation. Both of my two Linksys WHW03 V2 (Velop) units have the same issue. 

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

ipq40xx/generic

### Device

Linksys WHW03 V2 (Velop)

### Image kind

Official downloaded image

### Steps to reproduce

Every time 'reboot' command or the LuCI interface to perform a reboot

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-02-10:
Do you have serial access for backtrace of crash?

Jackie264, 2025-02-11:
> Do you have serial access for backtrace of crash?

I'm sorry, I don't have the means to test a serial console connection. I'm wondering if the kernel log I'm providing below might be of any help?

<6>[    0.000000] Booting Linux on physical CPU 0x0
<5>[    0.000000] Linux version 6.6.73 (builder@buildhost) (arm-openwrt-linux-muslgnueabi-gcc (OpenWrt GCC 13.3.0 r28427-6df0e3d02a) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Mon Feb  3 23:09:37 2025
<6>[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
<6>[    0.000000] CPU: div instructions available: patching division code
<6>[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
<6>[    0.000000] OF: fdt: Machine model: Linksys WHW03 V2 (Velop)
<6>[    0.000000] Memory policy: Data cache writealloc
<6>[    0.000000] OF: reserved mem: 0x87e00000..0x87e7ffff (512 KiB) nomap non-reusable smem@87e00000
<6>[    0.000000] OF: reserved mem: 0x87e80000..0x87ffffff (1536 KiB) nomap non-reusable tz@87e80000
<6>[    0.000000] Zone ranges:
<6>[    0.000000]   Normal   [mem 0x0000000080000000-0x000000009fffffff]
<6>[    0.000000]   HighMem  empty
<6>[    0.000000] Movable zone start for each node
<6>[    0.000000] Early memory node ranges
<6>[    0.000000]   node   0: [mem 0x0000000080000000-0x0000000087dfffff]
<6>[    0.000000]   node   0: [mem 0x0000000087e00000-0x0000000087ffffff]
<6>[    0.000000]   node   0: [mem 0x0000000088000000-0x000000009fffffff]
<6>[    0.000000] Initmem setup node 0 [mem 0x0000000080000000-0x000000009fffffff]
<6>[    0.000000] percpu: Embedded 13 pages/cpu s21524 r8192 d23532 u53248
<7>[    0.000000] pcpu-alloc: s21524 r8192 d23532 u53248 alloc=13*4096
<7>[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
<5>[    0.000000] Kernel command line: init=/sbin/init rootfstype=ubifs ubi.mtd=alt_rootfs root=ubi0:ubifs rootwait rw root=/dev/ubiblock0_0 rootfstype=squashfs ro
<6>[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
<6>[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
<6>[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129920
<6>[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
<6>[    0.000000] Memory: 505776K/524288K available (7663K kernel code, 643K rwdata, 1136K rodata, 1024K init, 248K bss, 18512K reserved, 0K cma-reserved, 0K highmem)
<6>[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
<6>[    0.000000] rcu: Hierarchical RCU implementation.
<6>[    0.000000] 	Tracing variant of Tasks RCU enabled.
<6>[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
<6>[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
<6>[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
<6>[    0.000000] arch_timer: cp15 timer(s) running at 48.00MHz (virt).
<6>[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb11fd3bfb, max_idle_ns: 440795203732 ns
<6>[    0.000001] sched_clock: 56 bits at 48MHz, resolution 20ns, wraps every 4398046511096ns
<6>[    0.000018] Switching to timer-based delay loop, resolution 20ns
<6>[    0.000332] Calibrating delay loop (skipped), value calculated using timer frequency.. 96.00 BogoMIPS (lpj=480000)
<6>[    0.000357] CPU: Testing write buffer coherency: ok
<6>[    0.000408] pid_max: default: 32768 minimum: 301
<6>[    0.010291] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
<6>[    0.010316] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
<6>[    0.018838] qcom_scm: convention: smc legacy
<6>[    0.020314] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=4.
<6>[    0.020491] Setting up static identity map for 0x80300000 - 0x8030003c
<6>[    0.020699] rcu: Hierarchical SRCU implementation.
<6>[    0.020706] rcu: 	Max phase no-delay instances is 1000.
<6>[    0.021399] smp: Bringing up secondary CPUs ...
<6>[    0.025093] smp: Brought up 1 node, 4 CPUs
<6>[    0.025117] SMP: Total of 4 processors activated (384.00 BogoMIPS).
<6>[    0.025127] CPU: All CPU(s) started in SVC mode.
<6>[    0.034109] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
<6>[    0.034291] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
<6>[    0.034320] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
<6>[    0.039443] pinctrl core: initialized pinctrl subsystem
<6>[    0.042767] NET: Registered PF_NETLINK/PF_ROUTE protocol family
<6>[    0.043306] DMA: preallocated 256 KiB pool for atomic coherent allocations
<6>[    0.044717] thermal_sys: Registered thermal governor 'step_wise'
<6>[    0.044822] cpuidle: using governor ladder
<6>[    0.044865] cpuidle: using governor menu
<6>[    0.063376] cryptd: max_cpu_qlen set to 1000
<6>[    0.067998] usbcore: registered new interface driver usbfs
<6>[    0.068056] usbcore: registered new interface driver hub
<6>[    0.068142] usbcore: registered new device driver usb
<6>[    0.068195] pps_core: LinuxPPS API ver. 1 registered
<6>[    0.068203] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[... 222 lines cut ...]
<4>[   23.696902] ath10k_ahb a800000.wifi: msdu-desc: 2500  skid: 32**
[... cut ...]
<6>[   32.035984] br-lan: port 1(lan) entered disabled state
<6>[   32.041157] qca8k-ipq4019 c000000.switch lan: entered allmulticast mode
<6>[   32.046142] ipqess-edma c080000.ethernet eth0: entered allmulticast mode
<6>[   32.053519] qca8k-ipq4019 c000000.switch lan: entered promiscuous mode
<6>[   34.151756] qca8k-ipq4019 c000000.switch lan: Link is Up - 1Gbps/Full - flow control rx/tx
<6>[   34.153039] br-lan: port 1(lan) entered blocking state
<6>[   34.159004] br-lan: port 1(lan) entered forwarding state

brada4, 2025-02-11:
Does not help to determine towards your claim  that linux crashes after reboot. Maybe reboot does not hapen - can you still ping it from wired lan minute after reboot command? Ie no rebot occurs?

Jackie264, 2025-02-11:
> Does not help to determine towards your claim that linux crashes after reboot. Maybe reboot does not hapen - can you still ping it from wired lan minute after reboot command? Ie no rebot occurs?

[image]

3 minutes after reboot
ping from linux
[image]

and
ping from my Mac
[image]

Jackie264, 2025-02-19:
After the recent few opkg upgrades, this issue seems to be resolved, but I don't know exactly which package or packages were updated to fix it.

closed 2025-05-08 by Jackie264