openwrt/openwrt #17938: No Radios / Fatal Kernel error

ExarKunX, 2025-02-10:
### Describe the bug

After installing standard image for RAX120v2, 

`
Mon Feb  3 23:13:00 2025 kern.err kernel: [  215.118818] DalChipInfo.c:289 Chip Major Version mismatch between Hardware:0x1 and Software:0x2
Mon Feb  3 23:13:00 2025 kern.err kernel: [  215.118818] param0 :1, param1 :2, param2 :zero.
Mon Feb  3 23:13:00 2025 kern.err kernel: [  215.118818] Thread ID      : 0x00000078  Thread name    : sys_m_smsm  Process ID     : 0
Mon Feb  3 23:13:00 2025 kern.err kernel: [  215.118818] Register:
Mon Feb  3 23:13:00 2025 kern.err kernel: [  215.118818] SP : 0x4b5bf650
Mon Feb  3 23:13:00 2025 kern.err kernel: [  215.118818] FP : 0x4b5bf658
Mon Feb  3 23:13:00 2025 kern.err kernel: [  215.118818] PC : 0x4b107f58
Mon Feb  3 23:13:00 2025 kern.err kernel: [  215.118818] SSR : 0x00000008
Mon Feb  3 23:13:00 2025 kern.err kernel: [  215.118818] BADVA : 0x00020000
Mon Feb  3 23:13:00 2025 kern.err kernel: [  215.118818] LR : 0x4b1076f4
Mon Feb  3 23:13:00 2025 kern.err kernel: [  215.118818]
Mon Feb  3 23:13:00 2025 kern.err kernel: [  215.118818] Stack Dump
Mon Feb  3 23:13:00 2025 kern.err kernel: [  215.118818] from : 0x4b5bf650
Mon Feb  3 23:13:00 2025 kern.err kernel: [  215.118818] to   : 0x4b5bf790
Mon Feb  3 23:13:00 2025 kern.err kernel: [  215.118818]
Mon Feb  3 23:13:00 2025 kern.err kernel: [  215.177249] remoteproc remoteproc0: crash detected in cd00000.q6v5_wcss: type fatal error
Mon Feb  3 23:13:00 2025 kern.err kernel: [  215.199486] remoteproc remoteproc0: handling crash #422 in cd00000.q6v5_wcss
Mon Feb  3 23:13:00 2025 kern.err kernel: [  215.207547] remoteproc remoteproc0: recovering cd00000.q6v5_wcss
Mon Feb  3 23:13:00 2025 kern.info kernel: [  215.250043] remoteproc remoteproc0: stopped remote processor cd00000.q6v5_wcss
`

### OpenWrt version

24.10.0

### OpenWrt release

r28427-6df0e3d02a

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Netgear RAX120v2

### Image kind

Official downloaded image

### Steps to reproduce

Just install using standard factory image using OEM Web page, after booting (that takes like 5 min), in LuCI No Wireless menu 

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

kirdesde, 2025-02-11:
Please provide the full kernel  log (dmesg)

ExarKunX, 2025-02-11:
> Invalid Release reported. `r28427-6df0e3d02a` Is this from a clean repository?

Yes, 

[image]

ExarKunX, 2025-02-11:
> Please provide the full kernel log (dmesg)

Here it is: 

[kernel.log](https://github.com/user-attachments/files/18749279/kernel.log)

kirdesde, 2025-02-11:
`DalChipInfo.c:289 Chip Major Version mismatch between Hardware:0x1 and Software:0x2`

 indicates this is SOC version 1 not 2.

Please provide `hexdump -C /proc/device-tree/soc_version_major`

ExarKunX, 2025-02-11:
> No that only contains the crash logs

I got this file running  `logread > /tmp/kernel.log`

the one before, was with `dmseg` but they are pretty much the same, don't know what else can I do to help you .

Thanks anyway
[kernel.log](https://github.com/user-attachments/files/18758167/kernel.log)

kirdesde, 2025-02-11:
> > `DalChipInfo.c:289 Chip Major Version mismatch between Hardware:0x1 and Software:0x2`
> > indicates this is SOC version 1 not 2.
> > Please provide `hexdump -C /proc/device-tree/soc_version_major`
> 
> ```
> root@OpenWrt:~# hexdump -C /proc/device-tree/soc_version_major
> 00000000  01 00 00 00                                       |....|
> 00000004
> root@OpenWrt:~# 
> ```

That means this is SOC version 1, that is not supported by ath11k.
Only version 2 is supported (RAX120v2)

https://openwrt.org/inbox/toh/netgear/netgear_rax120_nighthawk_ax12

ExarKunX, 2025-02-11:
> That means this is SOC version 1, that is not supported by ath11k. Only version 2 is supported (RAX120v2)
>
> https://openwrt.org/inbox/toh/netgear/netgear_rax120_nighthawk_ax12

Well, I don't know what happened but this used to work until before Release Candidates (Snapshots)