openwrt/openwrt #17940: dnsmasq won't start after sysupgrade -c 23.05.5 -> 24.10.0 - Not a dotted quad (24)

raintonr, 2025-02-11:
### Describe the bug

I previously had latest stable release installed on this device and it was working fine.

However, I noted my target device has limited storage so when 24.10.0 release was available built my own image to have the extra packages I'd like stored in ROM.

This image seems to work great, with the exception of `dnsmasq` which seems to fail due to some kind of configuration incompatibility.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer C7 v4

### Image kind

Self-built image

### Steps to reproduce

Install 23.05.5.
Build 24.10.0 using the config below.
Copy image to target device.
run `sysupgrade -c openwrt-ath79-generic-tplink_archer-c7-v4-squashfs-sysupgrade.bin`

### Actual behaviour

`dnsmasq` won't start.

Log shows:

```
Tue Feb 11 09:47:35 2025 daemon.crit dnsmasq[1]: bad dhcp-range at line 66 of /var/etc/dnsmasq.conf.cfg01411c
```

The offending line in that config is:

```
dhcp-range=set:lan,19,,,1h
```

This config appears to be created when running `/etc/init.d/dnsmasq start`

When trying to start, this appears in the terminal:

```
Not a dotted quad (24)
Not a dotted quad (24)
Not a dotted quad (24)
Not a dotted quad (24)
```

I think it comes out 4 times because `dnsmasq` is supposed to be running on 4 interfaces.

### Expected behaviour

Should just work.

### Diffconfig

```text
CONFIG_TARGET_ath79=y
CONFIG_TARGET_ath79_generic=y
CONFIG_TARGET_MULTI_PROFILE=y
CONFIG_TARGET_DEVICE_ath79_generic_DEVICE_tplink_archer-c7-v4=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_PREFER_CHACHA_OVER_GCM=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_IDEA=y
CONFIG_OPENSSL_WITH_MDC2=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SEED=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_OPENSSL_WITH_WHIRLPOOL=y
CONFIG_PACKAGE_acme=y
CONFIG_PACKAGE_acme-acmesh=y
CONFIG_PACKAGE_acme-acmesh-dnsapi=y
CONFIG_PACKAGE_acme-common=y
CONFIG_PACKAGE_adblock=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_coreutils=y
CONFIG_PACKAGE_coreutils-sort=y
CONFIG_PACKAGE_gawk=y
CONFIG_PACKAGE_ip-full=y
CONFIG_PACKAGE_kmod-crypto-kpp=y
CONFIG_PACKAGE_kmod-crypto-lib-chacha20=y
CONFIG_PACKAGE_kmod-crypto-lib-chacha20poly1305=y
CONFIG_PACKAGE_kmod-crypto-lib-curve25519=y
CONFIG_PACKAGE_kmod-crypto-lib-poly1305=y
CONFIG_PACKAGE_kmod-udptunnel4=y
CONFIG_PACKAGE_kmod-udptunnel6=y
CONFIG_PACKAGE_kmod-wireguard=y
CONFIG_PACKAGE_libatomic=y
CONFIG_PACKAGE_libbpf=y
CONFIG_PACKAGE_libelf=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_libncurses=y
CONFIG_PACKAGE_libopenssl=y
CONFIG_PACKAGE_libopenssl-conf=y
CONFIG_PACKAGE_libpcre2=y
CONFIG_PACKAGE_libreadline=y
CONFIG_PACKAGE_librt=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-acme=y
CONFIG_PACKAGE_luci-app-adblock=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-package-manager=y
CONFIG_PACKAGE_luci-app-pbr=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-lib-uqr=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-proto-wireguard=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_openssh-sftp-server=y
CONFIG_PACKAGE_openssl-util=y
CONFIG_PACKAGE_pbr=y
CONFIG_PACKAGE_resolveip=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rpcsys=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_socat=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_wget-ssl=y
CONFIG_PACKAGE_wireguard-tools=y
CONFIG_PACKAGE_zlib=y
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

raintonr, 2025-02-11:
After a bit of poking around I discovered the `/etc/init.d/dnsmasq` file on the device after image installation differs from `package/network/services/dnsmasq/files/dnsmasq.init` in the `v24.10.0` source tree.

Copying the file from the source fixes the problem and allows `dnsmasq` to start.

How is this possible?

Mosney, 2025-02-14:
Chang your lan dhcp range to some number will alleviate this bug. 

[image]

Believe this is fatal bug cause the default range 100 to 150 will trigger this, see real dnsmasq config file at `/tmp/etc/dnsmasq.conf.cfg01411c` you will see broken line on the bottom.

> dhcp-range=set:lan,19,,255.255.255.0,12h

Mosney, 2025-02-14:
#17455 #17273

ben-31, 2025-08-05:
Hello, 
I think the line `CONFIG_PACKAGE_gawk=y` causes the problem. `ipcalc.sh` uses an AWK syntax not compatible with `gawk`.

As a workaround, you can try this:
`# cd /usr/bin`
`# ln -sf busybox awk`