openwrt/openwrt #17941: Insanely high clock drift on Mikrotik RB911G-2Hpnd

zeno979, 2025-02-11:
### Describe the bug

Mikrotik RB911G-2Hpnd has very high clock drift (more than 20 seconds per minute). sysntpd stops keeping correct after first syncronization.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

ath79/mikrotik

### Device

MikroTik RouterBOARD 911G-(2,5)HPnD

### Image kind

Official downloaded image

### Steps to reproduce

Simply install system

### Actual behaviour

sysntpd can't keep clock syncronized, I've installed chrony configured with these params:
# uci show chrony
chrony.@pool[0]=pool
chrony.@pool[0].hostname='2.openwrt.pool.ntp.org'
chrony.@pool[0].maxpoll='4'
chrony.@pool[0].iburst='yes'
chrony.@dhcp_ntp_server[0]=dhcp_ntp_server
chrony.@dhcp_ntp_server[0].iburst='yes'
chrony.@dhcp_ntp_server[0].disabled='no'
chrony.@allow[0]=allow
chrony.@allow[0].interface='lan'
chrony.@makestep[0]=makestep
chrony.@makestep[0].threshold='1.0'
chrony.@makestep[0].limit='-1'
chrony.@nts[0]=nts
chrony.@nts[0].rtccheck='yes'
chrony.@nts[0].systemcerts='yes'

System clock seems to be very slow, chrony has these complaints:
Tue Feb 11 10:13:12 2025 daemon.warn chronyd[3181]: Frequency -600043.5 ppm exceeds allowed maximum
Tue Feb 11 10:13:12 2025 daemon.warn chronyd[3181]: System clock wrong by 375.780853 seconds
Tue Feb 11 10:13:12 2025 daemon.warn chronyd[3181]: System clock was stepped by 1878.199002 seconds

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

tmn505, 2025-02-11:
Probably this is Your issue: https://git.openwrt.org/?p=openwrt/staging/xback.git;a=commitdiff;h=63921a4df18d970084d11a3c329958bf17e774cc check Your board if that's the case.

zeno979, 2025-02-11:
yes, I've rebuilt with attached patch and now clock is steady. I have some of these boards and all of them has 25mhz crystal

[rb911g-clockref-patch.txt](https://github.com/user-attachments/files/18758580/rb911g-clockref-patch.txt)

tmn505, 2025-02-11:
Pinging @Leo-PL , since he introduced support for that board, maybe both of You can spot differences by which Mikrotik differentiates both variants and applies proper clock (maybe kernel command line option or property in hard_config).

Leo-PL, 2025-02-12:
I have two such boards, both RB911G-5HPnD, both have 25MHz crystal - although - I didn't do that from scratch.
RB911G is a cut-down RB912UAG, which had 40MHz set in its device tree - and it didn't even come to my mind to question that in the first place. And given commit a716ac55649707e8279de6f2ea66c7f6060c982c, it is different after all. So we need to separate that, after all. I'll prepare the fix.

Leo-PL, 2025-02-12:
@tmn505 @zeno979 please review my patch, if possible.

closed 2025-02-12 by openwrt-bot

Leo-PL, 2025-02-13:
@zeno979 please retest once fresh builds appear in snapshot downloads, and let me know if anything more is needed. Unlikely so, but just in case.

PR #17944 (2025-02-12): ath79: mikrotik: fix reference clock of Routerboard 911G
When support for Routerboard 911G was introduced, Routerboad 912UAG device tree was used as a base, and the common part. This led to use of 40MHz as the reference clock frequency for both [1], while RB911G uses 25MHz crystal on the board, causing heavy system clock drift.

Split the definition, and set the reference clock frequency for RB911G back to 25MHz.

Fixes: bcc44b1212b2 ("ath79: support for MikroTik RouterBOARD 911G-(2,5)HPnD")
[1] a716ac556497 ("ath79: fix reference clock for RouterBoard 912UAG")

Closes #17941