openwrt/openwrt #17942: Incorrect traffic counting on the sfp2 interface

root149, 2025-02-11:
### Describe the bug

Incorrect traffic counting on the sfp2 interface

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R3

### Image kind

Self-built image

### Steps to reproduce

build from image builder

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-02-11:
How? ethtool s or ifconfig or nstat or vs other end?

brada4, 2025-02-19:
Which of mentioned kernel counters seems "correct"  to you?  Wrong is how? Like Roman numerals?

root149, 2025-02-19:
i copy from NAS to device on sfp2 2TB, but on web interfase i see transfered 120mb

brada4, 2025-02-19:
Do you use firewall offloads?

root149, 2025-02-19:
yes, hardware offloading

brada4, 2025-02-19:
That bypasses kernel counters. Soft offload for example bypasses bridges leaving impression only at kernel netdev physical ports.

Please check counters in ethtool nstat etc. Then report to ../luci repository that counter from place XYZ is better than current ABC under specific (but common) conditions.

root149, 2025-02-19:
I understood what the problem was and the bug report can be closed

brada4, 2025-02-19:
Closing is in your hands, I still suggest asking for enhancement in web interface if you know which number is good.

closed 2025-02-19 by root149