openwrt/openwrt #17948: YT8521 leeps disconnecting on 24.10.0 (NanoPi R2C Pro, eth0)

pvlcek001, 2025-02-12:
### Describe the bug

I own a NanoPi R2C Pro. Every few minutes to a few hours (dependant on load), eth0 link breaks. It will never turn back on. If you reconnect the ethernet cable, it stays off. Onky thing that helps is to reboot the NanoPi.

### OpenWrt version

24.10.0

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

rockchip/armv8

### Device

NanoPi R2C Plus

### Image kind

Official downloaded image

### Steps to reproduce

Just leave the device on for a time period and return back after. The internet will have stopped working (depends on load on eth0).

### Expected behaviour

The interface should (obviously) stay on indefinitely or until the next restart.

### Additional info

I actually managed to fix the issue (with very high certainty). I installed ethtool and tinkered with the values I can switch on or off. I used the command ethtool -k eth0 to list them. When I use the command **ethtool -K eth0 tx-scatter-gather off** the problem disappears completely. The Nanopi has now been running with no issues for 5 hours with a heavy load. So my question is - how do I make this permanent? I used the local startup script function in openwrt for now. I do not know what or if any option values (and where I should use. Thank you for all you additional information and ideas.
Sincerely, Pavelo

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

pvlcek001, 2025-02-12:
Actually, I am now trxing to also add **ethtool -K eth0 generic-receive-offload off**. Shortly after I was satisfied with the outcome, the eth0 failed again. So I will investigate this further.

brada4, 2025-02-12:
What is in `ethtool -i eth0` ?

You can run commands in /etc/rc.local, but probably first should be reported as driver bug where due ;-)

pvlcek001, 2025-02-12:
> What is in `ethtool -i eth0` ?
> 
> You can run commands in /etc/rc.local, but probably first should be reported as driver bug where due ;-)

root@OpenWrt:~# ethtool -i eth0
driver: st_gmac
version: 6.6.73
firmware-version:
expansion-rom-version:
bus-info: ff540000.ethernet
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: yes
supports-priv-flags: no

Although, I've just had another crash. dmesg in an ssh session shows this

rk_gmac-dwmac ff540000.ethernet eth0: NETDEV WATCHDOG: CPU: 2: transmit queue 0 timed out 5610 ms
[ 5162.933683] rk_gmac-dwmac ff540000.ethernet eth0: Reset adapter.
[ 5163.015812] rk_gmac-dwmac ff540000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[ 5163.097912] rk_gmac-dwmac ff540000.ethernet eth0: PHY [stmmac-0:03] driver [YT8521 Gigabit Ethernet] (irq=POLL)
[ 5163.302697] rk_gmac-dwmac ff540000.ethernet: Failed to reset the dma
[ 5163.303318] rk_gmac-dwmac ff540000.ethernet eth0: stmmac_hw_setup: DMA engine initialization failed
[ 5163.304130] rk_gmac-dwmac ff540000.ethernet eth0: __stmmac_open: Hw setup failed

It does not seem to be overheating or anything

brada4, 2025-02-13:
Set your parameters in /etc/rc.local
ethtool -K eth0 sg off gro off
to avoid network device hanging followed by failed reset

pvlcek001, 2025-02-13:
Thank you. Trying this now. I will update in the morning

pvlcek001, 2025-02-13:
So **ethtool -K eth0 sg off gro off** did not help at all. Still the same behaviour. After couple of hours the interface shuts down (no link). What I found is that prior to this failure the traffic completely and abruptly stops with a couple of eth0 link up/down messages in dmesg. The traffic never recovers and stays at 0.

What I am pondering on now is just setting **ethtool -s eth0 autoneg on speed 100 duplex full**. So far this seems to fix the problem, However I now have only 100 Mb/s on WAN. That limits me a lot. I pay for 250 Mb/s bandwidth.

brada4, 2025-02-13:
Ok, so comment/remove -kK line as red herring.

What is in ethtool -a eth0 and ethtool --show-eee eth0 ?
If you set speed to 1000 what happens?

pvlcek001, 2025-02-13:
root@OpenWrt:~# ethtool -a eth0
Pause parameters for eth0:
Autonegotiate:  on
RX:             off
TX:             off
RX negotiated:  on
TX negotiated:  on

root@OpenWrt:~# ethtool --show-eee eth0
Cannot get EEE settings: Not supported

If I set the speed to 1000 the no-link problem returns. The driver seems flaky somehow on a gigabit speed...

brada4, 2025-02-13:
Pause is negotiated wrong, recheck with `ethtool eth0` "we" should not accept pause negotiated.

`ethtool -s eth0 advertise 0x28` ? (enable 1000 and 100+duplex only)

Something may be logged if one conductor is broken, maybe change cable, like reverse ends of it if possible, or run to a laptop or so.

pvlcek001, 2025-02-13:
OK, set the **ethtool -s eth0 advertise 0x28**. I'll let you know how it went.

pvlcek001, 2025-02-13:
After 3 hours and 21 minutes the disconnection happened again.

brada4, 2025-02-13:
What is being logged in `logread` before disconnect?

pvlcek001, 2025-02-13:
Sorry, I did not save it. There was nothing specific though. Just a simple interface eth0 link-down. Will provide when the issue happens again.

pvlcek001, 2025-02-13:
Still going ok. If the interface does not die, I believe I have a solution. Wanna see if it survives the night. I'll keep you posted.

brada4, 2025-02-13:
One problem is that network processor crashes, other is that it is not able to recover.

pvlcek001, 2025-02-14:
Holding for more than **ten** hours now. I ended up using **ethtool -s eth0 advertise 0x28** and the cpu core assignment for the ethernet adapters from [forum.openwrt.org/t/openwrt-support-for-nanopi-r2c/111620/12](https://forum.openwrt.org/t/openwrt-support-for-nanopi-r2c/111620/12). The latter probably did all the difference. So is it possible that heat spikes were the issue? According to sensors the temperature never exceeded 55 deg C but they could obviously have been lying...

Any idea how to make the code

**#eth0 IRQ on core 0
echo 1 > /proc/irq/35/smp_affinity
#eth1 has no IRQ?
#eth0 queue on core 1
echo 2 > /sys/class/net/eth0/queues/rx-0/rps_cpus
#eth1 queue on core 2
echo 4 > /sys/class/net/eth1/queues/rx-0/rps_cpus**

permanent? I have to enter this each time the Nanopi starts up in an ssh session.
Using /etc/rc.local changes no variables anywhere.

brada4, 2025-02-14:
That looks like exclude eth0 irq from irqbalance and enable steering.
You have to disable both irqbalance and steering for your override commands in /etc/rc.local to make sense.

Since the problem is about irq routing i suspect ethtool to change advertise is not needed.

pvlcek001, 2025-02-14:
Can you please tell me how to disable  that? Or at least where to look. I will study extensively.

pvlcek001, 2025-02-14:
So I have news. If I assign manually anything to CPU1, there is a chance (about 60%) NanoPi stops instantly routing and processing DNS, DHCP works fine. NanoPi reacts to everything, I can edit values, study logs (new log entries are **not** generated). No internet traffic processed whatsoever.

I went to basics, scratched everything and just disabled **packet steering**. Every variable for **/sys/class/net/eth*/queues/rx-0/rps_cpus** is 0 now and the problems are gone.

I do not need to benchmark the speeds, I am sure they will be slower than before. It does not matter, I can never exhaust the processing power in my setup.

pvlcek001, 2025-02-15:
I am going to close this for now because it seems to be power related. I disassembled the nanopi and when using 5V/3A power supply, I am not getting what I shoud get on some rails. I have downloaded datasheets to all the chips inside and some voltages are quite significantly lower than they should be. I will wait for some new information, possibly newer drivers. I even tried 4 power supplies, with some I had more luck but it is still not good enough for production.

closed 2025-02-15 by pvlcek001

brada4, 2025-02-15:
You can try with any modern phone charger which start at 25W

pvlcek001, 2025-02-15:
Sure, I tried with four of them, three GaNs and one ordinary. By the way, I often recommend Mikrotik devices. They have very robust power rail cascades and you can power them with anything that has 6-30V and it runs stable at any given voltage.

pvlcek001, 2025-02-15:
I do not want to raise voltage on the Nanopi. The case specifically says 5 V.

brada4, 2025-02-15:
usb-pd specifically negotiates that. You can use 200W laptop charger, it will not make smaller devices glow like lightbulbs.

pvlcek001, 2025-02-15:
Yeah but the manufacturer specifically states 5 V only and no support for usb-power delivery.

pvlcek001, 2025-02-24:
I have news. Because the unit was basically non-functional, I flashed the newest firmware from the vendor (FriendlyWRT) based on OpenWrt 23.05.5 r24106-10cc5fcd00. From that moment on the issues completely disappeared. It's been running for 5,5 days with zero freezes. So maybe issues with newer kernel? Some patches in drivers? Who knows, the unit is working now and is blazingly fast :)

pvlcek001, 2025-04-02:
I finally got it working. I compared OpenWRT to ImmortalWRT's patches regarding Rockchip architecture. The problem is actually related to the CPU frequency and board power. ImmortalWRT enables the overclocking and I noticed that when you run the CPU on a frequency lower than **816000**, the board might not have stable enough voltage and the bus connected to **eth0** sometimes just stops operating. It can never be revived unless you reboot the device. Also, the **schedutil** governor might be a better choice than **ondemand** because it gives me better performance. This is purely subjective, though.

Ihave set these values in /etc/rc.local:
**echo 816000 > /sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq**
**echo schedutil /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor**

The change to these two parameters is enough to make OpenWRT work stable.

jjm2473, 2025-07-14:
> [ 5162.933683] rk_gmac-dwmac ff540000.ethernet eth0: Reset adapter.
[ 5163.015812] rk_gmac-dwmac ff540000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[ 5163.097912] rk_gmac-dwmac ff540000.ethernet eth0: PHY [stmmac-0:03] driver [YT8521 Gigabit Ethernet] (irq=POLL)
[ 5163.302697] rk_gmac-dwmac ff540000.ethernet: Failed to reset the dma
[ 5163.303318] rk_gmac-dwmac ff540000.ethernet eth0: stmmac_hw_setup: DMA engine initialization failed
[ 5163.304130] rk_gmac-dwmac ff540000.ethernet eth0: __stmmac_open: Hw setup failed

@pvlcek001 Does this problem still occur in 24.10.2? Will executing this command `ip link set dev eth0 down; sleep 1; ip link set dev eth0 up` trigger the bug?

pvlcek001, 2025-07-15:
> [@pvlcek001](https://github.com/pvlcek001) After the failure occurred, did you see this kernel log `rk_gmac-dwmac ff540000.ethernet: Failed to reset the dma`?

Yes but not always. Sometimes logread stops refreshing. I can log into the device but do not see any transfer activity or logging activity. If logread keeps working, I indeed see **Failed to reset the DMA** message.

jjm2473, 2025-07-15:
This patch may fix error recovery, but may not fix TX timeout issues https://github.com/openwrt/openwrt/pull/19415

pvlcek001, 2025-07-15:
If this gets processed, I am willing to test the OpenWrt snapshot.

pvlcek001, 2025-07-15:
So far so good. No hiccups, freezing, disconnecting or any of the sort. I am going to wait a bit longer though. I just want to say that the performance already seems much better than anything I tried before...

jjm2473, 2025-07-16:
istoreos does not overclock the CPU. This is the irq and queue configuration script we use https://github.com/istoreos/istoreos/blob/38ae63309e5d97b28f787a81ae1e71ffe1d477ea/target/linux/rockchip/armv8/base-files/lib/board/init.sh

pvlcek001, 2026-07-03:
Very true. Will eventually report back.