openwrt/openwrt #17959: unbound doesn't appear in Luci->System->Startup

Nopel2020, 2025-02-13:
### Describe the bug

I have unbound installed on 24.10.0. The unbound service is started automatically on boot, and unbound is working fine, but it doesn't appear among the services in Luci->System->Startup.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

rockchip/armv8

### Device

FriendlyElec NanoPi R5C

### Image kind

Official downloaded image

### Additional info

The links in /etc/rc.d exist and do function correctly, but `ubus call rc list` doesn't pick up on them. 

`ubus call luci getInitList` does show the correct values for unbound.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-02-13:
Obvious duplicate of package bug report in package repo

Nopel2020, 2025-02-13:
Sorry, didn't see that bug, and was told in https://forum.openwrt.org/t/24-10-0-unbound-issues-luci-and-ipv4/224578/9 that this bug belonged in the main repo.

Nopel2020, 2025-02-13:
See https://github.com/openwrt/packages/issues/25963

dibdot, 2025-02-14:
Even if the title of this ticket says otherwise: it describes an ubus regression in 24.x and should be handled here!
Example output of `ubus call rc list` (used in LuCI 24.x) - wrong values:
```
[...]
	"unbound": {
		"enabled": false
	},
[...]
```

Example output of `ubus call luci getInitList` (used in LuCI 23.x) - correct values:
```
[...]
	"unbound": {
		"index": 19,
		"stop": 50,
		"enabled": true
	},
[...]
```

@rmilecki Please re-open  - thanks!

vincejv, 2025-02-15:
Issue should be moved here: https://github.com/openwrt/ubus/issues
If it's a ubus issue?