openwrt/openwrt #17962: AX3000T with Foresee NAND: Bad Blocks on 24.10.0, Works on 23.05.5

sc16me, 2025-02-13:
### Describe the bug

Some AX3000T routers with Foresee NAND chips experience issues on OpenWRT 24.10.0. During boot, multiple bad blocks appear. These bad blocks can be erased in U-Boot using nand erase.force, after which flashing the stock firmware or OpenWRT 23.05.5 (with patches for Foresee and AN8855) results in normal operation.

However, reflashing 24.10.0 causes the problem to reappear.

Not all AX3000T routers with Foresee NAND are affected, but the difference between problematic and non-problematic devices remains unclear.

I also tested ubootmod FIP from snapshot, but the issue persists there as well.

Currently, affected routers must be downgraded to 23.05.5, where no issues occur.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

mediatek/filogic

### Device

Xiaomi Mi Router AX3000T

### Image kind

Official downloaded image

### Steps to reproduce

	1.	Flash OpenWRT 24.10.0 on an AX3000T with a Foresee NAND chip.
	2.	Observe multiple bad blocks appearing during boot.
	3.	Erase bad blocks using nand erase.force in U-Boot.
	4.	Flash stock firmware or OpenWRT 23.05.5 and verify normal operation.
	5.	Flash OpenWRT 24.10.0 again and observe the issue returning.
	6.	Flash ubootmod FIP from snapshot and confirm that the problem persists.

### Actual behaviour

On some devices, 24.10.0 repeatedly encounters bad blocks, making the system unstable or unbootable.

### Additional info

	•	This issue is not present on all AX3000T routers with Foresee NAND.
	•	The exact cause of the problem is unknown.
	•	23.05.5 with the necessary patches works without issues.
	•	ubootmod FIP from snapshot does not resolve the issue.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

sc16me, 2025-02-13:
[Logs](https://mega.nz/file/nQJHBAKC#9_dt5ywXHyWfHcs3tYptMi1dkOkWBxobyUkW9t2ui2o)

lov1702, 2025-02-13:
I have the same problem. I have three ax3000t routers with FORESEE F35SQA001G memory and an Airoha AN8855 switch.
When switching from factory firmware to release 24.10.0, bad blocks are formed.

MrC0nst, 2025-02-13:
Same issue, foresee+an8855. Manufacture date 20214.12

AJleKcAHgP68, 2025-02-13:
3 routers. Production date 12.24/12.24/10.24All three have the bad block problem.

user26092024, 2025-02-13:
Actually....

dmsza, 2025-02-13:
I have an AX3000T with AN8855+Foresee, and gladly I am not seeing this issue. But comparing the logs, I noticed that every time my device boots, I am getting the following message:

<code>[    8.744629] UBIFS (ubi0:1): recovery needed
[    8.804850] UBIFS (ubi0:1): recovery completed</code> 

I've seen this also in other devices (such as Redmi AX6S), but after some search I had the understanding that his would be "normal".

However this error is not showing in the device showing bad blocks (see below). Not sure if this could be related, so just reporting.

[image]