openwrt/openwrt #17972: Xiaomi AX3600: Radio 0 not working with ucode wifi-scripts

JuliusBairaktaris, 2025-02-14:
### Describe the bug

Similar to the issues described in #17150, Radio 0 (5GHz) does not work on Qualcommax Xiaomi AX3600

```
Mon Feb 10 22:52:58 2025 daemon.notice netifd: radio0 (2071): wifi-scripts: Starting
Mon Feb 10 22:52:58 2025 daemon.notice hostapd: Set new config for phy phy1:
Mon Feb 10 22:52:58 2025 daemon.notice wpa_supplicant[1881]: Set new config for phy phy1
Mon Feb 10 22:52:58 2025 daemon.notice netifd: radio0 (2071): command failed: Not supported (-95)
Mon Feb 10 22:52:58 2025 kern.info kernel: [   22.339559] ECM init: selected_front_end=1
Mon Feb 10 22:52:58 2025 kern.info kernel: [   22.339662] ECM database jhash random seed: 0x73cddc7a
Mon Feb 10 22:52:58 2025 user.notice root: [ethtool] Disabling feature: rx-gro-list: disabled on (SQM_IFB_3a96d)
Mon Feb 10 22:52:58 2025 kern.info kernel: [   22.467565] ECM init complete
Mon Feb 10 22:52:58 2025 user.notice root: [ethtool] Disabling feature: rx-gro-list: disabled on (br-lan)
Mon Feb 10 22:52:58 2025 user.notice root: [ethtool] Disabling feature: rx-gro-list: disabled on (erspan0)
Mon Feb 10 22:52:58 2025 user.notice root: [ethtool] Disabling feature: rx-gro-list: disabled on (gre0)
Mon Feb 10 22:52:58 2025 user.notice root: [ethtool] Disabling feature: rx-gro-list: disabled on (gretap0)
Mon Feb 10 22:52:58 2025 user.notice root: [ethtool] Disabling feature: rx-gro-list: disabled on (lan1)
Mon Feb 10 22:52:58 2025 user.notice root: [ethtool] Disabling feature: rx-gro-list: disabled on (lan2)
Mon Feb 10 22:52:58 2025 user.notice root: [ethtool] Disabling feature: rx-gro-list: disabled on (lan3)
Mon Feb 10 22:52:58 2025 user.notice root: [ethtool] Disabling feature: rx-gro-list: disabled on (miireg)
Mon Feb 10 22:52:58 2025 user.notice root: [ethtool] Disabling feature: rx-gro-list: disabled on (wan)
Mon Feb 10 22:52:58 2025 daemon.notice procd: /etc/rc.d/S26qca-nss-ecm: net.bridge.bridge-nf-call-arptables = 0
Mon Feb 10 22:52:58 2025 daemon.notice procd: /etc/rc.d/S26qca-nss-ecm: net.bridge.bridge-nf-call-iptables = 0
Mon Feb 10 22:52:58 2025 daemon.notice procd: /etc/rc.d/S26qca-nss-ecm: net.bridge.bridge-nf-call-ip6tables = 0
Mon Feb 10 22:52:58 2025 daemon.notice procd: /etc/rc.d/S26qca-nss-ecm: dev.nss.general.redirect = 1
Mon Feb 10 22:52:58 2025 user.notice root: [ethtool] Disabling feature: rx-gro-list: failed to disable on (SQM_IFB_1357d)
Mon Feb 10 22:52:58 2025 daemon.notice hostapd: Set new config for phy phy1: /var/run/hostapd-phy1.conf
Mon Feb 10 22:52:58 2025 daemon.notice hostapd: Restart interface for phy phy1
Mon Feb 10 22:52:58 2025 user.notice skb_recycler: Enabling skb_recycler
Mon Feb 10 22:52:58 2025 user.notice skb_recycler: Setting max_skbs to 512
Mon Feb 10 22:52:58 2025 user.notice skb_recycler: Setting max_spare_skbs to 128
Mon Feb 10 22:52:58 2025 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=DE ieee80211h=1 ieee80211d=1 spectrum_mgmt_required=0 beacon_int=100 noscan=0 airtime_mode=0 rssi_reject_assoc_rssi=0 rssi_ignore_probe_request=0 acs_exclude_dfs=0 enable_background_radar=0 min_tx_power=0 hw_mode=a ieee80211n=1 ht_coex=0 ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][MAX-AMSDU-7935][DSSS_CCK-40][RX-STBC1] ieee80211ac=1 vht_oper_chwidth=2 vht_oper_centr_freq_seg0_idx=0 vht_capab=[RXLDPC][SHORT-GI-160][TX-STBC-2BY1][SU-BEAMFORMER][SU-BEAMFORMEE][MU-BEAMFORMER][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC1][SOUNDING-DIMENSION-4][BF-ANTENNA-4][MAX-MPDU-7991] he_bss_color=128 he_spr_sr_control=3 ieee80211ax=1 he_oper_chwidth=2 he_oper_centr_freq_seg0_idx=0 he_su_beamformer=1 he_su_beamformee=1 he_mu_beamformer=1 he_twt_required=0 he_default_pe_duration=4 he_rts_threshold=1023 he_mu_edca_qos_info_param_count=
Mon Feb 10 22:52:58 2025 kern.info kernel: [   22.807830] br-lan: port 4(phy1-ap0) entered blocking state
Mon Feb 10 22:52:58 2025 kern.info kernel: [   22.807910] br-lan: port 4(phy1-ap0) entered disabled state
Mon Feb 10 22:52:58 2025 kern.info kernel: [   22.812291] ath11k c000000.wifi phy1-ap0: entered allmulticast mode
Mon Feb 10 22:52:58 2025 kern.info kernel: [   22.819694] ath11k c000000.wifi phy1-ap0: entered promiscuous mode
Mon Feb 10 22:52:58 2025 kern.info kernel: [   22.824233] br-lan: port 4(phy1-ap0) entered blocking state
Mon Feb 10 22:52:58 2025 kern.info kernel: [   22.830365] br-lan: port 4(phy1-ap0) entered forwarding state
Mon Feb 10 22:52:58 2025 daemon.notice hostapd: phy1-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Mon Feb 10 22:52:58 2025 daemon.notice hostapd: ACS: Automatic channel selection started, this may take a bit
Mon Feb 10 22:52:58 2025 daemon.notice hostapd: phy1-ap0: interface state COUNTRY_UPDATE->ACS
Mon Feb 10 22:52:58 2025 daemon.notice hostapd: phy1-ap0: ACS-STARTED
Mon Feb 10 22:52:58 2025 kern.info kernel: [   22.868515] ath11k c000000.wifi phy1-ap0: left allmulticast mode
Mon Feb 10 22:52:58 2025 kern.info kernel: [   22.868640] ath11k c000000.wifi phy1-ap0: left promiscuous mode
Mon Feb 10 22:52:58 2025 kern.info kernel: [   22.874038] br-lan: port 4(phy1-ap0) entered disabled state
Mon Feb 10 22:52:58 2025 user.notice root: [ethtool] Disabling feature: rx-gro-list: failed to disable on (SQM_IFB_95c02)
Mon Feb 10 22:52:59 2025 kern.info kernel: [   22.939544] br-lan: port 4(phy1-ap0) entered blocking state
Mon Feb 10 22:52:59 2025 kern.info kernel: [   22.939599] br-lan: port 4(phy1-ap0) entered disabled state
Mon Feb 10 22:52:59 2025 kern.info kernel: [   22.943990] ath11k c000000.wifi phy1-ap0: entered allmulticast mode
Mon Feb 10 22:52:59 2025 kern.info kernel: [   22.957464] ath11k c000000.wifi phy1-ap0: entered promiscuous mode
Mon Feb 10 22:52:59 2025 kern.info kernel: [   22.957658] br-lan: port 4(phy1-ap0) entered blocking state
Mon Feb 10 22:52:59 2025 kern.info kernel: [   22.962559] br-lan: port 4(phy1-ap0) entered forwarding state
Mon Feb 10 22:52:59 2025 daemon.notice netifd: bridge 'br-lan' link is up
Mon Feb 10 22:52:59 2025 daemon.notice netifd: Interface 'lan' has link connectivity
Mon Feb 10 22:52:59 2025 daemon.notice netifd: Wireless device 'radio0' is now up
Mon Feb 10 22:52:59 2025 daemon.notice netifd: radio0 (2071): wifi-scripts: Configuring 'phy1' txantenna: 4294967295, rxantenna: 4294967295 distance: 0
Mon Feb 10 22:52:59 2025 daemon.notice netifd: radio0 (2071): wifi-scripts: network is not present in the schema
Mon Feb 10 22:52:59 2025 daemon.notice netifd: radio0 (2071): wifi-scripts: Preparing interface: phy1-ap0 with MAC: null
Mon Feb 10 22:52:59 2025 user.notice root: [ethtool] Disabling feature: rx-gro-list: disabled on (phy1-ap0)
Mon Feb 10 22:52:59 2025 daemon.notice netifd: bridge 'br-lan' link is down
Mon Feb 10 22:52:59 2025 daemon.notice netifd: Interface 'lan' has link connectivity loss
Mon Feb 10 22:52:59 2025 kern.info kernel: [   23.117282] br-lan: port 4(phy1-ap0) entered disabled state
Mon Feb 10 22:52:59 2025 user.notice root: [ethtool] Disabling feature: rx-gro-list: failed to disable on (SQM_IFB_f9b7d)
Mon Feb 10 22:52:59 2025 user.notice root: [ethtool] Disabling feature: rx-gro-list: failed to disable on (SQM_IFB_88af0)
Mon Feb 10 22:52:59 2025 auth.info sshd[2681]: Server listening on 0.0.0.0 port 22.
Mon Feb 10 22:52:59 2025 auth.info sshd[2681]: Server listening on :: port 22.
Mon Feb 10 22:52:59 2025 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Mon Feb 10 22:52:59 2025 daemon.err collectd[2946]: plugin_load: plugin "cpu" successfully loaded.
Mon Feb 10 22:52:59 2025 daemon.err collectd[2946]: plugin_load: plugin "interface" successfully loaded.
[... 210 lines cut ...]
### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Xiaomi AX3600

### Image kind

Self-built image

### Steps to reproduce

Use image from: https://github.com/JuliusBairaktaris/Qualcommax_NSS_Builder and try to enable 5G Wifi

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

kirdesde, 2025-02-14:
1. this is an un-official build (due to ecm/nss)
2. What do you mean by radio0 doesn't work?

The last line in the log states that DFS has been started and takes 600s to finish. What happens after that 600s?

JuliusBairaktaris, 2025-02-15:
The 5G radio can not be found by devices. The issue does not appear to be device/built specific: https://forum.openwrt.org/t/qualcommax-nss-build/148529/4405

brada4, 2025-02-15:
I suspect NSS enabled `option background_radar '1'` and you dont see this during continuous run, just once on startup, and wifi maintains dfs status-es in background.

qosmio, 2025-02-15:
@JuliusBairaktaris, did some digging in the ucode for hostapd, the bound checks for channel aren't correctly getting set.

Try this.

```diff
diff --git a/package/network/config/wifi-scripts/files-ucode/usr/share/ucode/wifi/hostapd.uc b/package/network/config/wifi-scripts/files-ucode/usr/share/ucode/wifi/hostapd.uc
index 0e3cfc5b90..83da7b7a19 100644
--- a/package/network/config/wifi-scripts/files-ucode/usr/share/ucode/wifi/hostapd.uc
+++ b/package/network/config/wifi-scripts/files-ucode/usr/share/ucode/wifi/hostapd.uc
@@ -242,7 +242,7 @@ function device_htmode_append(config) {
                                [ 29, 15 ], [ 61, 47 ], [ 93, 79 ], [ 125, 111 ],
                                [ 157, 143 ], [ 189, 175 ], [ 221, 207 ]];
                for (let k, v in vht_oper_centr_freq_seg0_idx_map)
-                       if (v[0] <= config.channel) {
+                       if (config.channel >= (v[0] - 28) && config.channel <= v[0]) {
                                config.vht_oper_centr_freq_seg0_idx = v[1];
                                break;
                        }
```

I'm also seeing sections missing parens with bitwise operators, causing the comparisons to fail causing `vht_capab` capabilities to be incomplete in the final conf.

Primarily these:
```
[MAX-A-MPDU-LEN-EXP7]
[MAX-MPDU-11454]
[RX-STBC-1] <- With the hyphen has a different bit flag in the code for VHT/HE 80 than HT 40
[SHORT-GI-80]
[VHT160]
```

For the full patch including above the following works:

```diff
diff --git a/package/network/config/wifi-scripts/files-ucode/usr/share/schema/wireless.wifi-device.json b/package/network/config/wifi-scripts/files-ucode/usr/share/schema/wireless.wifi-device.json
index 4354b4f816..eb556724f0 100644
--- a/package/network/config/wifi-scripts/files-ucode/usr/share/schema/wireless.wifi-device.json
+++ b/package/network/config/wifi-scripts/files-ucode/usr/share/schema/wireless.wifi-device.json
@@ -635,7 +635,8 @@
 		},
 		"short_gi_80": {
 			"description": "Short GI for 80 MHz",
-			"type": "boolean"
+			"type": "boolean",
+			"default": true"
 		},
 		"spectrum_mgmt_required": {
 			"description": "Set Spectrum Management subfield in the Capability Information field",
@@ -712,7 +713,8 @@
 			"description": "Indicates the maximum length of A-MPDU pre-EOF padding that the STA can recv",
 			"type": "number",
 			"minimum": 0,
-			"maximum": 7
+			"maximum": 7,
+			"default": 7
 		},
 		"vht_max_mpdu": {
 			"description": "Maximum MPDU length",
diff --git a/package/network/config/wifi-scripts/files-ucode/usr/share/ucode/wifi/hostapd.uc b/package/network/config/wifi-scripts/files-ucode/usr/share/ucode/wifi/hostapd.uc
index 0e3cfc5b90..223f84f7d7 100644
--- a/package/network/config/wifi-scripts/files-ucode/usr/share/ucode/wifi/hostapd.uc
+++ b/package/network/config/wifi-scripts/files-ucode/usr/share/ucode/wifi/hostapd.uc
@@ -242,7 +242,7 @@ function device_htmode_append(config) {
 				[ 29, 15 ], [ 61, 47 ], [ 93, 79 ], [ 125, 111 ],
 				[ 157, 143 ], [ 189, 175 ], [ 221, 207 ]];
 		for (let k, v in vht_oper_centr_freq_seg0_idx_map)
-			if (v[0] <= config.channel) {
+			if (config.channel >= (v[0] - 28) && config.channel <= v[0]) {
 				config.vht_oper_centr_freq_seg0_idx = v[1];
 				break;
 			}
@@ -329,7 +329,7 @@ function device_htmode_append(config) {
 			config.vht_capab += '[RX-ANTENNA-PATTERN]';
 		if (vht_capab & 0x20000000 && config.tx_antenna_pattern)
 			config.vht_capab += '[TX-ANTENNA-PATTERN]';
-		let rx_stbc = [ '', '[RX-STBC1]', '[RX-STBC12]', '[RX-STBC123]', '[RX-STBC-1234]' ];
+		let rx_stbc = [ '', '[RX-STBC-1]', '[RX-STBC-12]', '[RX-STBC-123]', '[RX-STBC-1234]' ];
 		config.vht_capab += rx_stbc[min(config.rx_stbc, (vht_capab >> 8) & 7)];
 
 		if (vht_capab & 0x800 && config.su_beamformer)
@@ -338,15 +338,15 @@ function device_htmode_append(config) {
 			config.vht_capab += '[BF-ANTENNA-' + min(((vht_capab >> 13) & 3) + 1, config.beamformer_antennas) + ']';
 
 		/* supported Channel widths */
-		if (vht_capab & 0xc == 8 && config.vht160 <= 2)
+		if ((vht_capab & 0xc) == 8 && config.vht160 <= 2)
 			config.vht_capab += '[VHT160-80PLUS80]';
-		else if (vht_capab & 0xc == 4 && config.vht160 <= 1)
+		else if ((vht_capab & 0xc) == 4 && config.vht160 <= 2)
 			config.vht_capab += '[VHT160]';
 
 		/* maximum MPDU length */
-		if (vht_capab & 3 > 1 && config.vht_max_mpdu > 11454)
+		if ((vht_capab & 3) > 1 && config.vht_max_mpdu >= 11454)
 			config.vht_capab += '[MAX-MPDU-11454]';
-		else if (vht_capab & 3 && config.vht_max_mpdu > 7991)
+		else if ((vht_capab & 3) && config.vht_max_mpdu >= 7991)
 			config.vht_capab += '[MAX-MPDU-7991]';
 
 		/* maximum A-MPDU length exponent */
```

JuliusBairaktaris, 2025-02-16:
@qosmio thanks a lot for the patch. I can confirm that everything works as it should with it applied. Please make a PR.

sppmasterspp, 2025-02-17:
> This isn't NSS related, but affects 160mhz capable devices mostly. Did some digging in the ucode for hostapd, the bound checks for channel aren't correctly getting set.
> 
> Try this.

This patch is OK and wifi now works.
Thanks.

qosmio, 2025-02-18:
@blogic Just pushed https://github.com/openwrt/openwrt/pull/18013

danpawlik, 2025-02-19:
@qosmio hi, nice patch! Just want to ensure, that this line should not be changed: https://github.com/openwrt/openwrt/blob/main/package/network/config/wifi-scripts/files-ucode/usr/share/ucode/wifi/hostapd.uc#L204 ?

closed 2025-11-29 by JuliusBairaktaris