openwrt/openwrt #17979: Sysupgrade from 23.05.5 to 24.20.0 failed on Linksys EA8300

dlretired, 2025-02-15:
Sysupgrade from 23.05.5 to 24.20.0 failed on Linksys EA8300 was configured and working as a dumb access point. The router begins a restart. The connected ethernet port shows traffic then goes blank, then lights sweep across all ports. I believe this happens twice and is normal. The WPS light now flashes orange on top of the unit. Then it goes dark.

To recover, I switch power off and on four consecutive times to swap to the alternate firmware image - 23.05.5.

The issue can be reproduced using LUCI Advance Reboot to boot to the new failing image. Oddly, the two images are identified differently - 

[image]

Notice that the 01 partition does not identify the OpenWrt Release - only the Linux release info.

(Also, “Perform power off” does not work. It does a reboot. Maybe this is hardware releated and should not work on the EA8300)

Kotopies, 2025-02-15:
If you had OpenWrt version 23, you should have "kernsize 500000," but double-check to be sure.
https://openwrt.org/toh/linksys/mr8300#installation_of_23052410_or_snapshot

dlretired, 2025-02-15:
> Kotopies

Yes. That was done before installing 23.05. See clip below:
...
image=dallas.img
imgsize=5800000
ipaddr=192.168.1.1
kernsize=500000
loadaddr=84000000
machid=8010006
...

Kotopies, 2025-02-16:
Connect the USB serial adapter and check what messages you receive during the update to version 24.10.

dlretired, 2025-02-16:
Thanks! I did not realize I could use the USB port for a serial adapter. Now all I need is to find my USB to serial adapter. :-)

dlretired, 2025-02-16:
> Connect the USB serial adapter and check what messages you receive during the update to version 24.10.

I do have a USB to serial adapter. It is 9pin DB RS232 to USB. I used it for console connections to switches and routers. What I didn't realize is that among the half dozen or so computers I have, none have a DB9 serial port. Is there a way I can go USB to USB? I'm assuming it will be RS232?
Thanks!

morepoc, 2025-02-16:
https://a.aliexpress.com/_EvoXoYy

 FT232RL - 5V - 3.3V - 1.8V

dlretired, 2025-02-16:
> https://a.aliexpress.com/_EvoXoYy
> 
> FT232RL - 5V - 3.3V - 1.8V

That's convienent. I had a couple other things from aliexpress I needed an excuse to order.

dlretired, 2025-03-01:
> gamma-577
>  left a comment
> (openwrt/openwrt#17979)
> Hi,
>
> I too had exactly same problem after force flashing OW ver 24.10. Fortunately I had an alt partition running OpenWrt 22.05 ( linux kernel 5.10) made exactly for contingencies like this.
>
> I requested a fw build via attended sysupgrade from primary partition ( OW ver 23.05). Stored the fw file on local laptop. Then took a config backup from luci. Then booted into alt partition and flashed the fw. I opted to preserve settings.
>
> I too was stuck with no access to primary partition. So I switched off and on thrice so as to boot into the alt partition. I then reflashed the same fw without preserving the settings. This worked and the router booted into primary partition. Then I restored the configs from luci config backup. Now everything is working fine. Back to Business As Usual. Yaay! :-)
>
> Learning point : Do not preserve settings while doing attended sysupgrade. Instead take a config backup and restore from that after the flash upgrade.
>
> Thank you.
>
> -Gamma

Thanks! I’ll try that.