openwrt/openwrt #17980: No PoE-OUT after upgrade MikroTik RouterBOARD 760iGS (hEX S) on 24.10

cschil, 2025-02-15:
### Describe the bug

The Hex S is Powered over PoE (802.3af/at) on Port 1 and powers Over PoE-Out at Port 5 an AP.

After upgrade from 23.05. to 24.10. PoE-Out is broken.

I can't get PoE-Out on with:
`uci set system.poe_passthrough.value=1
uci commit`

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

ramips/mt7621

### Device

MikroTik RouterBOARD 760iGS (hEX S)

### Image kind

Official downloaded image

### Steps to reproduce

After Upgrade with or without config PoE-Out not working .

### Actual behaviour

After set
`uci set system.poe_passthrough.value=1
uci commit`
PoE-Out on Port 5 is not working after reboot.

### Expected behaviour

After set
`uci set system.poe_passthrough.value=1
uci commit`
PoE-Out on Port 5 should work after reboot. 

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

joseppc, 2025-02-16:
Seeing the same issue, maybe a warning on the wiki page would be appreciated by future users:
https://openwrt.org/toh/mikrotik/rb760igs

ajmac, 2025-02-24:
Just confirming both the issue and the fix - unfortunately I didn't find this bug report before troubleshooting it and coming up with the same solution myself, I could have saved myself 15 minutes of worry...

muxator, 2025-02-27:
Thanks @joseppc! I had this exact same problem and your suggestion helped!

This stanza in `/etc/config/system` fixes PoE passthrough:

```
config gpio_switch 'poe_passthrough'
    option name 'PoE Passthrough'
    option gpio_pin '529'  # <-- 24.10.0 contains '497', which is wrong
    option value '1'
```

I have edited the wiki at https://openwrt.org/toh/mikrotik/rb760igs#poe and mentioned this thread, so that future users may be warned:

[image]

It would also be good to check if the future 24.10.1 already contains a fix for this issue. I'll try to do that in the next days.

Thanks again.

Djfe, 2025-03-03:
If I understand john-tho's answer correctly then the solution would be to wipe the old config, isn't? (sysupgrade -n)
https://github.com/openwrt/openwrt/pull/15782#issuecomment-2184830912 

sadly this wasn't mentioned in the release notes, but yes the Kernel changed gpio numbers and OpenWrt had to adjust for that but didn't make migration scripts for the old config.
Either you wipe the old config or update it manually to the correct new gpio pin.

@DragonBluep or is there any other issue going on that I'm missing? Because OpenWrt already has the new gpio numbers in place thanks to your commit :)
https://github.com/openwrt/openwrt/commit/2f1ff5b4ceb420761154afb1b2e0704c5f057720

DragonBluep, 2025-03-05:
> If I understand john-tho's answer correctly then the solution would be to wipe the old config, isn't? (sysupgrade -n) [#15782 (comment)](https://github.com/openwrt/openwrt/pull/15782#issuecomment-2184830912)

Yes. A full reset should fix the issue. By the way, it is not recommended to keep the configuration in the main version upgrade.

> 
> sadly this wasn't mentioned in the release notes, but yes the Kernel changed gpio numbers and OpenWrt had to adjust for that but didn't make migration scripts for the old config. Either you wipe the old config or update it manually to the correct new gpio pin.
> 
> [@DragonBluep](https://github.com/DragonBluep) or is there any other issue going on that I'm missing? Because OpenWrt already has the new gpio numbers in place thanks to your commit :) [2f1ff5b](https://github.com/openwrt/openwrt/commit/2f1ff5b4ceb420761154afb1b2e0704c5f057720)

matjon, 2025-03-09:
Hello,

Responding to the mail on openwrt-devel ( https://lists.openwrt.org/pipermail/openwrt-devel/2025-March/043815.html ):

> However, meddling around the repos, I was not able to figure out how the 
> contents of /etc/config/system is put together by the build scripts.
> I also tried to manually mount a development snapshot in a linux machine 
> (playing with mtd, loop device and yaffs2), but I was not able to do it.

An easy way to extract the image is to use `binwalk`, for example:

```
binwalk -e openwrt-24.10.0-ramips-mt7621-mikrotik_routerboard-760igs-squashfs-sysupgrade.bin
```

After grepping the directory for `poe_passthrough`, I found `squashfs-root/etc/board.d/03_gpio_switches` with the following text both on 24.10.0 and snapshot:

```
case "$board" in
mikrotik,routerboard-760igs)
	ucidef_add_gpio_switch "poe_passthrough" "PoE Passthrough" "529"
	;;
```