openwrt/openwrt #17982: filogic: GL.iNet GL-MT6000 does not enter failsafe with button press

Lanchon, 2025-02-16:
### Describe the bug

during boot, the normal 5-per-second 2-second flashing occurs signaling the failsafe window. but if the only button (reset) is pressed during this window, the command is ignored and the device boots normally. the hardware button itself is confirmed to be functional.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

mediatek/filogic

### Device

GL.iNet GL-MT6000

### Image kind

Official downloaded image

### Steps to reproduce

press reset during failsafe window.

### Actual behaviour

boots normally.

### Expected behaviour

boots in failsafe mode.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

Fail-Safe, 2025-09-01:
I'm intrigued to see if others have input on this as well. I've run into a similar situation lately where I've tried with two different MT6000s to enter failsafe mode. Pressing the reset button during the "rapid blink" on boot-up causes the LED to flash rapidly. Per my experience with other devices, this indicates successful entry into failsafe mode. However, the MT6000 will not respond on 192.168.1.1 on any phy on the unit.

I realize my situation/explanation differs a bit from the OP, so I can split this off to a dedicated bug report if that would be best.

heydabop, 2025-11-03:
Still seems to be the case in 24.10.4 as well. 10-15 seconds after initial boot I get the "semi-rapid 5-per-second blinking rhythm during four seconds" but pressing the reset button once, multiple times, or holding it doesn't enter failsafe mode, and instead the blinking is followed up by the "slower, 2.5-per-second blink continuing to the end of normal boot, if the failsafe was not triggered", and the router boots as normal.

Blinking timing taken from https://openwrt.org/docs/guide-user/troubleshooting/failsafe_and_factory_reset#entering_failsafe_mode

kfix, 2025-12-28:
I am having this problem with 24.10.5, as @schmorp described it. on LAN1 port, I get 3 responded pings to .1 while the LED is fast blinking and then the MT6000 resumes booting normally. pressing reset during the fast blink does nothing.

kaindl, 2026-03-28:
Still present on 25.12.2!

failsafe-Mode seems to be at least very unstable on this device. I sometimes had to reboot and try again many many times (sometimes over 10 times!), until the LED indicated it going into failsafe mode. But when it boots into failsafe-mode, it is not responding to 192.186.1.1 on any Port. 
I can see 2 ARP-Requests for 192.168.1.2 from 192.168.1.1 on lan2-lan5 before the timebox for pressing the button is signaled by flashing LED, but not in lan1 or wan1. I can see the broadcast package to Port 4919 only on lan1, but not on any other port.

Seems like the only way to fix a broken network config on these devices currently is the procedure described by @kfix, but with that you loose all your data!

Lanchon, 2026-06-07:
i added a warning to the [wiki](https://openwrt.org/toh/gl.inet/gl-mt6000).

rereading this thread we see:

- users for whom the RESET click is ignored and the device boots normally.
- users for whom the RESET click works and the device boots into failsafe, but does not respond via the network on any LAN port.
- a user that can enter failsafe normally, but can only access failsafe via port LAN1.
- a user that needs several boots for the RESET click to enter failsafe, but then the router does not respond anyway.

one user reports that the device responds to three .1 pings during the failsafe window, but the pings and clicks are ignored, and the device boots normally. they see no no port 4919 broadcast.

one user gives this detailed report:

> I can see 2 ARP-Requests for 192.168.1.2 from 192.168.1.1 on lan2-lan5 before the timebox for pressing the button is signaled by flashing LED, but not in lan1 or wan1. I can see the broadcast package to Port 4919 only on lan1, but not on any other port.

@kfix when you say:

> I "fixed" it by reflashing (from boot recovery) the latest GLnet firmware, and then sys upgrading to OpenWRT. I guess the GLnet firmware that shipped with the router is not a suitable base for a direct sysupgrade?

do you mean:
- you fixed your previously misconfigured router but failsafe continues not to work,
- or you really fixed the router and now it enters failsafe with no issues?

@TheSideFX can you try @kfix fix and tell us which is it?

@zhaojh329 thanks for your work on this port. do you have any idea of what could be happening?

@dangowrt have you seen anything like this?

could this be dependent on the uboot version currently on the device?

the [switch](https://datasheet4u.com/download/1603028/MT7531.html) on this device has a special SGMII interface for the 2.5G PHY of LAN1 (https://openwrt.org/toh/gl.inet/gl-mt6000#switch). misconfiguration of the switch might be a cause for these issues, but still does not explain openwrt ignoring the RESET click (unless the failsafe-triggered code hangs while interacting with the hardware before it can signal the rest of openwrt that failsafe has been invoked?).

the OEM web recovery procedure only works via LAN2-5 ports. could we disable LAN1 at the switch at this early stage?

i wonder what would happen if we connect a dumb eth hub to ports LAN1, LAN2 and the PC.

TCB13, 2026-10-05:
Just got a new unit today and upgrade to openwrt using the glinet webui 25.12.5 (sysupgrade), unfortunately it seem to have bricked, the device initially booted and I got at IP via DHCP but no access to 192.168.1.1 on browser or ssh.

Tried to enter recovery as described in https://openwrt.org/toh/gl.inet/gl-mt6000#oem_recovery and doesnt' seem to work.

If I press the reset button before pluggin the power it eventually gets to a non-flashing LED status and can't open 192.168.1.1. If I press it a second after connecting the power it starts flashing very fast and nothing happens, can't open the page. All tests on port 2-5.

Lanchon, 2026-10-06:
hi @TCB13,

this issue is not a support forum for GL-MT6000. please find the GL-MT6000 thread in the openwrt forums and ask for help there. (if you did that already, then you shouldn't cross-post.) then please delete your post here: this bug is about non-working failsafe on an otherwise working router. thank you!

TCB13, 2026-10-06:
> hi [@TCB13](https://github.com/TCB13),
>
> this issue is not a support forum for GL-MT6000. please find the GL-MT6000 thread in the openwrt forums and ask for help there. (if you did that already, then [...]

Hi, the only reason why I posted was because the recovery didn’t seem to work as well. At the end the day nothing was working because there was an overlap between the subnet of the router an a VMware vmnet on my machine as described in the update.