openwrt/openwrt #17983: Issue with 5G Wifi Scanning on Xiaomi Router 3G

jfox52, 2025-02-16:
### Describe the bug

When the WiFi country code is set to CN, the 5GHz band cannot scan nearby WiFi networks after the first boot. This affects automatic channel selection and wireless repeater functionality. Manually restarting the 5GHz WiFi may restore normal operation intermittently.

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

ramips/mt7621

### Device

Xiaomi Mi Router 3G

### Image kind

Official downloaded image

### Steps to reproduce

Enable 5GHz WiFi and apply the following settings: select AC mode for the operating frequency, set the channel to auto, channel width to 80MHz, and set the country code to CN. Leave other settings at their default values.
Reboot the device.
After rebooting, open the LuCI interface, check the status, go to Channel Analysis, and switch to radio1 (5GHz). At this point, only "Local Interface" is visible.

### Actual behaviour

After booting the device, it is unable to scan nearby 5GHz WiFi signals.

### Expected behaviour

After booting the device, the 5GHz WiFi should be able to scan nearby 5GHz WiFi signals without needing several manual restarts.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-02-16:
You need to wait 1-10min for DFS to settle.

jfox52, 2025-02-17:
> You need to wait 1-10min for DFS to settle.

However, I found this issue by chance after OpenWrt had been running for several days. At that time, my WAN port was connected via a wired connection. Since there was no involvement of wireless relay, I didn’t notice the issue right away. Later, I happened to discover it on the Channel Analysis page.

closed 2025-02-17 by jfox52