openwrt/openwrt #17984: Kernel config problem

schm0, 2025-02-16:
Hello,
recently I have the problem when running a build that it fails because of unset kernel config parameters.
They all seem to be related to WMI.
But it seems that these paramters are actually disabled in the target's (x86_64, I think it was the generic one) kernel config.
What is happening there? Some module package causing this?
Thanks!

brada4, 2025-02-16:
Mind to fill issue template?

schm0, 2025-03-31:
sorry i had no time looking into this.
i will provide a template issue later.
a small update on this, adding the kernel symbols into the generic config seems to fix it.
But the question is why the symbols that are already preset in the x86/64/legacy config aren't taking into account.

Also why are kernel symbols preset in the 64,legacy config that are already preset in the x86 config?