openwrt/openwrt #17988: Support for Xiaomi Ax6000 - RA-72

ejklock, 2025-02-16:
With the recent support for Qualcomm Ipq50xx support on open wrt we will be able to support ax6000 router?

One user on the past make some progress to add the device 

 https://github.com/r1ghty/openwrt-redmi-ax3000/commit/dd2fe7a8b09937934f1d61cb24b7776b6f581b9f

Its the case to add these commits to add a full support for this router ?

namiltd, 2025-02-16:
You have reported your query in "Issues" which is the wrong place. Go to https://forum.openwrt.org/

calvin2021y, 2025-09-28:
[create_exploit.zip](https://github.com/user-attachments/files/22581887/create_exploit.zip) the ssh tools

calvin2021y, 2025-09-28:
hi @kaykim 

please help me confirm, how to install it?

```sh
cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00080000 00020000 "0:SBL1"
mtd1: 00080000 00020000 "0:MIBIB"
mtd2: 00040000 00020000 "0:BOOTCONFIG"
mtd3: 00040000 00020000 "0:BOOTCONFIG1"
mtd4: 00100000 00020000 "0:QSEE"
mtd5: 00100000 00020000 "0:QSEE_1"
mtd6: 00040000 00020000 "0:DEVCFG"
mtd7: 00040000 00020000 "0:DEVCFG_1"
mtd8: 00040000 00020000 "0:CDT"
mtd9: 00040000 00020000 "0:CDT_1"
mtd10: 00080000 00020000 "0:APPSBLENV"
mtd11: 00140000 00020000 "0:APPSBL"
mtd12: 00140000 00020000 "0:APPSBL_1"
mtd13: 00100000 00020000 "0:ART"
mtd14: 00080000 00020000 "0:TRAINING"
mtd15: 00080000 00020000 "bdata"
mtd16: 00080000 00020000 "crash"
mtd17: 00080000 00020000 "crash_syslog"
mtd18: 02400000 00020000 "rootfs"
mtd19: 02400000 00020000 "rootfs_1"
mtd20: 01f00000 00020000 "overlay"
mtd21: 00080000 00020000 "rsvd0"
mtd22: 003a2000 0001f000 "kernel"
mtd23: 015cc000 0001f000 "ubi_rootfs"
mtd24: 01b20000 0001f000 "data"
```

should I use `mtd write /tmp/openwrt-qualcommax-ipq50xx-xiaomi_ax6000-squashfs-factory.ubi ubi_rootfs` ? (suggestion by chatgpt)

or use `initramfs-factory.ubi` ?  or both?

kaykim, 2025-09-30:
Here is the installation method I've found from the OpenWRT git commit logs:

Flash instructions:

Download XMIR Patcher: https://github.com/openwrt-xiaomi/xmir-patcher

First flash a ubinized OpenWrt initramfs that will serve as the intermediate step, since
OpenWrt uses unified rootfs in order to fully utilize NAND and provide enough space for
packages, through either of the below two methods:

Installation via XMIR Patcher:

1. Load the initramfs image: openwrt-qualcommax-ipq50xx-xiaomi_ax6000-initramfs-factory.ubi

Installation via ubiformat method, through SSH:

1. If needed, enable SSH using XMIR Patcher.
2. Copy the file openwrt-qualcommax-ipq50xx-xiaomi_ax6000-initramfs-factory.ubi to the /tmp directory
3. Open an SSH shell to the router
4. Check which rootfs partition is your router booted in (0 = rootfs | 1 = rootfs_1):
nvram get flag_boot_rootfs
5. Find the rootfs and rootfs_1 mtd indexes respectively:
cat /proc/mtd
Please confirm if mtd18 and mtd19 are the correct indexes from above!
6. Use the command ubiformat to flash the opposite mtd with UBI image:
If nvram get flag_boot_rootfs returned 0:
ubiformat /dev/mtd19 -y -f /tmp/openwrt-qualcommax-ipq50xx-xiaomi_ax6000-initramfs-factory.ubi && nvram set flag_boot_rootfs=1 && nvram set flag_last_success=1 && nvram commit
otherwise:
ubiformat /dev/mtd18 -y -f /tmp/openwrt-qualcommax-ipq50xx-xiaomi_ax6000-initramfs-factory.ubi && nvram set flag_boot_rootfs=0 && nvram set flag_last_success=0 && nvram commit
7. Reboot the device by:
reboot

Continue in order to pernamently flash OpenWrt:
1. Upload the sysupgrade image to /tmp/ using SCP:
scp -O <path to image> root@192.168.1.1:/tmp/
2. Open an SSH shell to 192.168.1.1 from a PC within the same subnet
3. Use sysupgrade to flash the sysupgrade image:
sysupgrade -n -v /tmp/openwrt-qualcommax-ipq50xx-xiaomi_ax6000-squashfs-sysupgrade.bin

Device will reboot with OpenWrt, and then sysupgrade can be used to upgrade the device when desired.

Source:
https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=d7f9e240c208891a7d26d7a1d308cabd70618cae

I hope this will help you and you'll let me know the outcome. Good luck!

VincxBlox, 2025-11-22:
Hello guys, I bought this router now. I have managed to boot into InitRamFS succesfully. Just in case it's useful, I was originally on the last fw, 1.0.122, but now after I try to sysupgrade, it flashes, but then reboots and get stucks in the bootloader I think. Like there something preventing me from doing this. I can't tell what is happening. The switch doesn't turn on, and the orange led just like lights up then back off after like 5-10second. I managed to recover with tftp, but even with MTD, flashing the rootfs file extracted myself, no, nothing works. I'm not sure how @calvin2021y  got it working. maybe the new firmware I have has simply patched the bootloader. Regardless, I also tried older snapshots and compiling my own from a commit a day before @calvin2021y  made his comment, yet it never booted. I think that's because the kernel ubi is write locked. And tftp checks the images. And booting a super old kernel with a new openwrt might not work. stock firmware on this is openwrt 18.06 snapshot. anyone has any ideas? @kaykim  maybe cuz i really don't know.

VincxBlox, 2025-11-30:
@kaykim hi, i found that this revision of mine is different.. in some way. the bootloader does some different things. cause, i got inside of it, and saw that when it tries to sysupgrade it couldnt see MTD19. which I believe is ubi rootfs. so the bootloader or kernel didn't mount it? I'm not sure about that. but that's why it kept bricking. I have logs that ill try to upload soon, but I did get openwrt flashed thru uart. rootfs_data didn't work for some reason, but I cannot experience anymore, because it blew a coil, and that also blew a resistor, and then that made the PMIC refuse to kick on any other rail than the 5volts and 2volts. flawed design. happent when I unplugged it and plugged it back in.. I have found stock tftp recovery to work fine (does check signatures and such), tftpboot allows getting images into RAM into uboot console, and that the "press any key to stop autoboot" timer is set to 0 in stock firmware. the other one I bought, because the dude I bought it from had another one, worked completely fine. While openwrt refused to utilize the 4 spacial streams, yet showing in iw list that it does support it, I got 650mbps on ch36 160mhz 24dbm AX. disappointed that I can't get 4 streams working, no errors in the logs about that, but I'll make a openwrt wiki device page about this one. but i won't be opening this one.

tl;dr one revision doesn't allow it to be flashed without UART access, and when it does work only 2 spacial streams work (do you know anything about that btw? I tried patching the kernel driver to force 4 tx and rx chains but It didn't seem to apply after reflashing it..)

thanks everyone. ill be uploading stock logs and openwrt logs and try to bring some proper support to this..

VincxBlox, 2025-12-06:
if anyone has a stock router with them, I'd really need the wifi calibration/board data files. the OpenWRT one limit us to 64QAM and two streams. right now, I cannot test more on mine since it is currently heavily relied on. if anyone could use xmir patcher, get ssh access and see the files in ```/lib/firmware/```. most likely in another subfolder (such as ath11k) but if you could just upload it all that's help. @sund00bie perhaps you could help, seeing as you have your own? btw the driver also doesn't currently have proper routing wan routing.. so pppoe is capped to around 650mbps. dhcp works fine tho. would need to recompile the kernel with the proper modules to get it working, and I can't be bothered to do so currently.

sund00bie, 2025-12-07:
> if anyone has a stock router with them, I'd really need the wifi calibration/board data files. the OpenWRT one limits us to two streams. right now, I cannot test more on mine since it is currently heavily relied on. if anyone could use xmir patcher, get ssh access and see the files in `/lib/firmware/`. most likely in another subfolder (such as ath11k) but if you could just upload it all that's help. [@sund00bie](https://github.com/sund00bie) perhaps you could help, seeing as you have your own? btw the driver also doesn't currently have proper routing wan routing.. so pppoe is capped to around 650mbps. dhcp works fine tho. would need to recompile the kernel with the proper modules to get it working, and I can't be bothered to do so currently.

@VincxBlox I'll have a look at this tonight and see if can get you the files you need

Mine is currently stock and is my daily device so don't went to flash openwrt untill it's working better. 

Let you know how I go

VincxBlox, 2025-12-07:
@sund00bie I managed to pull the files from the firmware file directly, but u could still be of help. in the stock firmware, is there a switch to use pppoe? or is it dhcp only? if they never implemented pppoe then I won't either, probably some hardware reason. and do you see any revision number other than the version anywhere in the web UI? cuz im trying to find what made my first one unflashable without uart. and the label on it doesn't say much. currently in the process of adapting the wifi calib files to work on the newer openwrt.. I have no idea wtf they did with all these. whatever you do, keep us updated, have a good night

sund00bie, 2025-12-07:
> [@sund00bie](https://github.com/sund00bie) I managed to pull the files from the firmware file directly, but u could still be of help. in the stock firmware, is there a switch to use pppoe? or is it [...]

Glad you ot the files. Thanks for all your efforts! 

Attached are some screenshots that confirm there is the option for dhcp and pppoe.

Doesn't look there is anything unique about revision numbers? Maybe the number before the serial number?

[image]
[image]
[image]

sund00bie, 2025-12-23:
> thanks, but for your information, as long as you don't need like full gigabit internet, it is currently stable. no 4x4 yet but im looking into that. has been running fine for a week or so on my end. and it doesn't look like u use pppoe. but, just in case, could you check if your label says "fuwu.mi.com" or just "mi.com"
> 
> if your just says mi.com im not sure but there a good chance it will simply refuse to flash without UART access (opening it..) that's the only difference between the two units. the different labels, which the things on it are differently placed and have a different url.

Sorry I missed this. 

When you say check the label, do you mean on the physical device or on a page?

j4801, 2025-12-23:
I only managed to downgrade to 1.0.85

VincxBlox, 2025-12-23:
@sund00bie on the physical device. I'm not sure if this is what determines it, but @j4801 you should also check your label so I can see if that is actually what determines if you can or not. but regardless of downgrading, my unit that works, was on the latest version when I did it, so was the bootloader. so I don't think it's a matter of such.

VincxBlox, 2026-01-04:
@velox11 ive been using mine as my home router for the last month completely fine. you will have to get by with 650-700mbps total routing speed, and 2x2 wifi instead of 4x4 tho. other than that i still prefer it due to its range and insane beamforming

VincxBlox, 2026-01-09:
hi @asaranovx 

with my revision and my openwrt from around november, it works fine on my end. What wan are you using

IRQBalance does not seem to provide an advantage on this router. I have found setting the packet steering to all cpu and setting the RPS steerings flows under it in LuCi to 1024 to help a tiny bit.

If nothing you do fixes it, I'd be happy to provide my files that I used for the process in case its version related. on my pppoe session the ping ranges for 3-8ms, never more. so it's probably not hardware. unless you have another revision that's is somehow majorly different...

craniusfawkes, 2026-10-10:
For what I have tried and observed, these new builds for ax6000 RA72 work, but there are still some problems that need to get fixed. When you install it, the best of download rate you might have is about 550Mbps, since it has no NSS integrated yet and that is what processes the PPPoE and some other packets. Without NSS, the processor will be the one processing everything and its processor won't be enough. I have found someone developing some new builds with NSS in GitHub, but it showed some problems. Though the download reached the peak of my plan (750Mbps), the upload got limited to 95Mbps and also some problems related to the 2Ghz radio... It is not perfect yet but I guess soon everything might be fixed and we might have a great router with OpenWRT.

craniusfawkes, 2026-11-09:
Good news, guys. There is finally a great new firmware update with NSS working for ax6000 ra72 without any issues until now. 

Page: https://github.com/kuncy7/openwrt-nss-edma/releases/tag/ipq50xx-2026.11.07
Direct link: https://github.com/kuncy7/openwrt-nss-edma/releases/download/ipq50xx-2026.11.07/openwrt-qualcommax-ipq50xx-xiaomi_ax6000-squashfs-sysupgrade.bin

velox11, 2026-11-14:
Hi
I successfully flashed ipq50xx-2026.11.07 on my Xiaomi AX6000 (RA72). The system boots fine, but the WAN interface cannot obtain an IP address via DHCP.

Here are the details from my diagnostics:

Physical link is up: The kernel log shows wan: Link is Up - 1Gbps/Full, and ip link show wan reports state UP and LOWER_UP.

NSS offload is active: Logs show qca-dwmac-nss: NSS data plane override on phys_if 1 and phys_if 1 on NSS fw data plane.

No RX packets: ip -s link show wan shows RX: 0 packets (0 bytes, 0 errors), while TX shows some packets.

DHCP fails: Running udhcpc -i wan -f -q -t 5 results in broadcasting discover repeatedly, but no offer is received.

MAC address is configured correctly: I set the WAN MAC address to the one required by my ISP. I did this entirely via SSH commands using UCI:

It seems the GMAC1 port (WAN) is not receiving any packets on the NSS data plane, even though the PHY link is established. I also tried reverting to the stock Xiaomi firmware and the internet works fine, so the hardware and ISP are not the issue.

Could you please advise if there is a fix or a workaround (e.g., moving GMAC1 descriptor rings to SDRAM) for this specific issue on the AX6000?

I'm new to OpenWRT maybe I missed something simple .. For the moment I returned to the stock fw because it is my main home router.