openwrt/openwrt #17998: ZTE MF286R can't use 4G modem after upgrade to 24.10

pcislocked, 2025-02-17:
### Describe the bug

I used attended sysupgrade to upgrade my device from 23.05 to 24.10. After this, the 4G modem stopped working. 

[image]

Network menu says Network device is not present, however, 3ginfo-lite reports modem as connected to 4G;

[image]

Can provide logs if needed. lmk what i need to look at pls thx

`root@ist-gw-2:~# cat /etc/config/network

config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fdae:17ef:ca5c::/48'

... (around 10-15 various interfaces later)

config interface 'wwan'
        option proto 'ncm'
        option force_link '1'
        option device '/dev/ttyACM0'
        option pdptype 'IP'
        option apn 'internet'
        option ipv6 '0'
        option delay '30'
        list dns '1.1.1.1'
        option metric '250'
        option mode 'preferlte'

...

root@ist-gw-2:~#`

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

ath79/nand

### Device

ZTE MF286R

### Image kind

Official downloaded image

### Steps to reproduce

Upgrade from 23.05 to 24.10 on zte mf286r

### Actual behaviour

4gmodem fails to initialize

### Expected behaviour

modem works

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

pcislocked, 2025-02-17:
Sure, here are the full logs after a fresh reboot;

```
Using username "root".
Error writing session log (ASCII mode) to file: putty.log
Authenticating with public key ""

BusyBox v1.36.1 (2025-02-14 10:31:13 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 24.10.0, r28427-6df0e3d02a
 -----------------------------------------------------
root@ist-gw-2:~# logger Hi there, from ssh

root@ist-gw-2:~# dmesg
[    0.000000] Linux version 6.6.73 (builder@buildhost) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r28427-6df0e3d02a) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 Mon Feb  3 23:09:37 2025
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
[    0.000000] MIPS: machine is ZTE MF286R
[    0.000000] SoC: Qualcomm Atheros QCA956X ver 1 rev 0
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 119000K/131072K available (7684K kernel code, 597K rwdata, 968K rodata, 1240K init, 240K bss, 12072K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] CPU clock: 775.000 MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4932285024 ns
[    0.000001] sched_clock: 32 bits at 388MHz, resolution 2ns, wraps every 5541893118ns
[    0.008305] Calibrating delay loop... 385.84 BogoMIPS (lpj=1929216)
[    0.074798] pid_max: default: 32768 minimum: 301
[    0.089174] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.096907] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.113446] RCU Tasks Trace: Setting shift to 0 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=1.
[    0.127979] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.138396] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.148080] pinctrl core: initialized pinctrl subsystem
[    0.156705] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.163475] thermal_sys: Registered thermal governor 'step_wise'
[    0.165472] /ahb/apb: Fixed dependency cycle(s) with /ahb/apb/interrupt-controller@18060010
[    0.193151] clocksource: Switched to clocksource MIPS
[    0.208777] NET: Registered PF_INET protocol family
[    0.214219] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.223257] tcp_listen_portaddr_hash hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.232191] Table-perturb hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.240382] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.248474] TCP bind hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.255941] TCP: Hash tables configured (established 1024 bind 1024)
[    0.263385] MPTCP token hash table entries: 512 (order: 0, 6144 bytes, linear)
[    0.271232] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
[... 772 lines cut ...]
Mon Feb 17 20:10:11 2025 daemon.info dnsmasq[1]: using nameserver 45.90.28.185#53
Mon Feb 17 20:10:11 2025 daemon.info dnsmasq[1]: using nameserver 45.90.30.185#53
Mon Feb 17 20:10:11 2025 daemon.info dnsmasq[1]: using nameserver 1.0.0.1#53
Mon Feb 17 20:10:11 2025 daemon.info dnsmasq[1]: using nameserver 8.8.8.8#53
Mon Feb 17 20:10:11 2025 daemon.info dnsmasq[1]: using only locally-known addresses for test
Mon Feb 17 20:10:11 2025 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Mon Feb 17 20:10:11 2025 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Mon Feb 17 20:10:11 2025 daemon.info dnsmasq[1]: using only locally-known addresses for local
Mon Feb 17 20:10:11 2025 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Mon Feb 17 20:10:11 2025 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Mon Feb 17 20:10:11 2025 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Mon Feb 17 20:10:11 2025 user.notice root: Hi there, from ssh
[... cut ...]
Mon Feb 17 20:10:18 2025 daemon.info hostapd: phy1-ap0: STA 00:00:00:00:00:00 IEEE 802.11: associated (aid 1)
Mon Feb 17 20:10:18 2025 daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED 00:00:00:00:00:00 auth_alg=open
Mon Feb 17 20:10:18 2025 daemon.info hostapd: phy1-ap0: STA 00:00:00:00:00:00 RADIUS: starting accounting session B17782DCFEB7EB94
Mon Feb 17 20:10:18 2025 daemon.info hostapd: phy1-ap0: STA 00:00:00:00:00:00 WPA: pairwise key handshake completed (RSN)
Mon Feb 17 20:10:18 2025 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED 00:00:00:00:00:00
root@ist-gw-2:~#

```

pcislocked, 2025-02-18:
Hi again, I think I saw that one and tried it, didn't work. Will try again, rebooting now, maybe it was me being a clumsy dumbass

pcislocked, 2025-02-18:
Yep, still not working.

```
Using username "root".
Error writing session log (ASCII mode) to file: putty.log
Authenticating with public key ""

BusyBox v1.36.1 (2025-02-14 10:31:13 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 24.10.0, r28427-6df0e3d02a
 -----------------------------------------------------
root@ist-gw-2:~# nano /lib/netifd/proto/ncm.sh
root@ist-gw-2:~# cat /lib/netifd/proto/ncm.sh
#!/bin/sh

[ -n "$INCLUDE_ONLY" ] || {
        . /lib/functions.sh
        . ../netifd-proto.sh
        init_proto "$@"
}

proto_ncm_init_config() {
        no_device=1
        available=1
        proto_config_add_string "device:device"
        proto_config_add_string ifname
        proto_config_add_string apn
        proto_config_add_string auth
        proto_config_add_string username
        proto_config_add_string password
        proto_config_add_string pincode
        proto_config_add_string delay
        proto_config_add_string mode
        proto_config_add_string pdptype
        proto_config_add_boolean sourcefilter
        proto_config_add_boolean delegate
        proto_config_add_int profile
        proto_config_add_defaults
}

proto_ncm_setup() {
        local interface="$1"

        local manufacturer initialize setmode connect finalize devname devpath ifpath

        local device ifname  apn auth username password pincode delay mode pdptype profile $PROTO_DEFAULT_OPTIONS
        json_get_vars device ifname apn auth username password pincode delay mode pdptype sourcefilter delegate profile $PROTO_DEFAULT_OPTIONS

        local context_type

        [ "$metric" = "" ] && metric="0"

        [ -n "$profile" ] || profile=1

        pdptype=$(echo "$pdptype" | awk '{print toupper($0)}')
        [ "$pdptype" = "IP" -o "$pdptype" = "IPV6" -o "$pdptype" = "IPV4V6" ] || pdptype="IP"

        [ "$pdptype" = "IPV4V6" ] && context_type=3
        [ -z "$context_type" -a "$pdptype" = "IPV6" ] && context_type=2
        [ -n "$context_type" ] || context_type=1

        [ -n "$ctl_device" ] && device=$ctl_device

        [ -n "$device" ] || {
[... 254 lines cut ...]
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 24.10.0, r28427-6df0e3d02a
 -----------------------------------------------------
root@ist-gw-2:~# cat /lib/netifd/proto/ncm.sh
#!/bin/sh

[ -n "$INCLUDE_ONLY" ] || {
        . /lib/functions.sh
        . ../netifd-proto.sh
        init_proto "$@"
}
...

        start=$(date +%s)
        while true; do
                manufacturer=$(gcom -d "$device" -s /etc/gcom/getcardinfo.gcom | awk 'NF && $0 !~ /AT\+CGMI/ { sub(/\+CGMI: /,""); print tolower($1); exit; }' | tr -d '\r\n')
                [ "$manufacturer" = "error" ] && {
                        manufacturer=""
                }
...
root@ist-gw-2:~#

```

pcislocked, 2025-02-18:
also just to fuck around, i ran that command few times. sometimes it outputs twice. sometimes it does nothing. sometimes it errors out. basically modem is being a dummy here.

```root@ist-gw-2:~# gcom -d "/dev/ttyACM0" -s /etc/gcom/getcardinfo.gcom

+CGMI: "Marvell"

OK

+CGMI: "Marvell"

OK
root@ist-gw-2:~# gcom -d "/dev/ttyACM0" -s /etc/gcom/getcardinfo.gcom

ERROR
root@ist-gw-2:~# gcom -d "/dev/ttyACM0" -s /etc/gcom/getcardinfo.gcom

+CGMI: "Marvell"

OK
root@ist-gw-2:~# gcom -d "/dev/ttyACM0" -s /etc/gcom/getcardinfo.gcom
root@ist-gw-2:~# gcom -d "/dev/ttyACM0" -s /etc/gcom/getcardinfo.gcom
root@ist-gw-2:~# gcom -d "/dev/ttyACM0" -s /etc/gcom/getcardinfo.gcom
root@ist-gw-2:~# gcom -d "/dev/ttyACM0" -s /etc/gcom/getcardinfo.gcom

+CGMI: "Marvell"

OK
root@ist-gw-2:~# gcom -d "/dev/ttyACM0" -s /etc/gcom/getcardinfo.gcom

+CGMI: "Marvell"

OK
root@ist-gw-2:~#
```

pcislocked, 2025-02-18:
DUDE. @naseef . sorry to bother yall here again, no offense, but i fixed it with chatgpt im still laughing lmaoo. its such a stupid fix

> It appears that the JSON lookup is still failing because in your JSON file the marvell key is defined with extra quotes. Notice in your `/etc/gcom/ncm.json` the entry is:
> 
> ```json
> "\"marvell\"": {
>     "initialize": [
>         "AT+CFUN=1"
>     ],
>     "configure": [
>         "AT+ZGDCONT=${profile},\\\"${pdptype}\\\",\\\"${apn}\\\",\\\"\\\",0,0",
>         "AT+ZGPCOAUTH=${profile},\\\"${username}\\\",\\\"${password}\\\",0"
>     ],
>     "connect": "AT+ZGACT=1,${profile}",
>     "disconnect": "AT+ZGACT=0,${profile}"
> }
> ```
> 
> This means the key in the JSON object is literally `"marvell"` (including the quotes). In your current patch you set:
> 
> ```sh
> manufacturer="marvell"
> ```
> 
> which does not match the key in the JSON. To fix the lookup, you can either change your JSON file to use a key without the extra quotes or adjust your hardcoded value.
> 
> For a quick test you can update your script to:
> 
> ```sh
> manufacturer="\"marvell\""
> ```
> 
> That is, include the literal double quotes around marvell so that the JSON lookup finds the matching key. Save the change and then try bringing the interface up again.
> 
> Let me know if that clears the “Unsupported modem” error and causes the network interface to be created.

https://chatgpt.com/share/67b51364-3c24-8006-a5eb-9c3614d151b1

Now, let's integrate this to a proper fix

Also, can you install zerotier and confirm that it's also crash looping on you as well? not sure if this is upgrade related or not

```
Wed Feb 19 01:40:02 2025 daemon.err zerotier-one[15537]: Starting Control Plane...
Wed Feb 19 01:40:02 2025 daemon.err zerotier-one[15537]: Starting V6 Control Plane...
Wed Feb 19 01:40:02 2025 daemon.notice netifd: Interface 'zerotier' is enabled
Wed Feb 19 01:40:02 2025 daemon.notice netifd: Network device 'zt44xlzd3z' link is up
Wed Feb 19 01:40:02 2025 daemon.notice netifd: Interface 'zerotier' has link connectivity
Wed Feb 19 01:40:02 2025 daemon.notice netifd: Interface 'zerotier' is setting up now
Wed Feb 19 01:40:02 2025 daemon.notice netifd: Interface 'zerotier' is now up
Wed Feb 19 01:40:05 2025 user.notice mwan3-hotplug[15580]: mwan3 hotplug on zerotier not called because interface disabled
Wed Feb 19 01:40:05 2025 user.notice firewall: Reloading firewall due to ifup of zerotier (zt44xlzd3z)
Wed Feb 19 01:40:08 2025 kern.info kernel: [  475.494808] do_page_fault(): sending SIGSEGV to zerotier-one for invalid write access to 00000002
Wed Feb 19 01:40:08 2025 kern.info kernel: [  475.504039] epc = 77e468a8 in libc.so[77dc0000+b2000]
Wed Feb 19 01:40:08 2025 kern.info kernel: [  475.509337] ra  = 77e47288 in libc.so[77dc0000+b2000]
Wed Feb 19 01:40:08 2025 daemon.notice netifd: Network device 'zt44xlzd3z' link is down
Wed Feb 19 01:40:08 2025 daemon.notice netifd: Interface 'zerotier' has link connectivity loss
Wed Feb 19 01:40:08 2025 daemon.notice netifd: Interface 'zerotier' is now down
Wed Feb 19 01:40:08 2025 daemon.info procd: Instance zerotier::instance1 s in a crash loop 6 crashes, 6 seconds since last crash
```

Zredy, 2025-02-28:
Hello,
I had the same issue here on my ZTE MF286R router, on the first install it booted up perfectly, but after a restart i began having the same issues.
The initial change to line 93 in `/lib/netifd/proto/ncm.sh` got me to the actual error of `Failed to get modem information`.
After changing that same line to `manufacturer="\"marvell\""`, basically forcing the router to check for the exact modem, with a reboot everything began working again.
I tried fiddling with the key in `ncm.sh` as this solution does not sit right with me, tried changing it to `"marvell"`, `"Marvell"`,, `"\"Marvell\""` but to no avail.
I can also confirm on my end that the command `gcom -d "/dev/ttyACM0" -s /etc/gcom/getcardinfo.gcom` sometimes returns absolutely nothing, sometimes an `Segmentation fault` error and sometimes the expected manufacturer output.

```
root@OpenWrt:~# gcom -d "/dev/ttyACM0" -s /etc/gcom/getcardinfo.gcom
root@OpenWrt:~# gcom -d "/dev/ttyACM0" -s /etc/gcom/getcardinfo.gcom
Segmentation fault
root@OpenWrt:~# gcom -d "/dev/ttyACM0" -s /etc/gcom/getcardinfo.gcom

+CGMI: "Marvell"

OK
root@OpenWrt:~# gcom -d "/dev/ttyACM0" -s /etc/gcom/getcardinfo.gcom
Segmentation fault
root@OpenWrt:~# gcom -d "/dev/ttyACM0" -s /etc/gcom/getcardinfo.gcom

+CGMI: "Marvell"

OK
root@OpenWrt:~# gcom -d "/dev/ttyACM0" -s /etc/gcom/getcardinfo.gcom
Segmentation fault
root@OpenWrt:~# gcom -d "/dev/ttyACM0" -s /etc/gcom/getcardinfo.gcom

+CGMI: "Marvell"

OK
root@OpenWrt:~# gcom -d "/dev/ttyACM0" -s /etc/gcom/getcardinfo.gcom

+CGMI: "Marvell"

OK
root@OpenWrt:~# gcom -d "/dev/ttyACM0" -s /etc/gcom/getcardinfo.gcom
Segmentation fault
```

I'm not that indept with parsing data so I cannot add much to this, will try to fiddle around further.
If anyone has any ideas worth testing I am more than happy to test them out.

shalzz, 2025-03-01:
I have a similar issue on a Huawei E3372s, where the AT command returns a slightly different output:

```
root@OpenWrt:~# gcom -d "/dev/ttyUSB0" -s /etc/gcom/getcardinfo.gcom

huawei

OK

^RSSI:27
root@OpenWrt:~# gcom -d "/dev/ttyUSB1" -s /etc/gcom/getcardinfo.gcom  | awk 'NF && $0 !~ /AT\+CGMI/ { sub(/\+CGMI: /,""); print tolower($1); exit; }'

```
so here in the output above you can see the AWK then gets and empty string

After I hardcoded the `manufacturer` variable in `/lib/netifd/proto/ncm.sh` to `huawei` then the interface comes up fine and everything works.
However after the interface is working, trying again the script/command again, the modem seems to return the  correct values for the script to parse as expected.

```
root@OpenWrt:~# gcom -d "/dev/ttyUSB0" -s /etc/gcom/getcardinfo.gcom  | awk 'NF && $0 !~ /AT\+CGMI/ { sub(/\+CGMI: /,""); print tolower($1); exit; }'
huawei
root@OpenWrt:~# gcom -d "/dev/ttyUSB0" -s /etc/gcom/getcardinfo.gcom
AT+CGMI
huawei

OK
```

Zredy, 2025-03-01:
> [@Zredy](https://github.com/Zredy) is it possible for you to test the command `gcom -d "/dev/ttyACM0" -s /etc/gcom/getcardinfo.gcom ` on openwrt 23.05?

I just tested it over 30 times, the output is normal every time. No Segmentation faults whatsoever.

Zredy, 2025-03-01:
> > > [@Zredy](https://github.com/Zredy) is it possible for you to test the command `gcom -d "/dev/ttyACM0" -s /etc/gcom/getcardinfo.gcom ` on openwrt 23.05?
> > 
> > 
> > I just tested it over 30 times, the output is normal every time. No Segmentation faults whatsoever.
> 
> Can you post the output here since it is reported that an extra new line (`\r\n`) is there in 24.10 and not in 23.05 ?

Here you go:
`
root@OpenWrt:~# gcom -d "/dev/ttyACM0" -s /etc/gcom/getcardinfo.gcom

+CGMI: "Marvell"

OK
root@OpenWrt:~# gcom -d "/dev/ttyACM0" -s /etc/gcom/getcardinfo.gcom

+CGMI: "Marvell"

OK
root@OpenWrt:~# gcom -d "/dev/ttyACM0" -s /etc/gcom/getcardinfo.gcom

+CGMI: "Marvell"

OK
root@OpenWrt:~# gcom -d "/dev/ttyACM0" -s /etc/gcom/getcardinfo.gcom

+CGMI: "Marvell"

OK
root@OpenWrt:~# gcom -d "/dev/ttyACM0" -s /etc/gcom/getcardinfo.gcom

+CGMI: "Marvell"

OK
root@OpenWrt:~# gcom -d "/dev/ttyACM0" -s /etc/gcom/getcardinfo.gcom

+CGMI: "Marvell"

OK
root@OpenWrt:~# gcom -d "/dev/ttyACM0" -s /etc/gcom/getcardinfo.gcom

+CGMI: "Marvell"

OK
`
Line 91 in ncm.sh is as follows: `manufacturer=$(gcom -d "$device" -s /etc/gcom/getcardinfo.gcom | awk 'NF && $0 !~ /AT\+CGMI/ { sub(/\+CGMI: /,""); print tolower($1); exit; }')`

pcislocked, 2025-04-08:
I will reset my MF286R to install latest firmware that includes BDF so I can finally use 5GHz without it being a pain in the ass. I will re-test this issue when I do so and let y'all know how it is currently.ş

pcislocked, 2025-08-17:
> I will reset my MF286R to install latest firmware that includes BDF so I can finally use 5GHz without it being a pain in the ass. I will re-test this issue when I do so and let y'all know how it is currently.ş

Ooof completely forgot. Anyway, 24.10 fresh install works withouy any .sh editing or whatever, on my device at least.