openwrt/openwrt #18002: please help with Huasifei wh3000 device

Doanduy09, 2025-02-17:
The HUASIFEI WH3000 and emmc model ( aka Fudy MT3000) are Wi-Fi 6 5G cellular routers, based on MediaTek MT7981A SoC.

Specifications:

SoC: Filogic 820 MT7981A (1.3GHz)
RAM: DDR4 1GB
Flash: eMMC 8G
WiFi: 2.4GHz and 5GHz with 3 antennas
Ethernet:
1x WAN (10/100/1000M)
1x LAN (10/100/1000/2500M)
USB: 1x USB 3.0 port
UART:
3.3V, TX, RX, GND / 115200 8N1

wgetge, 2025-02-17:
https://github.com/coolsnowwolf/lede/commits/master/target/linux/mediatek/dts/mt7981b-huasifei-wh3000-emmc.dts

fildunsky, 2025-02-18:
There is an immortalWRT 24.10.0 release for it. Hopefully someone could port it to official OpenWRT as well

Doanduy09, 2025-02-18:
it's already on immor but it doesn't have MDT and Hnat drivers

fildunsky, 2025-02-18:
> it's already on immor but it doesn't have MDT and Hnat drivers

I'm sorry, I don't know what is that. I will receive this router tomorrow. Does that mean I will not be able to normally use ImmortalWRT and it's better to keep qwrt for now?

Doanduy09, 2025-02-18:
> > it's already on immor but it doesn't have MDT and Hnat drivers
> 
> I'm sorry, I don't know what is that. I will receive this router tomorrow. Does that mean I will not be able to normally use ImmortalWRT and it's better to keep qwrt for now?

It is related to the new wifi transmission efficiency.

hecatae, 2025-02-26:
Pull request is here, https://github.com/openwrt/openwrt/pull/18118

This issue can be closed

Doanduy09, 2025-02-28:
> Pull request is here, [#18118](https://github.com/openwrt/openwrt/pull/18118)
> 
> This issue can be closed

You added mtk drive too, it's great

fildunsky, 2025-03-27:
Please review and approve, guys. The second month almost passed...
https://github.com/openwrt/openwrt/pull/18220

kkpcc33-stack, 2026-08-31:
Hello I bought a Huasifei WH3000 pro router a few days ago I received it and then I tried to update it to a newer version and I downloaded the update from your site github and indeed an update started and after two minutes of it started an update and flashed red then it stopped working and now when it turns on it flashes for a few seconds then it settles to red and does not show Wi-Fi and the factory reset button does not respond please help

kkpcc33-stack, 2026-09-02:
@fildunsky I want a backup that contains the openwrt system with a modem that recognizes the FM350-gl modems as well as the seria EM7455 DW5811e I don't know how to use that as I found this in immortalwrt but the fan is not working please send me a ready openwrt version including your application for the modem as well as the fan and the openwrt system I don't like to download

kkpcc33-stack, 2026-09-02:
@fildunsky The system I have is immortalwrt and the modem works as well as the router but it overheats a lot with the fan not working so I want to try openwrt with the modem working as well as a cooling fan and download the v2raya feature to bypass the blocking that's all I care about